cartographer-project/cartographer_ros #797: Localization failure with a Velodyne 64E Sensor in real vehicle

tigerk0430, 2018-04-02:
hello,

when I ran cartograhper `demo_backpack3d.launch` , 
It used 2 lidar sensor which are horizontal and vertical. 

but our vehicle system has just `one Velodyne 64E Lidar sensor` which surrounds 360 degree scenes. 

[image]
and It looks good but It's not making a map properly as much as demo file (I think it's because of the number of sensor).

To use cartographer's localization feature in our real vehicle system, I saved earlier `180320_dyrosvehicle_path1.pbstream` map file and load it with

`roslauch cartographer_ros dyros_test.launch map_filename:=/path/to/pbstream/180320_dyrosvehicle_path1.pbstream`

[image]

and I started driving in near start point (around 2m) and drive through parking lots. 
but It fails to merge 2 trajectories to one map :-/ 

[image]

Question is
- what is a major reason of failure to merge 2 different trajectories? (failure of Localization)

I think It's because of lack of map data against one velodyne sensor, but I'm not sure of it.

<br><br><br>
sorry for not having a bag file and I'm just having `.pbstream` files and I'll record a bag file soon
http://gofile.me/3TjkG/1RJiAbKLZ

tigerk0430, 2018-04-06:
Thank for the reply. 

additional screenshots are here.

[image]

[image]

[image]

### tf tree for our vehicle
----
[image]

### procedure was 
---
1. turn on `cartographer_ros` and moving around parking lots several times (started from the garage)
2. save an offline map using `cartographer_ros` (`rosservice call /write_state ~/Downloads/map.pbstream`)
3. shutdown all `cartographer_ros` programs
4. move vehicle to the front of the garage. 
5. rerun `cartographer_ros` with `map_filename:=~/Downloads/map.pbstream` 
6. then there are 2 trajectories ( 0: offline map trajectory,  1: current online trajectory)

<br><br>

gaschler, 2018-04-06:
Please check what types of constraints and how many have been found.
In rviz, enable Display -> Add -> By topic -> /constraint_list -> MarkerArray.
Then, again in the Display sidebar, expand MarkerArray -> Namespaces and click through the different kinds of constraints.
If there are no INTER constraints across different trajectories, this is the cause and needs to be fixed first.

Also, in the launch file, make sure to set argument `-keep_running=true` to the node so rviz is updated after the bag ran through.

tigerk0430, 2018-04-06:
@gaschler 

well.. I couldn't find any constraint between 2 trajectories..

I think It's because I don't have a `.bag` file and I just have `.pbstream` file that include 2 trajectories
then It opens just static offline map with 2 trajectories like pictures above.
and I don't think I put `-keep_running=true` argument into my `cartographer_ros`

I have to figure it out more deeply with recording `.bag` files. 

Thanks for the kind reply :-)   I'll close this issue

closed 2018-04-06 by tigerk0430

gaschler, 2018-04-09:
Oh, you are only opening pbstreams.
When you open a pbstream with `load_state_filename`, the default behavior is that its trajectories are frozen and no constraints are searched.
You can try again with `-load_frozen_state false`, but this is a rather special feature and not the typical usage of Cartographer. 
How to use arguments: When you start with roslaunch, you usually what to edit the .launch file and add the arguments to `args=""`.
Hope this helps.

tigerk0430, 2018-04-09:
@gaschler 

awesome!  I haven't known those commands so far. I'll try to do it 
Thank you for kind help :-)