cartographer-project/cartographer_ros #798: problems with use_nav_sat=true

elgarbe, 2018-04-02:
As suggested [here](https://github.com/googlecartographer/cartographer_ros/issues/795#issuecomment-377954158) I'm trying to add nav sat fix to my application.
So I set use_nav_sat=true in backpack_2d.lua and add a remap in backpack_2d.launch

```
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename backpack_2d.lua"
      output="screen">
	<remap from="imu" to="/mavros/imu/data" />
	<remap from="fix" to="/mavros/global_position/raw/fix" />
  </node>
```
I recompile and run, but get this error:

```
 roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/casa_imu_odom.bag... logging to /home/elgarbe/.ros/log/1bee9248-3691-11e8-b3c9-7c67a237b388/roslaunch-pc-20888.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pc:40246/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[cartographer_node-1]: started with pid [20905]
process[cartographer_occupancy_grid_node-2]: started with pid [20906]
process[rviz-3]: started with pid [20907]
process[playbag-4]: started with pid [20916]
[ INFO] [1522686574.134628558]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1522686574.134892114]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522686574.134929799]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522686574.134985196]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522686574.135018041]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522686574.135147762]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522686574.135180909]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522686574.135232198]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522686574.135263899]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522686574.135365387]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522686574.135396843]: I0402 13:29:34.000000 20905 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522686574.142904251]: I0402 13:29:34.000000 20905 submap_2d.cc:203] Added submap 1
[ INFO] [1522686574.142994397]: I0402 13:29:34.000000 20905 map_builder_bridge.cc:130] Added trajectory with ID '0'.
0x1d7a4f0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x113dcd0) ): Attempt to set a screen on a child window.
0x1d72980 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x113dcd0) ): Attempt to set a screen on a child window.
0x1d7e4e0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x113dcd0) ): Attempt to set a screen on a child window.
0x1d7db60 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x113dcd0) ): Attempt to set a screen on a child window.
[ INFO] [1522686574.392884864, 1522539150.314767008]: I0402 13:29:34.000000 20905 sensor_bridge.cc:88] Using NavSatFix. Setting ecef_to_local_frame with lat = -33.508, long = -60.0621.
[ INFO] [1522686574.392962356, 1522539150.314767008]: I0402 13:29:34.000000 20905 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636581359500876942'.
F0402 13:29:34.393121 20905 pose_graph_2d.cc:173] Not yet implemented for 2D.
[FATAL] [1522686574.393361148, 1522539150.314767008]: F0402 13:29:34.000000 20905 pose_graph_2d.cc:173] Not yet implemented for 2D.
*** Check failure stack trace: ***
    @     0x7f6fb3e155cd  google::LogMessage::Fail()
    @     0x7f6fb3e17433  google::LogMessage::SendToLog()
    @     0x7f6fb3e1515b  google::LogMessage::Flush()
    @     0x7f6fb3e17e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x66baa0  cartographer::mapping::PoseGraph2D::AddFixedFramePoseData()
    @           0x6d69b0  cartographer::mapping::(anonymous namespace)::GlobalTrajectoryBuilder<>::AddSensorData()
    @           0x6cf7fd  cartographer::sensor::Dispatchable<>::AddToTrajectoryBuilder()
    @           0x6cdeba  cartographer::mapping::CollatedTrajectoryBuilder::HandleCollatedSensorData()
    @           0x6ce8ee  _ZNSt17_Function_handlerIFvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteISB_EEEZNS9_7mapping25CollatedTrajectoryBuilderC4EPNSA_17CollatorInterfaceEiRKSt3setINSG_26TrajectoryBuilderInterface8SensorIdESt4lessISM_ESaISM_EES8_ISL_SC_ISL_EEEUlS7_SE_E_E9_M_invokeERKSt9_Any_dataS7_OSE_
    @           0x5fa88a  _ZNSt17_Function_handlerIFvSt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteIS3_EEEZNS2_8Collator13AddTrajectoryEiRKSt13unordered_setINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4hashISF_ESt8equal_toISF_ESaISF_EERKSt8functionIFvRKSF_S6_EEEUlS6_E_E9_M_invokeERKSt9_Any_dataOS6_
    @           0x5fcf61  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x5fd9c4  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x5fac60  cartographer::sensor::Collator::AddSensorData()
    @           0x6ccd0d  cartographer::mapping::CollatedTrajectoryBuilder::AddData()
    @           0x6cf1b5  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @           0x5b9813  cartographer_ros::SensorBridge::HandleNavSatFixMessage()
    @           0x58f6d9  cartographer_ros::Node::HandleNavSatFixMessage()
    @           0x5a236e  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5ad149  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f6fb176de2d  ros::SubscriptionQueue::call()
    @     0x7f6fb17136f8  ros::CallbackQueue::callOneCB()
    @     0x7f6fb17150fb  ros::CallbackQueue::callAvailable()
    @     0x7f6fb1771ef9  ros::SingleThreadedSpinner::spin()
    @     0x7f6fb1756edb  ros::spin()
    @           0x589b74  cartographer_ros::(anonymous namespace)::Run()
    @           0x586dd4  main
    @     0x7f6fb0162830  __libc_start_main
    @           0x589699  _start
    @              (nil)  (unknown)
[cartographer_node-1] process has died [pid 20905, exit code -6, cmd /home/elgarbe/catkin_ws_cart/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua imu:=/mavros/imu/data fix:=/mavros/global_position/raw/fix __name:=cartographer_node __log:=/home/elgarbe/.ros/log/1bee9248-3691-11e8-b3c9-7c67a237b388/cartographer_node-1.log].
log file: /home/elgarbe/.ros/log/1bee9248-3691-11e8-b3c9-7c67a237b388/cartographer_node-1*.log

```

ojura, 2018-04-02:
`Not yet implemented for 2D.` =)

closed 2018-04-20 by elgarbe

albertchang1989, 2018-11-26:
> `Not yet implemented for 2D.` =)
@ojura 
Has it been implemented for 3D？