cartographer-project/cartographer_ros #801: how to use imu in 2d cartographer?

390388, 2018-04-03:
when I use cartographer without imu,everything is OK. But when I add  imu ,it do not work .I can only get imu odom scan,but no map.This is my bag
[2018-03-28-10-13-18.bag.tar.gz](https://github.com/googlecartographer/cartographer_ros/files/1870472/2018-03-28-10-13-18.bag.tar.gz)

xcy@xcy-laptop:~$ cartographer_rosbag_validate -bag_filename /home/xcy/2018-03-28-10-13-18.bag
E0403 10:51:40.965397  4894 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)
E0403 10:51:40.966027  4894 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan2 has serialization time 1522203199.667674920 but sensor time 1522201906.152137571 differing by -1293.52 s.
E0403 10:51:40.966107  4894 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan2 has serialization time 1522203199.667752026 but sensor time 1522201906.219218460 differing by -1293.45 s.
E0403 10:51:40.966152  4894 rosbag_validate_main.cc:342] frame_id "odom" on topic /odom has serialization time 1522203199.668143839 but sensor time 1522201906.276377838 differing by -1293.39 s.
W0403 10:51:40.966935  4894 rosbag_validate_main.cc:103] frame_id imu_link time 1522201907193958167: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0
W0403 10:51:40.967041  4894 rosbag_validate_main.cc:103] frame_id imu_link time 1522201907253826571: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0
W0403 10:51:40.967131  4894 rosbag_validate_main.cc:103] frame_id imu_link time 1522201907311168010: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0
W0403 10:51:40.983608  4894 rosbag_validate_main.cc:161] Sensor with frame_id "laser" measurements overlap in time Previous range message, ending at time stamp 1522201931.019193514, must finish before current range message, which ranges from 1522201930.986657571 to 1522201931.019987452
W0403 10:51:41.119607  4894 rosbag_validate_main.cc:198] Sensor with frame_id "laser" range measurements have longest overlap of 0.0325359 s
E0403 10:51:41.119637  4894 rosbag_validate_main.cc:357] Average IMU linear acceleration is 12.4045 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0403 10:51:41.119690  4894 rosbag_validate_main.cc:381] IMU data (frame_id: "imu_link") has a large gap, largest is 0.0868308 s, recommended is [0.0005, 0.005] s with no jitter.
I0403 10:51:41.119928  4894 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/IMU_data" (frame_id: "imu_link"):
Count: 3752  Min: 0.032650  Max: 0.086831  Mean: 0.059999
[0.032650, 0.038068)	                    	Count: 6 (0.159915%)	Total: 6 (0.159915%)
[0.038068, 0.043486)	                    	Count: 11 (0.293177%)	Total: 17 (0.453092%)
[0.043486, 0.048904)	                    	Count: 23 (0.613006%)	Total: 40 (1.066098%)
[0.048904, 0.054322)	                   #	Count: 118 (3.144989%)	Total: 158 (4.211087%)
[0.054322, 0.059740)	                 ###	Count: 638 (17.004265%)	Total: 796 (21.215351%)
[0.059740, 0.065158)	     ###############	Count: 2776 (73.987206%)	Total: 3572 (95.202560%)
[0.065158, 0.070577)	                   #	Count: 130 (3.464819%)	Total: 3702 (98.667374%)
[0.070577, 0.075995)	                    	Count: 33 (0.879531%)	Total: 3735 (99.546906%)
[0.075995, 0.081413)	                    	Count: 13 (0.346482%)	Total: 3748 (99.893387%)
[0.081413, 0.086831]	                    	Count: 4 (0.106610%)	Total: 3752 (100.000000%)
E0403 10:51:41.119973  4894 rosbag_validate_main.cc:374] Point data (frame_id: "laser") has a large gap, largest is 0.133058 s, recommended is [0.0005, 0.05] s with no jitter.
I0403 10:51:41.120187  4894 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/scan2" (frame_id: "laser"):
Count: 4608  Min: 0.000794  Max: 0.133058  Mean: 0.066668
[0.000794, 0.014020)	                    	Count: 1 (0.021701%)	Total: 1 (0.021701%)
[0.014020, 0.027247)	                    	Count: 0 (0.000000%)	Total: 1 (0.021701%)
[0.027247, 0.040473)	                    	Count: 0 (0.000000%)	Total: 1 (0.021701%)
[0.040473, 0.053700)	                    	Count: 0 (0.000000%)	Total: 1 (0.021701%)
[0.053700, 0.066926)	  ##################	Count: 4109 (89.171005%)	Total: 4110 (89.192711%)
[0.066926, 0.080152)	                  ##	Count: 497 (10.785590%)	Total: 4607 (99.978302%)
[0.080152, 0.093379)	                    	Count: 0 (0.000000%)	Total: 4607 (99.978302%)
[0.093379, 0.106605)	                    	Count: 0 (0.000000%)	Total: 4607 (99.978302%)
[0.106605, 0.119832)	                    	Count: 0 (0.000000%)	Total: 4607 (99.978302%)
[0.119832, 0.133058]	                    	Count: 1 (0.021701%)	Total: 4608 (100.000000%)
I0403 10:51:41.120407  4894 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
Count: 4607  Min: 0.028901  Max: 0.106415  Mean: 0.066667
[0.028901, 0.036652)	                    	Count: 3 (0.065118%)	Total: 3 (0.065118%)
[0.036652, 0.044403)	                    	Count: 5 (0.108530%)	Total: 8 (0.173649%)
[0.044403, 0.052155)	                    	Count: 18 (0.390710%)	Total: 26 (0.564359%)
[0.052155, 0.059906)	                    	Count: 79 (1.714782%)	Total: 105 (2.279140%)
[0.059906, 0.067658)	    ################	Count: 3763 (81.680054%)	Total: 3868 (83.959190%)
[0.067658, 0.075409)	                 ###	Count: 662 (14.369438%)	Total: 4530 (98.328629%)
[0.075409, 0.083161)	                    	Count: 58 (1.258954%)	Total: 4588 (99.587585%)
[0.083161, 0.090912)	                    	Count: 14 (0.303885%)	Total: 4602 (99.891472%)
[0.090912, 0.098663)	                    	Count: 2 (0.043412%)	Total: 4604 (99.934883%)
[0.098663, 0.106415]	                    	Count: 3 (0.065118%)	Total: 4607 (100.000000%)

gaschler, 2018-04-03:
> E0403 10:51:41.119637 4894 rosbag_validate_main.cc:357] Average IMU linear acceleration is 12.4045 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.

Your input IMU acceleration data seems wrong. Over a longer dataset, it should really average on the gravity acceleration.

Please inspect your IMU data with `rqt_bag`.

> W0403 10:51:40.967131 4894 rosbag_validate_main.cc:103] frame_id imu_link time 1522201907311168010: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0

There are multiple incorrect IMU messages (the log stops after 3 such messages).

> E0403 10:51:40.966027 4894 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan2 has serialization time 1522203199.667674920 but sensor time 1522201906.152137571 differing by -1293.52 s.

Expect the (extrapolated) output of cartographer_ros to be wrong. (Mapping itself only looks at the header=sensor time.)

390388, 2018-04-03:
Thanks,I will modify it later. But I still want to know why when I made the map ,it told me no map received. And is it caused by the errors you found?  Here are some files I changed and the results I got.

[backpack_2d.lua.tar.gz](https://github.com/googlecartographer/cartographer_ros/files/1870655/backpack_2d.lua.tar.gz)
[backpack_2d.launch.tar.gz](https://github.com/googlecartographer/cartographer_ros/files/1870656/backpack_2d.launch.tar.gz)
[trajectory_builder_2d.lua.tar.gz](https://github.com/googlecartographer/cartographer_ros/files/1870658/trajectory_builder_2d.lua.tar.gz)
[image]
[image]
[frames.pdf](https://github.com/googlecartographer/cartographer_ros/files/1870659/frames.pdf)

gaschler, 2018-04-03:
> https://github.com/googlecartographer/cartographer_ros/files/1870659/frames.pdf

It looks like your `base_link` to `laser` transform is never published. If it is constant, just provide it in urdf and drop the whole robot state publisher.

Generally, for terminal output, please paste the text.
To share your configuration, please prepare a fork so we can see the diff of all config files at once as described in https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/.github/ISSUE_TEMPLATE
Thanks.

390388, 2018-04-03:
Because of the problem of Internet, I can't open the link.And I give the wrong frame.This is the frame of my bag.
[frames.pdf](https://github.com/googlecartographer/cartographer_ros/files/1871260/frames.pdf)

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87