cartographer-project/cartographer_ros #809: z position drift during front end slam

slamer-run, 2018-04-05:
With 3d cartographer with single Lidar and IMU, z position have negligible drift at front end slam.  Even if the robot doesn't move at all, the z position drift more than 1 m in a few minutes. 

- launch: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_3d.launch

- lua: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua
The only one parameter is changed as following. 
before: num_point_clouds = 2
after: num_point_clouds = 1

- rosbag & urdf: Sorry, but I can't upload rosbag for some reasons. The tested rosbag is validated, though   

Should I add barometric sensor in order to suppress z position drift at front end slam? Or, Is it necessary to have double lidar to use 3d cartographer slam?

Thanks in advance.

gaschler, 2018-04-05:
What kind of area are you mapping? What angle is your lidar mounted, what type is it, does it see 360 degrees? Does it see the ground?

gaschler, 2018-04-05:
Since the general setup sounds fine, unfortunately, this needs real tuning on the data.
Please tune local SLAM only first (disabling optimization as in the tuning guide).
Perhaps your IMU/Lidar transform is not calibrated not set precise enough?

Does 2D SLAM work fine for you? Indoor 2D is usually easy to configure.

slamer-run, 2018-04-05:
>Does 2D SLAM work fine for you? Indoor 2D is usually easy to configure.
2D SLAM work fine. Your SLAM system is amazing!

>Please tune local SLAM only first (disabling optimization as in the tuning guide).
Referring to below link, I tuned many parameters like TRAJECTORY_BUILDER_3D.ceres_scan_matcher.translation_weight, rotation_weight and so on... But, I can't get good estimated position result. 
http://google-cartographer-ros.readthedocs.io/en/latest/tuning.html

By the way, why did you ask me? "Does it see the ground?"
You meant if Lidar can see the ground, can it result in constraint for z position, pitch and roll estimation? 
What is more, 3d cartographer slam can consider ground plane as important factor?

ojura, 2018-04-05:
From my experience, when doing e.g. 3D SLAM on a vehicle, highly accurate IMU (especially pitch) calibration _really_ matters. It's also simple to conceptually visualize: imagine you're going up (or down) on a very long (e.g. one or two kilometers) slope/incline which with a constant tilt angle. LIDAR is always going to see the ground of the slope, so you won't drift away from the slope in your maps. However, observations from the LIDAR do not make the tilt of the whole slope observable - you need the IMU and the gravity reference for that. Basically, the tilt of all submaps built when driving on this slope will depend on the pitch your IMU says you've had during your whole trajectory. 

So, if you were actually going straight but have a miscalibrated IMU, Cartographer could end up thinking you've been climbing or descending the whole time.

gaschler, 2018-04-06:
> You meant if Lidar can see the ground, can it result in constraint for z position, pitch and roll estimation?
> What is more, 3d cartographer slam can consider ground plane as important factor? 

No, it treats "ground" like any other point cloud. I just wanted to rule out the case that you only see walls because that would be an obvious reason for drift.

> the tilt of all submaps built when driving on this slope will depend on the pitch your IMU 

Thanks @ojura for this excellent explanation. I also believe this is the reason.

slamer-run, 2018-04-08:
@gaschler @ojura 
thank you for your suggestion. 
I got acceleration sensor especially pitch angle accuracy is very important. I'm going to calibrate acceleration sensor and check if it is improved.

closed 2018-04-08 by slamer-run

gaschler, 2018-04-09:
@slamer-run 
If your IMU calibration is good enough for mapping to succeed and you enable `log_solver_summary`, final optimization will print a log message "IMU correction was: ..." that outputs *optimized IMU calibration* that you can use.

See https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/3d/pose_graph/optimization_problem_3d.cc#L549

Cristian-wp, 2021-12-13:
@slamer-run I have the same problem, I also use a VLP-16 as 3D lidar and a Vectornav vn-100 as IMU.
Sometimes the pose is perfect, but other times the drift is without sense...
Can you give me some advice?

@gaschler can you please explain or maybe give me a link to the IMU-LIDAR calibration part? In my case are "alligned", both 3 axis are in the same direcirection