cartographer-project/cartographer_ros #810: Push broom 2D LIDAR

davidromanos, 2018-04-06:
Hi, one very short issue. 

A bit of background here:
I am trying to use cartographer outdoors. My system is a golfcart with a LIDAR placed at a knee high approximately. The LIDAR is a Sick LMS151 with operating range: 0.5 m - 50 m (Max. range with 10 % reflectivity: 18 m). I am trying to map parts of my university campus, i.e. streets, parking lots and others. To do so I use IMU measurements as well, but the results don't look good.

The actual issue:
To get more information of the environment it seemed a good idea to use a tilted LIDAR (push broom configuration) placed on the roof of the golfcart. I've seen that the 2D public data supplied in http://google-cartographer-ros.readthedocs.io/en/latest/data.html includes push broom LIDAR data but I have some doubts with respect to cartographer and this configuration:

- Can I use cartographer with a push broom configuration LIDAR (obviously with the correct tf)? 

- In such case, should I use `MAP_BUILDER.use_trajectory_builder_2d = true` or `MAP_BUILDER.use_trajectory_builder_3d = true`?

- Can I use both range data from the horizontal LIDAR and the push broom LIDAR?

Thanks in advance for your help! 

p.s.: I managed to get good results with only horizontal LIDAR indoors but not with the push broom LIDAR. Find [here](https://www.dropbox.com/s/nkyvy4gelz13wzx/systemoverview.pdf?dl=0) a picture of the golfcart with the aforementioned configuration.

gaschler, 2018-04-06:
> Can I use cartographer with a push broom configuration LIDAR (obviously with the correct tf)?

In principle, yes.
However, for good quality, you need to scan all directions and see more features than just a hilly ground.

> In such case, should I use MAP_BUILDER.use_trajectory_builder_2d = true or MAP_BUILDER.use_trajectory_builder_3d = true?

If you have 2D lidars, you can only use the first config.

> Can I use both range data from the horizontal LIDAR and the push broom LIDAR?

Sure, you can have multiple lidars.

davidromanos, 2018-04-06:
Thanks for the quick reply @gaschler ! What do you mean by "scan all directions"? The push broom LIDAR is a Sick TIM551 with operating range: 0.05 m to 10 m (Max. range with 10 % reflectivity: 8 m ) and the aperture angle is limited between 0° and 180°. Furthermore, both LIDARs are not multi echo lasers.

You say in principle I can use cartographer with a push broom LIDAR but, do you think I can get good quality with this configuration and considering I want to map regular urban outdoor environments? 

I really appreciate your help thanks!!