cartographer-project/cartographer_ros #815: Why can't I get correct 3D map with VLP16 data

sysuzyc, 2018-04-08:
I want to use the cartographer for 3D data with velodyne 16. But I have some  problems encountered in the study. This is the result:
[image]
But there is **no trajectory** in this map, and the **frame is not changed** during the bag play time.
This is the [backpack_3d.urdf](https://github.com/sysuzyc/File/blob/master/backpack_3d.urdf).The launch files are [backpack_3d.launch](https://github.com/sysuzyc/File/blob/master/backpack_3d.launch) and [demo_backpack_3d.launch](https://github.com/sysuzyc/File/blob/master/demo_backpack_3d.launch).
How could I get the correct map? Thanks for any help

gaschler, 2018-04-09:
Please specify messages from the log.
The easiest explanation would be that the Collator just waits for a sensor but does not receive data from that sensor.

Also, please give more details following the issue template https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/.github/ISSUE_TEMPLATE
specifically, the output of the bag file validate tool.

sysuzyc, 2018-04-09:
Thanks for your help!
###  1、when I run rosbag_validate, the result is:
[image] 

###  2、The file required can be find [here](https://gist.github.com/sysuzyc/b2306c90fc726063acc763ca59c116a3).
There are three files, backpack_3d.urdf, backpack_3d.launch, demo_backpack_3d.launch

###  3、The bag file can be find [here](https://drive.google.com/open?id=1y7-7qej0asf7kHWcntIOEOBZMP-qFkEg)

gaschler, 2018-04-09:
The rosbag_validate results do not point out any serious issues with the bag.

Please paste the (interesting) log messages that the node writes into the terminal.
I suspect there is some "waiting for sensor" in there.

Which lua config are you using? The default backpack_3d.lua expects two laser scanners, which would explain why it doesn't map but waits for the second input topic.

sysuzyc, 2018-04-09:
Hi, I use one velodyne 16 lidar and imu sensor to get data. So I changed something in backpack_3d.urdf, backpack_3d.launch, demo_backpack_3d.launch which can be found [**here**](https://gist.github.com/sysuzyc/b2306c90fc726063acc763ca59c116a3). 
And the lua file and log file as shown below.
### 1、backpack_3d.lua
 The lua file can be found  [**here**](https://gist.github.com/sysuzyc/ccb3a7c248eed2aeb97bf63cd4f0d75f). 
### 2、information in terminal
The log information in the terminal is:
[image]
[image]
[image]
### 3、log file
The log file(playbag-6*.log) which is shown in the terminal can be found [**here**](https://github.com/sysuzyc/File/blob/master/playbag-6-stdout.log)

sysuzyc, 2018-04-10:
Hi, I change some parameter in the pose_graph.lua and trajectory_builder_2d.lua and get the **correct map** 
[image]
**But how can I get the correct map with 3D datasets above.**

gaschler, 2018-04-10:
Specify a correct IMU to laser scanner transform in the urdf config.

sysuzyc, 2018-04-10:
The sensor's frame_id is **map** and **velodyne**:
[image]
I have changed the frame_id in backpack_3d.urdf:
[image]
The whole urdf file can be find [**here**](https://gist.github.com/sysuzyc/b2306c90fc726063acc763ca59c116a3#file-backpack_3d-urdf)

derekhe89, 2018-04-11:
hello, i use the velodyne too, but i don't know how to get the 2d data of velodyne, can you tell me? Thank you for you help.@sysuzyc

sysuzyc, 2018-04-12:
The velodyne lidar just get 3D data from the environment. So you should use [**pointcloud_to_laserscan**](https://github.com/ros-perception/pointcloud_to_laserscan) change 3D data to 2D data. Then you can get the 2D data from velodyne lidar. @derekhe89

mantangL, 2018-04-12:
hello,can you get a correct map with 3D dataset? I have same problems with you.

sysuzyc, 2018-04-16:
**Well, I have got the correct map. Thanks for the help received.**

closed 2018-04-16 by sysuzyc

Duke-Allen, 2019-02-28:
@sysuzyc 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?

Duke-Allen, 2019-03-16:
@sysuzyc 
Hi, I have been running 3D cartographer successfully with no error by a 16-lines LIDAR and a MPU6050. But the track is inconsistent with the real movement, like this:
[image]
The result of bag is:
- I0316 12:20:33.418805 11044 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 18056  Min: 0.004353  Max: 0.013893  Mean: 0.010000
[0.004353, 0.005307)	                    	Count: 1 (0.005538%)	Total: 1 (0.005538%)
[0.005307, 0.006261)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.006261, 0.007215)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.007215, 0.008169)	                    	Count: 281 (1.556269%)	Total: 282 (1.561808%)
[0.008169, 0.009123)	                    	Count: 73 (0.404298%)	Total: 355 (1.966105%)
[0.009123, 0.010077)	 ###################	Count: 17197 (95.242577%)	Total: 17552 (97.208687%)
[0.010077, 0.011031)	                    	Count: 151 (0.836287%)	Total: 17703 (98.044968%)
[0.011031, 0.011985)	                    	Count: 266 (1.473194%)	Total: 17969 (99.518166%)
[0.011985, 0.012939)	                    	Count: 86 (0.476296%)	Total: 18055 (99.994461%)
[0.012939, 0.013893]	                    	Count: 1 (0.005538%)	Total: 18056 (100.000000%)
E0316 12:20:33.421423 11044 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101069 s, recommended is [0.0005, 0.05] s with no jitter.
I0316 12:20:33.421811 11044 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1791  Min: 0.100548  Max: 0.101069  Mean: 0.100801
[0.100548, 0.100600)	                    	Count: 10 (0.558347%)	Total: 10 (0.558347%)
[0.100600, 0.100652)	                    	Count: 7 (0.390843%)	Total: 17 (0.949190%)
[0.100652, 0.100704)	                    	Count: 10 (0.558347%)	Total: 27 (1.507538%)
[0.100704, 0.100756)	                   #	Count: 51 (2.847571%)	Total: 78 (4.355109%)
[0.100756, 0.100809)	       #############	Count: 1190 (66.443329%)	Total: 1268 (70.798439%)
[0.100809, 0.100861)	               #####	Count: 478 (26.689001%)	Total: 1746 (97.487434%)
[0.100861, 0.100913)	                    	Count: 21 (1.172529%)	Total: 1767 (98.659966%)
[0.100913, 0.100965)	                    	Count: 10 (0.558347%)	Total: 1777 (99.218315%)
[0.100965, 0.101017)	                    	Count: 9 (0.502513%)	Total: 1786 (99.720825%)
[0.101017, 0.101069]	                    	Count: 5 (0.279174%)	Total: 1791 (100.000000%)

Here is my [bag](https://share.babel.cc/share.do?s=4553105679638233). 
The account: 15838868117
Password: 123456789
Any suggestions would be highly appreciated!

wzw19940427, 2019-08-09:
> Hi, I change some parameter in the pose_graph.lua and trajectory_builder_2d.lua and get the **correct map**
> [image]
> **But how can I get the correct map with 3D datasets above.**

hi,sorry for disturbing.Can you share some message about how to change the parameter in the  pose_graph.lua and trajectory_builder_2d.lua? I meet the same problem.Thanks a lot for that!