cartographer-project/cartographer_ros #816: Best tuning for VLP16 indoor data

villajuvius, 2018-04-09:
Hello, I'm trying to tune Cartographer for my single Velodyne VLP16 but result is not precise enough (weird "shaking" effect & wide wall "thickness"). What are the parameters to tune in order to have good result in a rich indoor environment ?

Configuration files and launch file can be found [here](https://github.com/villajuvius/test_tuning) and bag file [here](https://drive.google.com/file/d/1uWsRaa4SQAppDWucTRdnxgJ0Szc1yqH2/view).

[image]

rosbag validate result:

```
W0409 03:12:53.558775 23173 rosbag_validate_main.cc:344] frame_id "imu_link" on topic /imu/data has serialization time 1522069582.088021415 but sensor time 1522069581.983204338 differing by -0.104817 s.
I0409 03:13:37.202939 23173 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/imu/data" (frame_id: "imu_link"):
Count: 120348  Min: 0.000011  Max: 0.020604  Mean: 0.004001
[0.000011, 0.002070)	             #######	Count: 41676 (34.629574%)	Total: 41676 (34.629574%)
[0.002070, 0.004130)	                  ##	Count: 11014 (9.151793%)	Total: 52690 (43.781368%)
[0.004130, 0.006189)	           #########	Count: 54448 (45.242130%)	Total: 107138 (89.023499%)
[0.006189, 0.008248)	                  ##	Count: 9609 (7.984345%)	Total: 116747 (97.007843%)
[0.008248, 0.010307)	                    	Count: 1426 (1.184897%)	Total: 118173 (98.192741%)
[0.010307, 0.012367)	                    	Count: 1827 (1.518098%)	Total: 120000 (99.710838%)
[0.012367, 0.014426)	                    	Count: 292 (0.242630%)	Total: 120292 (99.953468%)
[0.014426, 0.016485)	                    	Count: 49 (0.040715%)	Total: 120341 (99.994186%)
[0.016485, 0.018544)	                    	Count: 6 (0.004986%)	Total: 120347 (99.999168%)
[0.018544, 0.020604]	                    	Count: 1 (0.000831%)	Total: 120348 (100.000000%)
I0409 03:13:37.210742 23173 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 362207  Min: 0.000005  Max: 0.025250  Mean: 0.001329
[0.000005, 0.002530)	####################	Count: 356097 (98.313118%)	Total: 356097 (98.313118%)
[0.002530, 0.005054)	                    	Count: 6003 (1.657340%)	Total: 362100 (99.970459%)
[0.005054, 0.007579)	                    	Count: 97 (0.026780%)	Total: 362197 (99.997238%)
[0.007579, 0.010103)	                    	Count: 4 (0.001104%)	Total: 362201 (99.998344%)
[0.010103, 0.012627)	                    	Count: 1 (0.000276%)	Total: 362202 (99.998619%)
[0.012627, 0.015152)	                    	Count: 2 (0.000552%)	Total: 362204 (99.999168%)
[0.015152, 0.017677)	                    	Count: 1 (0.000276%)	Total: 362205 (99.999451%)
[0.017677, 0.020201)	                    	Count: 0 (0.000000%)	Total: 362205 (99.999451%)
[0.020201, 0.022726)	                    	Count: 1 (0.000276%)	Total: 362206 (99.999725%)
[0.022726, 0.025250]	                    	Count: 1 (0.000276%)	Total: 362207 (100.000000%)
```

gaschler, 2018-04-09:
Please check if you are really accumulating one laser scanner rotation in one LocalSlamResult.
In rviz, look at the PointCloud2 topic that shows the accumulated points.
This should neither flicker nor show more or less points.

Then, tune `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 80`

Entropy512, 2018-04-25:
"In rviz, look at the PointCloud2 topic that shows the accumulated points." - Which topic is this?  I only see my input point clouds and /scan_matched_points2?

I'm not sure if I can provide my bag file and configuration at the moment (corporate rules on putting piles of data out there...), but some pointers as to where to start looking would be helpful.

The Toyota HSR in 3D mode appears to be using a VLP-16 based on the subscribed topic in https://github.com/googlecartographer/cartographer_toyota_hsr/blob/master/cartographer_toyota_hsr/launch/hsr_3d.launch - but they are not setting num_accumulated_range_data at all in https://github.com/googlecartographer/cartographer_toyota_hsr/blob/master/cartographer_toyota_hsr/configuration_files/toyota_hsr_3d.lua

Meanwhile for two VLP-16 units, https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua is setting a value of 160 - I can see the number of accumulated range data doubling for two units, and I can see why this value might change based on the configured Velodyne RPM setting, but a 160x range???

In the case of backpack_3d, what was the Velodyne configuration for motor RPM?  I'm assuming at lower RPM there are more packets per revolution and then num_accumulated_range_data must increase.

Entropy512, 2018-04-25:
OK, it appears there's a not-very-well-documented "npackets" option for the Velodyne driver.

If you are running dual returns and leaving npackets at default, then you want num_accumulated_range_data equal to 2*num_scanners

If you're not running dual returns, num_accumulated_range_data = num_scanners
(Am I making a mistake here, should this just be per scanner?)

If you set npackets=1 in the Velodyne node config when recording your data and are running 600 RPM, then num_accumulated_range_data should be approximately 76 for single return and 151 for dual return.  (Multiplied by num_scanners?  Not sure...  Need to do more investigation/testing)

Duke-Allen, 2019-03-03:
@villajuvius 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?

villajuvius, 2019-03-03:
@Duke-Allen Hello Duke, you should probably start your own thread to get an answer related to your current issue. Since my problem was solved, I can give you a few hints to start with.

In your case, I think your IMU is not well calibrated prior to your lidar sensor. That why you should provide a correct URDF file **before tunning anything else**. Use CAD data of your system to do so, or use an open source calibration software like [Ethzasl Kalibr](https://github.com/ethz-asl/kalibr.git), VINS Mono calibration capability or even Cartographer itself since it can fix minor rotation issue (set POSE_GRAPH.optimization_problem.log_solver_summary = true).

Then, when you feel confident enough with your URDF you should tune local SLAM alone (set POSE_GRAPH.optimize_every_n_nodes = 0). More detailed instructions [here](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html).

Hope it would help you, please consider opening your own thread next time.

closed 2019-03-03 by villajuvius

Duke-Allen, 2019-03-03:
@villajuvius 
Thanks for your reply. I have started my thread to get an answer. I will tune it as you said. Thank you for your help again.