cartographer-project/cartographer_ros #820: Use cartographer trusting odometry and tf message.

fnobis, 2018-04-10:
I am using the cartographer with the robot with a very reliable source of odometry information. The same information is published as an odometry message and an tf between "odom" and "base_link" frames. Both topics are then enabled in the cartographer configuration.

Now I am trying to configure the cartographer to heavily rely on this sort of information to build the map. Unfortunately the produced maps and pose estimations still differ greatly from my "ground truth" for the pose. 
Is there a standard configuration for such a case, where I want the cartographer basically to work as an "mapping" algorithm without tuning much of the "localization" part via the scan matcher?
Basically take the poses from my odometry or tf and map the lidar data accordingly to those positions?
Thank  you!

gaschler, 2018-04-11:
If you want to increase the weight of odometry in global slam, you are lucky this was just implemented yesterday in https://github.com/googlecartographer/cartographer/pull/1029.
Please take a look at the bottom of the tuning guide, new documentation should just be merged.

In local slam, the position difference of odometry is part of the cost function for scan matching.
The odometry pose difference is computed here when input is available (and then replaces the scan matching pose difference as a velocity estimation):
https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/pose_extrapolator.cc#L240
Its translation part enters the local slam scan matching problem here:
https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/2d/scan_matching/ceres_scan_matcher_2d.cc#L80 (as `target_translation`)

> The same information is published as an odometry message and a tf between "odom" and "base_link" frames

No, this is not the same as the odometry input. odom to base_link is the local SLAM result of Cartographer.

For more help, we would need to understand the actual setup, config and results.

fnobis, 2018-04-11:
@gaschler Thanks for your input, I am going to check out the new version and see if I can improve the results. By tuning the parameters to minimize the influence of the scan matcher.

@MichaelGrupp  yes! That is the way I would like to try. Do you have any hints on where to start for this "custom hack"?

MichaelGrupp, 2018-04-11:
Regarding tuning the weights: 

Please keep in mind that this does not affect the scan insertion. Scans are inserted into the submaps only once in *local* SLAM via scan matching. Setting the new weights that @gaschler mentioned to a higher value means that consecutive trajectory poses are not allowed to deviate too much from the odometry when doing *global* optimization. It's a regularizer for optimization, nothing more. The submaps are not affected by this though.

You can study the local trajectory builder to get familiar with th local SLAM pipeline: https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/2d/local_trajectory_builder_2d.h  (or 3d if you want to use 3d)

fnobis, 2018-04-12:
Directly when mapping starts the odom frame is rotated against the static map frame by a small angle. The mapping result would already be much better if this offset would not occur. Unfortunately I was not able to detect a parameter which influences this. I would appreciat if you can give me some more insight. I upload my bag and workspace.

The results of the bag check can be found here:
https://gist.github.com/fnobis/2e0434c405038d4f29a398da344c72d7
My cartographer configuration:
https://github.com/fnobis/carto_upload
bagfile:
https://gigamove.rz.rwth-aachen.de/d/id/59p78ChqSJ7xnH?6&id=59p78ChqSJ7xnH

MichaelGrupp, 2018-04-13:
I personally don't have much time to look at your data, but two remarks:

* generally it's normal that the odom frame is not aligned with the map frame all the time (http://www.ros.org/reps/rep-0105.html#odom)
* you set ` publish_frame_projected_to_2d = true`, but your odometry data is 3-dimensional

fnobis, 2018-04-13:
@MichaelGrupp  I am aware of the drift. What is odd is that the "drift" occurs on start up. Its seems to mess up the following mapping.
I will have a look at your second remark concerning the 3D movement.  Thanks!

ojura, 2018-04-13:
> generally it's normal that the odom frame is not aligned with the map frame all the time (http://www.ros.org/reps/rep-0105.html#odom)

Not just normal, but expected and wanted. The difference between the odom frame and the map frame is the difference (drift) between the odometry pose and the SLAM localization pose. If that difference would be zero, that would mean that the odometry is perfect and we would not be doing localization. I.e. we would not have SLAM, but rather mapping with known poses.

fnobis, 2018-04-15:
@ojura 
I agree, the frame should correct the drift. The behavior I am experiencing however is that the cartographer is creating a map<->odom transform with an initial rotational offset, which is resulting in a faulty alignment of the map. At initial position, there shouldnt be a drift. So frames should still be aligned. 

Mapping with known poses would deliver a more accurate result in this case. At least with the current parameters I am using.

litzar, 2018-09-04:
@fnobis May I ask you a question? I'd like to add the odometer to cartographer. But I don't know the odometer pose is the global pose or the delta pose between current and previous. Could you help me?

RonaldEnsing, 2019-01-30:
I had the same situation. My odometry is very reliable, so I wanted to use only the odometry input for local SLAM and allow global SLAM to optimize using the point clouds for a loop closure. I've experimented with it by making some small modifications [1, 2] as suggested by @MichaelGrupp 

[1] https://github.com/RonaldEnsing/cartographer/blob/90f53eae1e7208537489162a906a02720e698a90/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L90
[1] https://github.com/RonaldEnsing/cartographer/blob/90f53eae1e7208537489162a906a02720e698a90/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L289

litzar, 2019-02-17:
How about the test results？

------------------
Lin LiTzar 
 Cell:(+86)159 9958 6612

RonaldEnsing, 2019-02-22:
These modifications improved the mapping for my recorded rosbag containing accurate odometry information. The loop closure is accurate with this modified code and not accurate (local optimum) without these modifications.

closed 2020-10-21 by MichaelGrupp

Leo6069, 2024-10-07:
> I had the same situation. My odometry is very reliable, so I wanted to use only the odometry input for local SLAM and allow global SLAM to optimize using the point clouds for a loop closure. I've experimented with it by making some small modifications [1, 2] as suggested by @MichaelGrupp
> 
> [1] https://github.com/RonaldEnsing/cartographer/blob/90f53eae1e7208537489162a906a02720e698a90/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L89-L105 [2] https://github.com/RonaldEnsing/cartographer/blob/90f53eae1e7208537489162a906a02720e698a90/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L289-L290

Hi there, based on this modification, do you have any idea about how to disable global SLAM further to stop using the point clouds for a loop closure? Thanks in advance.