cartographer-project/cartographer_ros #823: RViz: marker visualizations disappearing behind submaps

MichaelGrupp, 2018-04-11:
The trajectory and constraint markers are disappearing and re-appearing from time to time when submap visualization is enabled. Here's a video that shows this: [click](https://drive.google.com/open?id=1Ss15b_zyp4zldVYuVFtGq9ZaBQwTlTeL)

The video is a capture of running this:
```bash
roslaunch cartographer_toru offline_toru_2d.launch bag_filenames:=$(pwd)/hallway_return.bag
```
as described here: https://github.com/magazino/cartographer_magazino#documentation

This is a minor issue, but maybe someone who is familiar with the RViz visualization could take a look? @kdaun, you recently changed the trajectory visualization?

kdaun, 2018-04-17:
Thanks for the note.
Currently, the submap and the trajectory visualization are drawn at the same z-coordinate. Therefore, it is undefined which one is shown at top if both visualizations are enabled at the same time.
AFAIK, this is the default behavior for Cartographer for quite some while. 
The recent change in the trajectory visualization https://github.com/googlecartographer/cartographer_ros/pull/756 changes the transparency of parts of the trajectory depending on whether the part is constrained by inter-trajectory, inter-submap or intra-submap constraints.

ojura, 2018-04-17:
Here's a related older issue. https://github.com/googlecartographer/cartographer_ros/issues/426