cartographer-project/cartographer_ros #828: Cartographer 3D with IMU, odometry, and 1 LiDAR

vaibhavviswanathan, 2018-04-16:
Hi there,

I am looking to use Cartographer 3D with an IMU, odometry, and 1 LiDAR. I am using the "demo_backpack_3d.launch" as a reference. However, I notice that it has 2 different LiDARs. How do I set my launch file and configuration files to match my needs?

Best,
Vai

gaschler, 2018-04-17:
Hi,

- launch file: Have only one `remap` tag with your incoming topic name
- backpack_3d.lua: `num_point_clouds = 1, use_odometry = true`
- backpack_3d.urdf: Remove the unneeded joint, specify your IMU to base_link rotation (the default 0% chance to be correct for you). Same for odometry.

Also, always start experimenting with 2D first and definitely without odometry first. See the tuning guide.

vaibhavviswanathan, 2018-04-17:
Great! Thank you.

Another question, I am currently building cartographer in a separate workspace outside of my main ROS workspace. What is the convention for this? Where do I put my launch files?

Best,
Vai

vaibhavviswanathan, 2018-04-17:
Hi,

I have an error when I am setting `use_odometry = true`. If I use the same bagfile without odometry, there are no errors. The error I am getting is:

```
[FATAL] [1523991642.691777084, 1523988701.740793128]: F0417 12:00:42.000000 15592 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /camera should not start with a /. See 1.7 in http://wiki.ros.org/tf2/Migration.
*** Check failure stack trace: ***
    @     0x7f4264a485cd  google::LogMessage::Fail()
    @     0x7f4264a4a433  google::LogMessage::SendToLog()
    @     0x7f4264a4815b  google::LogMessage::Flush()
    @     0x7f4264a4ae1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x5bce89  cartographer_ros::(anonymous namespace)::CheckNoLeadingSlash()
    @           0x5bd3c4  cartographer_ros::SensorBridge::ToOdometryData()
    @           0x5936b0  cartographer_ros::Node::HandleOdometryMessage()
    @           0x5a680e  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5b2199  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f42623a144d  ros::SubscriptionQueue::call()
    @     0x7f4262347210  ros::CallbackQueue::callOneCB()
    @     0x7f4262348683  ros::CallbackQueue::callAvailable()
    @     0x7f42623a5511  ros::SingleThreadedSpinner::spin()
    @     0x7f426238a4cb  ros::spin()
    @           0x58dd24  cartographer_ros::(anonymous namespace)::Run()
    @           0x58af74  main
    @     0x7f4260d97830  __libc_start_main
    @           0x58d849  _start
    @              (nil)  (unknown)
[cartographer_node-3] process has died [pid 15592, exit code -6, cmd /home/igino/bfr_cartographer/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/igino/bfr_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename bfr.lua points2:=velodyne_points imu:=imu/data odom:=integrated_to_init __name:=cartographer_node __log:=/home/igino/.ros/log/9aaab6de-4271-11e8-a711-f44d3068a052/cartographer_node-3.log].
log file: /home/igino/.ros/log/9aaab6de-4271-11e8-a711-f44d3068a052/cartographer_node-3*.log
```

gaschler, 2018-04-18:
As the error message says, frame_id should not start with a `/`, details are in http://wiki.ros.org/tf2/Migration#Removal_of_support_for_tf_prefix.

vaibhavviswanathan, 2018-04-18:
Thank you! I solved this issue.

To continue this conversation, the results of including an odometry topic in Cartographer makes it significantly worse. In fact, the resulting path is even worse than just the odometry frame. 

This is an image of only the odometry topic. As you can see, there is some error on the turns. The result should also be a closed loop.
[image]

Here is a result of the path using Cartographer without odometry. This trajectory seems much more accurate, but the loop closure error is approx 3.5 meters. 
[image]

Here is the result of using Cartographer with the above odometry. The results are clearly much worse. The position estimate drifts a lot even before the robot actually starts moving. What could be the possible errors?
[image]

gaschler, 2018-04-18:
I believe this is expected as long as your input data issues https://github.com/googlecartographer/cartographer_ros/issues/831 are unresolved.

vaibhavviswanathan, 2018-04-18:
I fixed that issue by changing the odometry_sampling_ratio to 0.1 -- I no longer have that problem.

wxt-github, 2018-04-22:
hi,everyone！
I would like to ask if I only have a 3D laser radar, how do I set up my launch file and lua file?
thanks.

gaschler, 2018-04-23:
@wxt-github, please see my previous answer: https://github.com/googlecartographer/cartographer_ros/issues/828#issuecomment-381888864

wxt-github, 2018-04-26:
Thanks for your reply. I'm sorry to reply to you now. I've been trying to modify these files as you said these days, but there are still problems. The more mistakes I make, the more I am modifying them. Do you have a few documents that have been modified to give me a reference? Thank you very much! @gaschler

albertchang1989, 2018-09-17:
@vaibhavviswanathan @gaschler 
Hi, when I am using Cartographer 3D with an IMU and one LiDAR.
Here is my launch file, the topic name of LiDAR data in my bag is "velodyne_points" and "/imu/data" for IMU data

/home/albert/Desktop/lua.png

Here is the udrf file
/home/albert/Desktop/urdf.png

In lua file, I just change num_point_clouds = 1.

The problem I meet now is below
/home/albert/Pictures/scr.png

Is there anything I miss?

Best
Albert

Mafuqiang, 2019-05-29:
can you share the configuration files like urdf file? Because I meet some mistakes when i run cartographer with odometry, but it is work with IMU. Thanks.

bochen87, 2020-09-11:
closing as inactive

closed 2020-09-11 by bochen87