cartographer-project/cartographer_ros #831: map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first

vaibhavviswanathan, 2018-04-18:
Results of rosbag_validate: [https://gist.github.com/vaibhavviswanathan/6d3f3952d8e86033d094678a7cd23b5f](https://gist.github.com/vaibhavviswanathan/6d3f3952d8e86033d094678a7cd23b5f
)

What is causing this error? I am running 3D SLAM with odometry.

gaschler, 2018-04-18:
> E0417 17:53:28.068562  3700 rosbag_validate_main.cc:350] frame_id "imu_link" on topic /imu/data has serialization time 1524007103.861264414 but sensor time 1518636554.987154960 differing by -5.37055e+06 s.

Expect the (extrapolated) output of cartographer_ros to be wrong. (Mapping itself only looks at the header=sensor time.) Easiest fix is to re-record on the machine were the sensors are.

> E0417 17:53:28.068958  3700 rosbag_validate_main.cc:121] frame_id odom time 1518636555022079640: Odometry pose is invalid. pose position: 
>   x: 4.59
>   y: 60.474
>   z: 0.668
> orientation: 
>   x: 4
>   y: 0
>   z: 0
>   w: 0

The orientation is not a valid unit quaternion.

> E0417 17:53:28.069325  3700 rosbag_validate_main.cc:315] Sensor with frame_id "map" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
> E0417 17:53:28.096035  3700 rosbag_validate_main.cc:160] Sensor with frame_id "/camera" is not sequential in time.Previous range message ends at time 636542333570838600, current one at time 636542333570838600

This is not supported as input. Each sensor input must be sequential.

There are really multiple issues, also with timestamps, did you process your initial recording?

vaibhavviswanathan, 2018-04-18:
SOlved issue by changing odometry_sampling_ratio from 1 to 0.1

closed 2018-04-18 by vaibhavviswanathan

cschuet, 2018-04-19:
@vaibhavviswanathan That's not a fix. That's just hiding the issue. You throw away 9 out of 10 IMU messages.

Duke-Allen, 2019-02-22:
@cschuet 
I also meeting this problem while running cartographer with IMU and a RSLIDAR_16. 
[image]
Results of rosbag_validate:
[image]
[my configuration files](https://github.com/Duke-Allen/cartographer-configuration.git)
How can I solve it?