cartographer-project/cartographer_ros #836: Performing 2d SLAM using cartographer_ros with Scanse Sweep lidar

mahendranicl, 2018-04-20:
Hi All,
       I am trying to perform 2d SLAM using cartographer_ros with Scanse sweep lidar (2d). I have installed the sweep_ros and this allowed me to get point cloud from Scanse sweep lidar and view it. Then I installed cartographer_ros and ran the demo bag file, cartographer is working fine too. 

Now I want to perfrom SLAM with cartorgapher_ros  using sweep_ros. The launch file, urdf and lua files used can be in the below github link

https://github.com/mahendranicl/Scanse-sweep_ros-config-urdf-and-launch-files-for-cartographer_ros

However, unable to perform slam and getting error messages. The history can be found in the above github link (filename - error)

error message 

..."]:33: attempt to index global 'SPARSE_POSE_GRAPH' (a nil value)ailed: status == 0 (2 vs. 0) [string "include "map_builder.lua"
..."]:33: attempt to index global 'SPARSE_POSE_GRAPH' (a nil value)ameter_dictionary.cc:83] Check failed: status == 0 (2 vs. 0) [string "include "map_builder.lua"

then  

[cartographer_node-5] process has died [pid 19664, exit code -6, cmd /home/mahendran/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/mahendran/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename sweep.lua __name:=cartographer_node __log:=/home/mahendran/.ros/log/f7bbfdba-449c-11e8-b606-a782b0d5ce85/cartographer_node-5.log].
log file: /home/mahendran/.ros/log/f7bbfdba-449c-11e8-b606-a782b0d5ce85/cartographer_node-5*.log

I kindly request you all to help me resolve this. Maybe there is an issue with the codes I am using or haven't pasted the files in the proper place or maybe I am doing something wrong somewhere. Please help and any help is much appreciated. 

Thanks

gaschler, 2018-04-20:
`SPARSE_POSE_GRAPH` has been renames to `POSE_GRAPH` a few months ago.
Make sure you use the latest versions.

Update your config:
https://github.com/mahendranicl/Scanse-sweep_ros-config-urdf-and-launch-files-for-cartographer_ros/blob/master/sweep.lua#L33

Hope this helps.

mahendranicl, 2018-04-20:
Hi gaschler,
         Ah, ok. Very useful. Done it, but getting the following error now 

error message

F0420 15:03:58.366683 21814 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'scans_per_accumulation' was used the wrong number of times.
[FATAL] [1524233038.366955371]: F0420 15:03:58.000000 21814 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'scans_per_accumulation' was used the wrong number of times.

then 

[cartographer_node-5] process has died [pid 21814, exit code -6, cmd /home/mahendran/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/mahendran/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename sweep.lua __name:=cartographer_node __log:=/home/mahendran/.ros/log/ab34dfb4-44a3-11e8-b59b-54e1ad75622a/cartographer_node-5.log].
log file: /home/mahendran/.ros/log/ab34dfb4-44a3-11e8-b59b-54e1ad75622a/cartographer_node-5*.log

Thanks

cschuet, 2018-04-21:
In November '17 we renamed `scans_per_accumulation` to `num_accumulated_range_data`. I think it is easier in your case to start with a tip-of-tree config file, e.g. `backpack_2d.lua`, and modify that to suit your needs.

mahendranicl, 2018-06-20:
Hello All,
          Changed the config files and 2d slam was working fine. Until recently I noticed my map (RVIZ) spinning out of control and I switched my LIDAR (luckily have 3) and it worked fine. Then again today the map is spinning out of control i.e. with my replacement lidar too ! Any ideas ?

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87