cartographer-project/cartographer_ros #843: Build broken with renamed .h files

Chambana, 2018-04-26:
Recent commits to Master have renamed these files and broken the include statements: 

https://github.com/googlecartographer/cartographer_ros/blob/5ce6e68dd3c97e3d6bb7dab533f7dd9f8d73b0c8/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.h#L24-L25

The official build instructions (http://google-cartographer-ros.readthedocs.io/en/latest/) will checkout Master to build cartographer, so it might be better to update those instructions to checkout the latest stable build instead of master.

kdaun, 2018-04-26:
Merging is in process: https://github.com/googlecartographer/cartographer_ros/pull/838

kdaun, 2018-04-26:
Merging is completed.