cartographer-project/cartographer_ros #845: Errors occurred while installing cartographer_ros

jn-c, 2018-04-26:
Hello! I'm trying to install `cartographer_ros` in ubuntu16.04 and kinetic for my robot project. However, some errors occurred during the installation process, and the following is a section of the error message([complete output](https://gist.github.com/jn-c/5a9f9c6b0abeb97710b96388ea138b1b)):

```
FAILED: /home/dauker/cartographer_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_offline_node 
: && /usr/bin/c++   -O3 -DNDEBUG   cartographer_ros/CMakeFiles/cartographer_offline_node.dir/offline_node_main.cc.o  -o /home/dauker/cartographer_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_offline_node  -rdynamic /home/dauker/cartographer_ws/devel_isolated/cartographer_ros/lib/libcartographer_ros.a /home/dauker/cartographer_ws/install_isolated/lib/libcartographer.a /home/dauker/cartographer_ws/install_isolated/lib/libceres.a -lglog -lgflags -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lglog -lgflags -lcairo /usr/local/lib/libprotobuf.a -lpthread -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_serialization -lboost_chrono -lboost_atomic -lpthread -lpcl_common /opt/ros/kinetic/lib/libeigen_conversions.so /opt/ros/kinetic/lib/liborocos-kdl.so.1.3.0 /opt/ros/kinetic/lib/librosbag.so /opt/ros/kinetic/lib/librosbag_storage.so /opt/ros/kinetic/lib/libroslz4.so -llz4 /opt/ros/kinetic/lib/libtopic_tools.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options -ltinyxml /opt/ros/kinetic/lib/libtf2_ros.so /opt/ros/kinetic/lib/libactionlib.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libtf2.so /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_serialization -lboost_chrono -lboost_atomic -lpthread -lpcl_common /opt/ros/kinetic/lib/libeigen_conversions.so /opt/ros/kinetic/lib/liborocos-kdl.so.1.3.0 /opt/ros/kinetic/lib/librosbag.so /opt/ros/kinetic/lib/librosbag_storage.so /opt/ros/kinetic/lib/libroslz4.so -llz4 /opt/ros/kinetic/lib/libtopic_tools.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options -ltinyxml /opt/ros/kinetic/lib/libtf2_ros.so /opt/ros/kinetic/lib/libactionlib.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libtf2.so /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_iostreams -lboost_regex -lconsole_bridge -Wl,-rpath,/opt/ros/kinetic/lib: && :
/home/dauker/cartographer_ws/devel_isolated/cartographer_ros/lib/libcartographer_ros.a(urdf_reader.cc.o)：在函数‘cartographer_ros::ReadStaticTransformsFromUrdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tf2_ros::Buffer*)’中：
urdf_reader.cc:(.text+0x24a)：对‘urdf::Model::initFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
collect2: error: ld returned 1 exit status
[27/39] Building CXX object cartograph...s/cartographer_node.dir/node_main.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/dauker/cartographer_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/dauker/cartographer_ws/build_isolated/cartographer_ros && /home/dauker/cartographer_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

```

  In order to make a comparison, then I followed the instructions from this [tutorial](https://google-cartographer-ros.readthedocs.io/en/latest/), and some errors([complete output](https://gist.github.com/jn-c/5a9f9c6b0abeb97710b96388ea138b1b)) showed up after executing `catkin_make_isolated --install --use-ninja`:

```
<== Failed to process package 'cartographer_ros': 
  Command '['/home/dauker/cartographer_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/dauker/cartographer_ws/build_isolated/cartographer_ros && /home/dauker/cartographer_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.
```

  I also post the content of [CMakeError.log](https://gist.github.com/jn-c/45de39b101ca6e2085bb01627d73659b) and [CMakeOutput.log](https://gist.github.com/jn-c/f4df80dc1e641d8f6e9116e422b5a375). What should I do to fix this? Any advice and suggestions will be greatly appreciated

gaschler, 2018-05-07:
Thanks for providing enough details.
I'm not sure what the actual issue is, however, I'd recommend to 

* Fix your "pthread not found" issue first that is indicated by https://gist.github.com/jn-c/45de39b101ca6e2085bb01627d73659b, this really should not happen. Ensure that pthread is installed. 
* As always, delete all build/build_isolated/devel/install folders before compiling again.

forlinchow, 2018-07-06:
have you fix the problem?

cschuet, 2018-07-06:
Note that there are now binary packages here, so if you can't get the package to compile this might be an option:

http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-cartographer-ros/

jn-c, 2018-07-08:
@forlinchow Not yet.

forlinchow, 2018-07-09:
I solved my problem last week

jn-c, 2018-07-09:
@forlinchow Can you tell me your method?

forlinchow, 2018-07-10:
https://github.com/googlecartographer/cartographer_ros/issues/924  That's my problem

closed 2018-10-22 by jn-c