cartographer-project/cartographer_ros #846: Cartographer not generating submap although getting sensor data

ekow-mensah, 2018-04-26:
Hi, I am a master's student at the University of Loughborough and I am trying to implement online 2D SLAM using the VLP-16 LiDAR. The cartographer node subscribes to the /scan topic produced by the LiDAR  which is responsible for publishing the sensor data, however, the sub-map does not build. The image below shows my .lua file. I will really appreciate some help as to why the sub-map is not being constructed. Thank you very much 
[image]

ekow-mensah, 2018-04-26:
no reason in particular. Since I am new to SLAM and Cartographer as a whole, I thought it will be easier to implement 2D first and then 3D after successfully implementing 2D.

gaschler, 2018-04-27:
> implement 2D first and then 3D after successfully implementing 2D.

Agree, this is often easier because you don't need IMU extrinsic calibration for 2D.

> why the sub-map is not being constructed

Please post text instead of screenshots or ideally fork cartographer_ros and upload config files so people can see the diff.
The two screenshots do not indicate anything.
Please post at least test_slam.lua, the terminal logs of the node, and explain what data arrives rviz.

ekow-mensah, 2018-04-30:
I am sorry for the late reply I did not have access to the lab I am working in over the weekend. I have created the fork of the google cartographer project. The link is down below: 
https://github.com/ekow-mensah/cartographer_ros

I have also uploaded the test.launch and testslam.lua files. I will post the terminal logs of the node and an explanation of what data arrives very soon.

ekow-mensah, 2018-04-30:
I found a solution to the problem in this issue https://github.com/googlecartographer/cartographer_ros/issues/623 
and I made the changes to the test.launch, test_slam.lua files accordingly. I also made a new file called vlp16.urdf but now I have a new error.  lua_parameter_dictionary.cc:399] Check failed: HasKey(key) Key 'use_nav_sat' not in dictionary. 

More details on this error can be found in the attached log file. 
[cartographer_node.log](https://github.com/googlecartographer/cartographer_ros/files/1960685/cartographer_node.log)

but I don't understand the error i've searched all the configuration files and there is no key called use_nav_sat.

kdaun, 2018-04-30:
> lua_parameter_dictionary.cc:399] Check failed: HasKey(key) Key 'use_nav_sat' not in dictionary.

states that the option "use_nav_sat" is missing in your configuration file. Add `use_nav_sat = false,` to your configuration file. You can use [backpack_2d.lua ](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_2d.lua#L28) as reference.

ekow-mensah, 2018-05-01:
Thank you all for your advice and help it's all working now. I will close this issue now.

closed 2018-05-01 by ekow-mensah

daiencan, 2018-09-11:
@gaschler Hi, I am a master's student at the University of Cqupt and I am trying to implement  3D SLAM using the RSLiDAR-16. The cartographer node subscribes to the /rslidar_points topic , /imu topic and /odom topic. however, the sub-map does not build. There is no error or warning in the terminal. I will really appreciate some help as to why the sub-map is not being constructed. Thank you very much

kingcheng2000, 2019-12-12:
> @gaschler Hi, I am a master's student at the University of Cqupt and I am trying to implement 3D SLAM using the RSLiDAR-16. The cartographer node subscribes to the /rslidar_points topic , /imu topic [...]

Did you get your issue fixed ?