cartographer-project/cartographer_ros #847: Different Submaps resolution with pure localization

AlexReimann, 2018-04-27:
Can there be any problems / drawbacks running with pure localization on a pbstream which was made with a different submap resolution?

gaschler, 2018-04-27:
Interesting question. 
Mixing trajectories with different `TrajectoryOptions` is untested I would say. It may or may not run through. If it doesn't crash, at least scores will be biased.
Please share your results.

> You guys want questions like this here or actually in the cartographer repo?

You are right this is actually a cartographer question. Reason for the link from cartographer's front page to point to cartographer_ros is that 90% of questions are really related to ROS.

AlexReimann, 2018-04-27:
So should I move this over to the cartographer?
I basically got my answer already though.

From my results, it still worked but had some problems with global optimization score being rather low. Was not sure if this is because of the difference submap resolution or something else.

If it is not intended usage I won't try to get it working better as it's not worth investing time in it imho.

Might want to make a warning about this somewhere in the documentation?

gaschler, 2018-04-27:
> So should I move this over to the cartographer?

No worries.

> If it is not intended usage I won't try to get it working better as it's not worth investing time in it imho.

Right. We have plenty of options and many combinations are not really supported.
If you have an interesting use case, feel free to discuss in the open house meeting.
Of course, we do review and accept bug fixes, but are a little careful adding obscure features.

AlexReimann, 2018-04-27:
I might do a PR for this.

The [tuning guide for pure localization](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#pure-localization-in-a-given-map) currently says to use the [tuning guide for low latency](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency), which tells to increase `submaps.resolution`.

So I would at least add a warning in the low latency guide and then maybe a check in the cartographer code itself for the resolution of the loaded trajectory and the current used resolution?

PR #858 (2018-05-25): [docs] Add warning on pure_localization submap resolutions
Closes #847
commit 037ea79e (2018-05-25): [docs] Add warning on pure_localization submap resolutions (#858)

Closes #847
closed 2018-05-25 by wally-the-cartographer