cartographer-project/cartographer_ros #850: build error with google::protobuf::internal

coldrainy, 2018-04-28:
I get an error when building the cartographer_ros,the error log is bellow,can anyone help me?
==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/coldrain/catkin_ws/carto_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/coldrain/catkin_ws/carto_ws/build_isolated/cartographer_ros'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/coldrain/catkin_ws/carto_ws/build_isolated/cartographer_ros'
[8/45] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/local/include -isystem /usr/local/include/pcl-1.8 -isystem /usr/include/eigen3 -isystem /home/coldrain/catkin_ws/carto_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/include/pcl-1.7 -I. -I/home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o -c /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc
In file included from /home/coldrain/catkin_ws/carto_ws/install_isolated/include/cartographer/mapping/2d/proto/map_limits.pb.h:33:0,
                 from /home/coldrain/catkin_ws/carto_ws/install_isolated/include/cartographer/mapping/2d/map_limits.h:26,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.h:25,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc:17:
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid2d::set_allocated_translation(cartographer::transform::proto::Vector2d*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:1900:21: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       translation = ::google::protobuf::internal::GetOwnedMessage(
                     ^
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid2f::set_allocated_translation(cartographer::transform::proto::Vector2f*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:1982:21: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       translation = ::google::protobuf::internal::GetOwnedMessage(
                     ^
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid3d::set_allocated_translation(cartographer::transform::proto::Vector3d*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:2064:21: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       translation = ::google::protobuf::internal::GetOwnedMessage(
                     ^
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid3d::set_allocated_rotation(cartographer::transform::proto::Quaterniond*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:2118:18: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       rotation = ::google::protobuf::internal::GetOwnedMessage(
                  ^
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid3f::set_allocated_translation(cartographer::transform::proto::Vector3f*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:2176:21: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       translation = ::google::protobuf::internal::GetOwnedMessage(
                     ^
/usr/local/include/cartographer/transform/proto/transform.pb.h: In member function ‘void cartographer::transform::proto::Rigid3f::set_allocated_rotation(cartographer::transform::proto::Quaternionf*)’:
/usr/local/include/cartographer/transform/proto/transform.pb.h:2230:18: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       rotation = ::google::protobuf::internal::GetOwnedMessage(
                  ^
In file included from /usr/local/include/cartographer/mapping/proto/trajectory_node_data.pb.h:32:0,
                 from /usr/local/include/cartographer/mapping/trajectory_node.h:25,
                 from /home/coldrain/catkin_ws/carto_ws/install_isolated/include/cartographer/mapping/2d/map_limits.h:28,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.h:25,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc:17:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h: In member function ‘void cartographer::sensor::proto::ImuData::set_allocated_linear_acceleration(cartographer::transform::proto::Vector3d*)’:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h:1004:29: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       linear_acceleration = ::google::protobuf::internal::GetOwnedMessage(
                             ^
/usr/local/include/cartographer/sensor/proto/sensor.pb.h: In member function ‘void cartographer::sensor::proto::ImuData::set_allocated_angular_velocity(cartographer::transform::proto::Vector3d*)’:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h:1054:26: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       angular_velocity = ::google::protobuf::internal::GetOwnedMessage(
                          ^
/usr/local/include/cartographer/sensor/proto/sensor.pb.h: In member function ‘void cartographer::sensor::proto::OdometryData::set_allocated_pose(cartographer::transform::proto::Rigid3d*)’:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h:1132:14: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       pose = ::google::protobuf::internal::GetOwnedMessage(
              ^
/usr/local/include/cartographer/sensor/proto/sensor.pb.h: In member function ‘void cartographer::sensor::proto::FixedFramePoseData::set_allocated_pose(cartographer::transform::proto::Rigid3d*)’:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h:1210:14: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       pose = ::google::protobuf::internal::GetOwnedMessage(
              ^
/usr/local/include/cartographer/sensor/proto/sensor.pb.h: In member function ‘void cartographer::sensor::proto::LandmarkData_Landmark::set_allocated_transform(cartographer::transform::proto::Rigid3d*)’:
/usr/local/include/cartographer/sensor/proto/sensor.pb.h:1327:19: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       transform = ::google::protobuf::internal::GetOwnedMessage(
                   ^
In file included from /usr/local/include/cartographer/mapping/trajectory_node.h:25:0,
                 from /home/coldrain/catkin_ws/carto_ws/install_isolated/include/cartographer/mapping/2d/map_limits.h:28,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.h:25,
                 from /home/coldrain/catkin_ws/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc:17:
/usr/local/include/cartographer/mapping/proto/trajectory_node_data.pb.h: In member function ‘void cartographer::mapping::proto::TrajectoryNodeData::set_allocated_gravity_alignment(cartographer::transform::proto::Quaterniond*)’:
/usr/local/include/cartographer/mapping/proto/trajectory_node_data.pb.h:323:27: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
[... cut ...] lativeScanMatcherOptions*)’:
/usr/local/include/cartographer/mapping/sparse_pose_graph/proto/constraint_builder_options.pb.h:607:50: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
       fast_correlative_scan_matcher_options_3d = ::google::protobuf::internal::GetOwnedMessage(
                                                  ^
/usr/local/include/cartographer/mapping/sparse_pose_graph/proto/constraint_builder_options.pb.h: In member function ‘void cartographer::mapping::

gaschler, 2018-05-07:
> `sparse_pose_graph_options`

This was renamed to pose_graph_options several months ago. Please make sure you are working on the latest versions and not mixing versions of cartographer and cartographer_ros.

coldrainy, 2018-05-08:
@gaschler Thank you for your answer!I delete the whole catkin_ws folder and create a new one,then I follow the cartographer_ros Building & Installation instruction again,then the problem is solved.But i am still confused about the problem.What I did in the first time was that I updated the program by deleting the old source code and downloaded the new code,then I skipped the previous steps and directly do the last step.The result as you know,I got an error.So why would this happen?Why I can't update the program by just deleting the code of old version and replace it by the new one?

closed 2020-10-23 by MichaelGrupp