cartographer-project/cartographer_ros #851: How do we get transformation between base_link and odom frames?

jediofgever, 2018-04-28:
Hi there, 
This is a question rather than an actual issue, sorry for the inconvenience. However I am confused about few things and one might help.

I build map using 2D slam by cartographer, and then I want to navigate my robot in this map, however I cannot localize my robot in this map. I use `amcl` package to localize, which needs transform between odom frame and base_link frame. I see that google cartographer provides transform from odom frame to base_link frame. 

How do you do this, can you show me in the code ?
Thank you

kdaun, 2018-04-30:
> I build map using 2D slam by cartographer, and then I want to navigate my robot in this map

This use-case is well supported by Cartographer. You can use Cartographer in SLAM mode to create  map of an environment (see [demo_backpack_2d.launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_2d.launch)). After creating the map, you can use Cartographer in localization-only mode to localize within the map (see [demo_backpack_2d_localization.launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_2d_localization.launch)). As an alternative to Cartographer in localization-only mode, you can use `amcl` to localize within the map.

>I see that google cartographer provides transform from odom frame to base_link frame.
>How do you do this, can you show me in the code ?

The provided tf Transforms are described in the documentation under [Provided tf Transforms
](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#provided-tf-transforms) and [configuration](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html). 
AFAIK `amcl` expects  the odom to base_link transform to contain integrated odometry observations whereas Cartographer publishes the local, non-loop-closed, continuous pose.
Therefore, using Cartographer to generate odom to base_link transform to feed into `amcl` does not seem reasonable to me. Instead, I would recommend using Cartographer localization 

> however I cannot localize my robot in this map

Please set up your system similar to  [demo_backpack_2d.launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_2d.launch) and  [demo_backpack_2d_localization.launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_2d_localization.launch).
If the result is not satisfactory, follow guidelines for filing an issue:
 1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:
   `rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>`
 2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
 3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.

jediofgever, 2018-05-05:
Dear @kdaun , 
Thank you very much for your detailed and helpful reply. I followed your instrusctions I was able to sucessfully do the localization using 2D slam of Google cartographer. I am closing issue as I have resolved with your help . :)

closed 2018-05-05 by jediofgever