cartographer-project/cartographer_ros #852: Pure Localisation -  Using Cartographer to perform localisation from point cloud data

ekow-mensah, 2018-05-02:
Hi, This is more of a question rather than an issue and I really need some help.  Is it possible to perform localisation only with Google Cartographer while an external program displays a map of the environment? Velodyne has a ROS package which displays a 3D point cloud of the environment. Could I use that as the map and then use cartographer_ros to perform localisation?

gaschler, 2018-05-02:
As long as cartographer_ros can subscribe to the topic of PointCloud2 messages, there are no limitations.

> Could I use that as the map and then use cartographer_ros to perform localisation?

Not sure what you mean. Cartographer can only localize on maps it made itself (there may be work-arounds in 2D).

ekow-mensah, 2018-05-04:
oh okay could you please explain the possible work-arounds?

ojura, 2018-05-04:
This an often requested feature. However, it's not trivial, and unfortunately not implemented (yet).
E.g. https://github.com/googlecartographer/cartographer_ros/issues/706#issuecomment-364882438

ekow-mensah, 2018-05-04:
Okay yes i'd imagine it will not be trivial to implement

eugenfunk, 2018-10-26:
It's trivial. :) Build the map as gazebo model, drive around with cartographer and save it. :))

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87