cartographer-project/cartographer_ros #856: What is the timestamp format in PBStream

mjseabright, 2018-05-07:
Hi, I've managed to pull the trajectory out of a PBStream file but I cannot work out what the timestamps mean.

Exampe:
`timestamp: 636533061005937350
pose {
  translation {
    x: 1.8459875994728516
    y: -4.098475079319226
  }
  rotation {
    z: -0.999340312949446
    w: 0.036317198599333245
  }
}
node_index: 1251`
I would have thought the timestamps would corresponded to times in the bag file. So the example above would be somewhere around 1517709302.83 or Feb 04 2018 14:55:02.83.

I noticed a similar timestamp in https://github.com/googlecartographer/cartographer_ros/issues/460 so I don't think it's that I'm decoding it incorrectly.

What does this timestamp mean? What time does it correspond to?

Thank you very much.

mjseabright, 2018-05-07:
A co-worker has pointed me in the right direction. [This timestamp proto](https://github.com/google/protobuf/blob/master/src/google/protobuf/timestamp.proto) shows how the timestamp is formed. 

The timestamps are relative to 0001-01-01T00:00:00Z rather than the Unix epoch. The units are 1/10000000 seconds.

ojura, 2018-05-07:
Use [`FromUniversal/ToUniversal`](https://github.com/googlecartographer/cartographer/blob/697d6091cd88c4fe869553abcbce147f21be74d4/cartographer/common/time.h#L54) to convert from/to proto timestamps.

woshidaye, 2018-09-09:
@mjseabright Hi, I want to extract data from pbstream. How do you extract trajectory from pbstream? I look forward to your reply.

mjseabright, 2018-09-09:
@woshidaye I used [this code](https://gist.github.com/mjcarroll/f620ef8d88efc3a03711259d6fc9a578). It is a little outdated now as I had to change all references of sparse_pose_graph to pose_graph. It may require more changes to be compatible with the latest version of Cartographer.

woshidaye, 2018-09-11:
@mjseabright I carried out according to your above steps, but I still made a mistake. /home/sy/图片/12.png

woshidaye, 2018-09-11:
@mjseabright I follow your steps, but there are still errors. When 'python export_pose_graph.py example.pbstream' is entered at the terminal, it still prompts errors  'ImportError: No module named cartographer.mapping'.

mjseabright, 2018-09-11:
Try it with python3. `python3 export_pose_graph.py example.pbstream`

woshidaye, 2018-09-12:
@mjseabright When I ran according to your command, a new error occurred :
File "export_pose_graph.py", line 5, in <module>
    import cartographer.mapping.proto.pose_graph_pb2
  File "/home/sy/py_export/cartographer/mapping/proto/pose_graph_pb2.py", line 6, in <module>
    from google.protobuf import descriptor as _descriptor
ImportError: No module named 'google'
How do I modify it here? Thank you.

woshidaye, 2018-09-12:
@mjseabright Thank you for your answer. However, there was another error: 
Traceback (most recent call last):
  File "export_pose_graph.py", line 6, in <module>
    import cartographer.mapping.proto.serialization_pb2
  File "/home/sy/py_export/cartographer/mapping/proto/serialization_pb2.py", line 16, in <module>
    from cartographer.mapping.proto import submap_pb2 as cartographer_dot_mapping_dot_proto_dot_submap__pb2
  File "/home/sy/py_export/cartographer/mapping/proto/submap_pb2.py", line 15
    from cartographer.mapping.proto.2d import grid_2d_pb2 as cartographer_dot_mapping_dot_proto_dot_2d_dot_grid__2d__pb2
                                    ^
SyntaxError: invalid syntax

Looking forward to your reply again. Thank you.

fsblu, 2019-03-15:
@woshidaye Could you solve the last error?

TSRyan, 2019-03-19:
@woshidaye @fsblu The `invalid syntax` error is due to the leading digit after the dot (...proto**.2d**) in
```
from cartographer.mapping.proto.2d import ...
```
You can get past it by putting in a leading underscore like this:
```
from cartographer.mapping.proto._2d import ...
```
You'll have to rename the associated folders and change it everywhere this happens.  You'll also have to change references to `SparsePoseGraph` and `sparse_pose_graph` to `PoseGraph` and `pose_graph`, respectively, but once that's done, it should at least run.

Having done that, I'm still not sure it works correctly, though.  When I try to run it on a test file I get:
```
$ ./export_pose_graph.py example.pbstream 
Found:  0  trajectories
0 0
```
If anyone can shed some light on this, I would appreciate it.

Bitfroest, 2019-03-20:
@TSRyan 
I guess all the data can be obtained using only the serialization because it already serializes the PoseGraph messages.
```
import cartographer.mapping.proto.serialization_pb2
...
def __init__(...):
    ...
    # create a new list
    self.serdatalist = []
    
def read_serialized_data():
    ...
    # add current message to list of messages in readable string format
    self.serdatalist.append(str(self.sd.__str__()))
```

[available functions doc](https://developers.google.com/protocol-buffers/docs/reference/python/google.protobuf.message.Message-class)

at the end you can write the whole list or just the parts you need to a file.

TSRyan, 2019-03-20:
Thanks, @Bitfroest.  That will get me where I need to go.

Bitfroest, 2019-03-25:
In addition to my previous post you can use the following code snipped to get the PoseGraph as dict
```
from google.protobuf.json_format import MessageToDict
...
        dict_obj = MessageToDict(self.sd)
        if 'poseGraph' in dict_obj:
            self.serdatalist.append(dict_obj)
        return True
```
Source: https://stackoverflow.com/questions/19734617/protobuf-to-json-in-python

In case someone else wants to convert the times here is my way.

> Range is from 0001-01-01T00:00:00Z to 9999-12-31T23:59:59.999999999Z.

> The units are 1/10000000 seconds.

> If you need to convert from those numbers to Unix time, substract 621355968000000000L, which is the Unix epoch expressed in ticks.

Here the python conversion with using the combined information.
```
>> import time
>> print( time.ctime( (636533061005937350 - 621355968000000000) / 10000000) )
Sun Feb  4 02:55:00 2018
1517709300.593735
```

Source: https://stackoverflow.com/questions/1832714/18-digit-timestamp

artemiialessandrini, 2019-10-10:
While running 
` python3 export_pose_graph.py ~/.ros/map.pbstream
`, 

I'm getting the following error:
`Traceback (most recent call last):
  File "export_pose_graph.py", line 34, in <module>
    from cartographer.mapping.proto.serialization_pb2 import SerializationHeader, SerializedData
  File "/home/artemii/catkin_ws/src/cartographer_ros/cartographer_ros/scripts/cartographer/mapping/proto/serialization_pb2.py", line 17, in <module>
    from cartographer.mapping.proto import submap_pb2 as cartographer_dot_mapping_dot_proto_dot_submap__pb2
  File "/home/artemii/catkin_ws/src/cartographer_ros/cartographer_ros/scripts/cartographer/mapping/proto/submap_pb2.py", line 16, in <module>
    from cartographer.mapping.proto.i2d import grid_2d_pb2 as cartographer_dot_mapping_dot_proto_dot_2d_dot_grid__2d__pb2
  File "/home/artemii/catkin_ws/src/cartographer_ros/cartographer_ros/scripts/cartographer/mapping/proto/i2d/grid_2d_pb2.py", line 16, in <module>
    from cartographer.mapping.proto.i2d import map_limits_pb2 as cartographer_dot_mapping_dot_proto_dot_2d_dot_map__limits__pb2
  File "/home/artemii/catkin_ws/src/cartographer_ros/cartographer_ros/scripts/cartographer/mapping/proto/i2d/map_limits_pb2.py", line 16, in <module>
    from cartographer.mapping.proto.i2d import cell_limits_pb2 as cartographer_dot_mapping_dot_proto_dot_2d_dot_cell__limits__pb2
  File "/home/artemii/catkin_ws/src/cartographer_ros/cartographer_ros/scripts/cartographer/mapping/proto/i2d/cell_limits_pb2.py", line 22, in <module>
    serialized_pb=_b('\n/cartographer/mapping/proto/i2d/cell_limits.proto\x12\x1a\x63\x61rtographer.mapping.proto\"6\n\nCellLimits\x12\x13\n\x0bnum_x_cells\x18\x01 \x01(\x05\x12\x13\n\x0bnum_y_cells\x18\x02 \x01(\x05\x62\x06proto3')
  File "/home/artemii/.local/lib/python3.6/site-packages/google/protobuf/descriptor.py", line 879, in __new__
    return _message.default_pool.AddSerializedFile(serialized_pb)
TypeError: Couldn't parse file content!
`

Someone faced with it?

artemiialessandrini, 2019-10-12:
MessageToDict is a great tool.

I assume `import json` could also solve this transformation without additional installation:
`print (json.dumps(message))`

closed 2020-10-21 by MichaelGrupp