cartographer-project/cartographer_ros #861: Submaps poor quality

alfred31, 2018-05-09:
Hello.
I am using cartographer with a rotating planar 2d LIDAR. The thing is that the submaps are really poor. Besides the system does not close the loop.
I saw that the Z axis decreases continuously even though the robot is moving in a planar floor.
Any idea why is this happening?
These are the results of rosbag validation are showed below:
W0509 11:56:48.749516 15162 rosbag_validate_main.cc:103] frame_id imu_link time 1525707113829814862: IMU linear acceleration is 0.458578 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.0434229 -0.228816 00.395034
E0509 11:56:48.749853 15162 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /sensors/imu and /sensors/imu_raw
W0509 11:56:48.750057 15162 rosbag_validate_main.cc:352] frame_id "foxizuyo_hokuyo" on topic /foxizuyo/scan has serialization time 1525707113.912360266 but sensor time 1525707113.789593280 differing by -0.122767 s.
W0509 11:56:48.750074 15162 rosbag_validate_main.cc:103] frame_id imu_link time 1525707113842995174: IMU linear acceleration is 0.230427 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 000.156535 -0.0262548 000.167045
W0509 11:56:48.750098 15162 rosbag_validate_main.cc:103] frame_id imu_link time 1525707113859665360: IMU linear acceleration is 0.841206 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.028564 00.46092 0.703111
E0509 11:56:48.750174 15162 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0509 11:56:48.750187 15162 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /sensors/imu and /sensors/imu_raw
E0509 11:56:48.750205 15162 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0509 11:56:48.750217 15162 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /sensors/imu and /sensors/imu_raw
E0509 11:56:48.750239 15162 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0509 11:56:49.294574 15162 rosbag_validate_main.cc:365] Average IMU linear acceleration is 5.9268 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0509 11:56:49.294968 15162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/foxizuyo/scan" (frame_id: "foxizuyo_hokuyo"):
Count: 8905  Min: 0.027727  Max: 0.040933  Mean: 0.033686
[0.027727, 0.029048)	                    	Count: 33 (0.370578%)	Total: 33 (0.370578%)
[0.029048, 0.030368)	               #####	Count: 2114 (23.739471%)	Total: 2147 (24.110050%)
[0.030368, 0.031689)	                  ##	Count: 710 (7.973049%)	Total: 2857 (32.083099%)
[0.031689, 0.033009)	                   #	Count: 662 (7.434026%)	Total: 3519 (39.517124%)
[0.033009, 0.034330)	                ####	Count: 1778 (19.966311%)	Total: 5297 (59.483437%)
[0.034330, 0.035651)	                  ##	Count: 1077 (12.094329%)	Total: 6374 (71.577766%)
[0.035651, 0.036971)	                 ###	Count: 1512 (16.979225%)	Total: 7886 (88.556992%)
[0.036971, 0.038292)	                   #	Count: 520 (5.839416%)	Total: 8406 (94.396408%)
[0.038292, 0.039612)	                    	Count: 90 (1.010668%)	Total: 8496 (95.407074%)
[0.039612, 0.040933]	                   #	Count: 409 (4.592926%)	Total: 8905 (100.000000%)
E0509 11:56:49.295166 15162 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.0534519 s, recommended is [0.0005, 0.005] s with no jitter.
I0509 11:56:49.296026 15162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/sensors/imu" (frame_id: "imu_link"):
Count: 48309  Min: -0.040272  Max: 0.053452  Mean: 0.006210
[-0.040272, -0.030899)	                    	Count: 2 (0.004140%)	Total: 2 (0.004140%)
[-0.030899, -0.021527)	                    	Count: 1 (0.002070%)	Total: 3 (0.006210%)
[-0.021527, -0.012155)	                    	Count: 0 (0.000000%)	Total: 3 (0.006210%)
[-0.012155, -0.002782)	                    	Count: 0 (0.000000%)	Total: 3 (0.006210%)
[-0.002782, 0.006590)	         ###########	Count: 27754 (57.450993%)	Total: 27757 (57.457203%)
[0.006590, 0.015963)	           #########	Count: 20544 (42.526237%)	Total: 48301 (99.983437%)
[0.015963, 0.025335)	                    	Count: 6 (0.012420%)	Total: 48307 (99.995857%)
[0.025335, 0.034707)	                    	Count: 0 (0.000000%)	Total: 48307 (99.995857%)
[0.034707, 0.044080)	                    	Count: 1 (0.002070%)	Total: 48308 (99.997932%)
[0.044080, 0.053452]	                    	Count: 1 (0.002070%)	Total: 48309 (100.000000%)
I0509 11:56:49.296891 15162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/odometry/loc" (frame_id: "odom"):
Count: 29973  Min: 0.000000  Max: 0.031161  Mean: 0.010008
[0.000000, 0.003116)	              ######	Count: 8251 (27.528109%)	Total: 8251 (27.528109%)
[0.003116, 0.006232)	                   #	Count: 923 (3.079438%)	Total: 9174 (30.607546%)
[0.006232, 0.009348)	                   #	Count: 1342 (4.477363%)	Total: 10516 (35.084911%)
[0.009348, 0.012464)	                  ##	Count: 3051 (10.179161%)	Total: 13567 (45.264072%)
[0.012464, 0.015580)	               #####	Count: 8094 (27.004303%)	Total: 21661 (72.268372%)
[0.015580, 0.018696)	               #####	Count: 7465 (24.905748%)	Total: 29126 (97.174126%)
[0.018696, 0.021813)	                    	Count: 584 (1.948420%)	Total: 29710 (99.122543%)
[0.021813, 0.024929)	                    	Count: 158 (0.527141%)	Total: 29868 (99.649681%)
[0.024929, 0.028045)	                    	Count: 98 (0.326961%)	Total: 29966 (99.976646%)
[0.028045, 0.031161]	                    	Count: 7 (0.023354%)	Total: 29973 (100.000000%)

And my backpack 3d.lua:

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 0.3,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,

}

TRAJECTORY_BUILDER_3D.submaps.num_range_data = 10
TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 12
MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 10
POSE_GRAPH.constraint_builder.sampling_ratio = 0.01
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

alfred31, 2018-05-09:
But shouldn't 3D SLAM work with a rotating LIDAR? I am able to localize the robot , the problem is just the loop closure and the Z axis (This one might by caused by the IMU data).
The thing that worries me the most is the poor quality of the submaps, then the /map (projected map in 2D) is not generated

gaschler, 2018-05-09:
> But shouldn't 3D SLAM work with a rotating LIDAR? 

If you mean a 2D lidar that is *tilting* so its laser plane rotates up and down, yes.
But a tilting 2D lidar is hard to calibrate and configure correctly.
In that case, you would need an accurate tf publisher of the tilted lidar transform and have a flawless urdf config (which you haven't posted yet).

> loop closure

First tune local slam alone and make sure your submaps look good. Please see the tuning guide how to tune local slam alone.

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87