cartographer-project/cartographer_ros #865: cartographer with hokuyo2d and laser_scan_matcher

BakhtawarRehman, 2018-05-14:
bag_validate:= https://gist.github.com/BakhtawarRehman/c89a0f3960311e50814b15aa1b8b1c34
files:= https://github.com/BakhtawarRehman/cartographer
bag:= https://drive.google.com/file/d/1G_5SbVE9IkSpOwtvv-66E-3cx3D3hU2s/view?usp=sharing

Problem:

I cannot create the map. Though the topic "map" shows upon entering command "rostopic list" and rqt_graph but in rviz map topic dose not exist.

I have used laser_scan_matcher and hokuyo lidar to record the map (as i am using only lidar without any odometry).
In rviz only visualization available is the laser readings and thats all.

gaschler, 2018-05-14:
So you don't see a trajectory or multiple submaps in rviz.
Can you please share the terminal log output of cartographer_node?
Perhaps it is just waiting for input on a different topic.

Also, try swapping
```
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 1,
```

BakhtawarRehman, 2018-05-14:
Hello Gascheler, Thanks for instant response.

I cannot see anything here other than laser (image: https://drive.google.com/file/d/1JVouHZ6C1UR_l-Aq8G44jVSp7tmFgUKf/view?usp=sharing).
I hav tried swapping the values but no help.

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87