cartographer-project/cartographer_ros #866: Cartographer latency

cressell, 2018-05-14:
Hi!

I would like to be able to see what the latency between laser scan callback to map update is in Cartographer using ROS. I haven't been able to find anything built in to do this, do you perhaps have any suggestions?

Best-
Cressell

gaschler, 2018-05-15:
You manually measure most of these latencies in cartographer_ros `Node`.
For instance, the latency from when a laser scan was received to the local SLAM result can be measured
in `Node::PublishTrajectoryStates` by computing `ros::Time::now() - trajectory_state.local_slam_data->time`. Of course, it won't be exact because `PublishTrajectoryStates` runs only at a certain frequency.

Can you detail what you mean by "map update". Do you mean the end of a global SLAM optimization run?

cressell, 2018-05-15:
Thank you for your answer! That latency will help me. 
Both the global optimization run and and the insertion to the local submap would be interesting to see.

closed 2018-05-31 by cressell