cartographer-project/cartographer_ros #867: How can I get a 3D pointcloud map using cartographer?

DriftingTimmy, 2018-05-15:
After tuning the urdf and lua configuration, I run the cartographer and get a useful trajectory.
According to the tutorial, I called the "/write_state" service to get a .pbstream file, using the command:

        rosservice call /write_state ${HOME}/Downloads/test_save.1.bag.pbstream.

 And I get an error like:

        ERROR: Unable to load type [cartographer_ros_msgs/WriteState].
        Have you typed 'make' in [cartographer_ros_msgs]?

I know the .pbstream file is the input of the asset_writer node to generate the 3D pointcloud map. So I want to know how to get it down. Because it is not about the tuning, I decide not to upload the config files.

Thanks a lot!

kentsommer, 2018-05-15:
This message likely means that the terminal that you are calling the command from has not sourced the cartographer workspace (`source install_isolated/setup.bash`)

DriftingTimmy, 2018-05-15:
@kentsommer  Thanks! You are right! I thought the rosservice do not need to source the path.
And after I get may .pbstream file "test_save.pbstream", I run the command ：

roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:=/home/lx/Rosbags/segmentmatching/try2.bag pose_graph_filename:=/home/lx/Downloads/test_save.1.bag.pbstream

And I get the node corrupted as ： 

W0515 14:48:38.401289 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
W0515 14:48:38.401427 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
W0515 14:48:38.401464 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
W0515 14:48:38.401489 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
W0515 14:48:38.401512 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
W0515 14:48:38.401536 25744 xray_points_processor.cc:147] Not writing output: bounding box is empty.
I0515 14:48:38.401553 25744 counting_points_processor.cc:43] Processed 0 and finishing.
================================================================================REQUIRED process [cartographer_assets_writer-1] has died!
process has finished cleanly
log file: /home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1*.log
Initiating shutdown!
================================================================================

Thanks again～

kentsommer, 2018-05-15:
Does the bag file that you are passing match the data used to generate the `.pbstream`? Are you sure it is actually slamming? Those errors would imply to me that cartographer was never slamming.

DriftingTimmy, 2018-05-15:
What do u mean that cartographer never slamming? I just run the asset_writer node. The bag file and the .pbstream file is matched of course~ Should I upload my config file and the corresponding bag file?
Thanks~

kentsommer, 2018-05-15:
@DriftingTimmy 

I mean that it doesn't look like cartographer is actually building a map (SLAM is not happening). 

If cartographer is working properly check:
* tracking frame in `assets_writer_backpack_3d.lua` (should match the tracking frame you have set in your `xyz.lua` file that you use when running cartographer)
* urdf config file in `assets_writer_backpack_3d.launch` (should point to the urdf you use while running cartographer)

DriftingTimmy, 2018-05-15:
@kentsommer  Well, I think u mean that I need to make sure that the configuration file is correct.
Actually, I think my configuration is all right and the corresponding frames are the same. The tracking frame is "imu". 
Here are some probable problems. Should I run the cartographer_node and the asset_writer node at the same time? I tried but I failed on the same error. Then maybe I run the wrong command at the first time getting the  .pbstream  file? I just call this service when rosbag is done, should I call this service at the beginning of the rosbag time? And I m going to check the asset_writer docs to find the mistakes.
Thanks~

kentsommer, 2018-05-15:
@DriftingTimmy 

You have two options (ways to run):

Online Cartographer (before killing cartographer_ros run these):
* `rosservice call /finish_trajectory 0`
* `rosservice call /write_state xyz_your_file_name.pbstream`

Offline Cartographer:
* Don't have to run anything manually, it generates the necessary .pbstream file

After finishing the above:
* `roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:=path_to_bag pose_graph_filename:=path_to_pbstream`

DriftingTimmy, 2018-05-15:
@kentsommer  I just followed your suggestions and run the offline_node instead. When I run the offline node, I get warnings:

[ WARN] [1526372510.416963827, 1526043655.019913171]: W0515 16:21:50.000000 28105 tf_bridge.cc:52] "imu" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1526372510.417029835, 1526043655.035860380]: W0515 16:21:50.000000 28105 tf_bridge.cc:52] "imu" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1526372510.417066582, 1526043655.035860380]: W0515 16:21:50.000000 28105 tf_bridge.cc:52] "imu" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1526372510.417131340, 1526043655.067925180]: W0515 16:21:50.000000 28105 tf_bridge.cc:52] "imu" passed to lookupTransform argument source_frame does not exist.

And I get the corresponding  .pbstream  file. After running the asset_writer, I get fatal errors: 

F0515 16:25:41.712893 29841 assets_writer.cc:151] Check failed: pose_graph_.trajectory_size() == bag_filenames_.size() (0 vs. 1) Pose graphs contains 0 trajectories while 1 bags were provided. This tool requires one bag for each trajectory in the same order as the correponding trajectories in the pose graph proto.
*** Check failure stack trace: ***
    @     0x7f8ca1128daa  (unknown)
    @     0x7f8ca1128ce4  (unknown)
    @     0x7f8ca11286e6  (unknown)
    @     0x7f8ca112b687  (unknown)
    @           0x4b0f7b  cartographer_ros::AssetsWriter::AssetsWriter()
    @           0x4ab4c1  main
    @     0x7f8c9dfeef45  (unknown)
    @           0x4ad0d7  (unknown)
    @              (nil)  (unknown)
================================================================================REQUIRED process [cartographer_assets_writer-1] has died!
process has died [pid 29841, exit code -6, cmd /home/lx/catkin_cartographer/install_isolated/lib/cartographer_ros/cartographer_assets_writer -configuration_directory /home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename assets_writer_backpack_3d.lua -urdf_filename /home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/urdf/robo_3d.urdf -bag_filenames /home/lx/Rosbags/segmentmatching/try2.bag -pose_graph_filename /home/lx/Rosbags/segmentmatching/try2.bag.pbstream __name:=cartographer_assets_writer __log:=/home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1.log].
log file: /home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1*.log
Initiating shutdown!
================================================================================

This time the problem is different. What does it mean that the trajectories are not matched?
Thx a lot!

kentsommer, 2018-05-15:
It means that your configuration is likely wrong:

* `"imu" passed to lookupTransform argument source_frame does not exist` (this likely means your urdf is incorrect for your setup)
* `Pose graphs contains 0 trajectories` (this means cartographer never built a map i.e. nothing happened and your pose graph is empty)

DriftingTimmy, 2018-05-15:
@kentsommer  https://github.com/DriftingTimmy/carto_config Here is my configuration files.
The bag file is too big, I m considering record a new bag.
But I just figured out the problem of my bag file is that the message style is not satisfied with cartographer. And I change it to the matched style and then I run:

roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:=/home/lx/Rosbags/segmentmatching/try2_mod.bag pose_graph_filename:=/home/lx/Rosbags/segmentmatching/try2_save.pbstream

In the beginning, the programme seems to be alright loading the bagfile(its size is about 13G) and other configurations. But after few minutes, I get a new error like this:

... logging to /home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/roslaunch-lx-29488.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lx:35013/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    cartographer_assets_writer (cartographer_ros/cartographer_assets_writer)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cartographer_assets_writer-1]: started with pid [29506]
I0515 19:42:33.251335 29506 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/assets_writer_backpack_3d.lua' for 'assets_writer_backpack_3d.lua'.
I0515 19:42:33.251737 29506 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I0515 19:42:33.251763 29506 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I0515 19:42:33.256165 29506 assets_writer.cc:194] Processing /home/lx/Rosbags/segmentmatching/try2_mod.bag...
I0515 19:42:33.341763 29506 assets_writer.cc:260] Processed 0 of 635.477 bag time seconds...
F0515 19:48:19.481357 29506 hybrid_grid.h:387] Check failed: new_bits <= 8 (9 vs. 8) 
*** Check failure stack trace: ***
    @     0x7f45a3e8edaa  (unknown)
    @     0x7f45a3e8ece4  (unknown)
    @     0x7f45a3e8e6e6  (unknown)
    @     0x7f45a3e91687  (unknown)
    @           0x4e9e5e  cartographer::mapping::DynamicGrid<>::mutable_value()
    @           0x4e945c  cartographer::io::XRayPointsProcessor::Insert()
    @           0x4e97bc  cartographer::io::XRayPointsProcessor::Process()
    @           0x4e47d9  cartographer::io::CountingPointsProcessor::Process()
    @           0x4e1fc8  cartographer::io::MinMaxRangeFiteringPointsProcessor::Process()
    @           0x4af6cf  cartographer_ros::AssetsWriter::Run()
    @           0x4ab4ee  main
    @     0x7f45a0d54f45  (unknown)
    @           0x4ad0d7  (unknown)
    @              (nil)  (unknown)
================================================================================REQUIRED process [cartographer_assets_writer-1] has died!
process has died [pid 29506, exit code -6, cmd /home/lx/catkin_cartographer/install_isolated/lib/cartographer_ros/cartographer_assets_writer -configuration_directory /home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename assets_writer_backpack_3d.lua -urdf_filename /home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/urdf/robo_3d.urdf -bag_filenames /home/lx/Rosbags/segmentmatching/try2_mod.bag -pose_graph_filename /home/lx/Rosbags/segmentmatching/try2_save.pbstream __name:=cartographer_assets_writer __log:=/home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1.log].
log file: /home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1*.log
Initiating shutdown!
================================================================================

I do not know what is wrong here. Does it mean that the data is too large or something else?

gaschler, 2018-05-15:
`F0515 19:48:19.481357 29506 hybrid_grid.h:387] Check failed: new_bits <= 8 (9 vs. 8)`
means that a voxel grid is too large.
It think the limitation is 8192 voxels, so it should usually be in hundreds of meters.
See https://github.com/googlecartographer/cartographer/issues/937
(Note that in SLAM itself, this limitation only applies to the size of a submap, not to the entire map.)

DriftingTimmy, 2018-05-16:
@gaschler  Thanks for your answer!! I will record a new data in a smaller environment. I just enlarge the voxel size to 10cm to make sure the assets_writer can run well. And I really got some  .png  pictures like this:

[image]
 
But I can not get the ply file either. Because I got a new error which I think is caused by the large bag file data:

process[cartographer_assets_writer-1]: started with pid [30630]
I0516 10:56:18.579820 30630 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/assets_writer_backpack_3d.lua' for 'assets_writer_backpack_3d.lua'.
I0516 10:56:18.580185 30630 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I0516 10:56:18.580199 30630 configuration_file_resolver.cc:41] Found '/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I0516 10:56:18.584894 30630 assets_writer.cc:194] Processing /home/lx/Rosbags/segmentmatching/try2_mod.bag...
I0516 10:56:18.668066 30630 assets_writer.cc:260] Processed 0 of 635.477 bag time seconds...
I0516 11:07:00.415024 30630 counting_points_processor.cc:43] Processed 335311832 and finishing.
================================================================================REQUIRED process [cartographer_assets_writer-1] has died!
process has finished cleanly
log file: /home/lx/.ros/log/768f62ce-53ef-11e8-9151-1c1b0d00bd07/cartographer_assets_writer-1*.log
Initiating shutdown!
================================================================================

So if I have a new smaller bag file, everything will be alright? Thx

DriftingTimmy, 2018-05-16:
@kentsommer Thank u so much for your help! I think the points are too many, is there any configuration params can be modified to downsample the data?
And I just tried to generate the  .ply  file by adding the command you mentioned. But I cannot get the  . ply  file after the cartographer_assets_writer shutting itself down. Do I have to modify other params?

kentsommer, 2018-05-16:
@DriftingTimmy 

The size of your data is not an issue. Did you rebuild and source the workspace? 

In other words, ensure that the file here `/home/lx/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/assets_writer_backpack_3d.lua'` has the `write_ply` section in it.

DriftingTimmy, 2018-05-16:
@kentsommer  Thank U so much! I rebuild the work space and check it again. It worked well.
And the final map size is too big(6.5G), is there any parameters we can change to downsample the map data during the assets_writer step?
Thx again~

kentsommer, 2018-05-16:
@DriftingTimmy 

One option is modifying the `min_max_range_filter`:
```
{
      action = "min_max_range_filter",
      min_range = 1.,
      max_range = 60.,
    },
```

There is also another processor that might be useful (`voxel_filter_and_remove_moving_objects`). I don't' remember the parameters for this one though.

DriftingTimmy, 2018-05-16:
@kentsommer  hhh That's enough for me~ I tried to convert my ply file to pcd file using the command "pcl_ply2pcd" and the process dumped like this:
Convert a PLY file to PCD format. For more information, use: pcl_ply2pcd -h
PCD output format: binary
Segmentation fault (core dumped)

I think the ply file is too big(about 6.5G), so I need to downsample the input data~

kentsommer, 2018-05-17:
@DriftingTimmy 

The file size is really not an issue. There are many people that process much larger datasets. That being said, `pcl_ply2pcd` is a part of PCL not cartographer. So, you would likely be better served opening an issue there (https://github.com/PointCloudLibrary/pcl)

DriftingTimmy, 2018-05-17:
ok～ Thanks a lot~ U really helped me so much~
I m going to close this issue~

kentsommer, 2018-05-21:
@DriftingTimmy 

I suggest you take a look at the [tuning documentation](http://google-cartographer-ros.readthedocs.io/en/latest/tuning.html). 

The parameter you are looking for is `linear_z_search_window`

gaschler, 2018-05-22:
@kentsommer Thanks for all your helpful answers.

In addition, make sure your transformation between the IMU and the laser scanner in the urdf config is accurately calibrated.
In case you use odometry, temporarily disable it in cartographer_ros because it is another possible source of error.

DriftingTimmy, 2018-05-22:
@kentsommer  @gaschler 
Thank u so much! I've tried to tuning the params to make my data detect a loop.
And it is really a difficult job to make sure the params can satisfy all the data environment, but I finally succeeded once.
Thanks again~

closed 2018-05-22 by DriftingTimmy

daiencan, 2018-09-11:
@DriftingTimmy  Hi, I am a master's student at the University of Cqupt and I am trying to implement 3D SLAM using the RSLiDAR-16. The cartographer node subscribes to the /rslidar_points topic , /imu topic and /odom topic. however, the sub-map does not build. There is no error or warning in the terminal. I will really appreciate some help as to why the sub-map is not being constructed. Thank you very much

DriftingTimmy, 2018-11-10:
@daiencan Sorry for replying you so late cause I was busy with my job and graduation. I don't know if u have solved this problem or not?

Duke-Allen, 2019-02-22:
@DriftingTimmy
Hello, I also got the problem when running cartographer with IMU and 16-lines lidar.
The rosbag_validate result is:
[image]
My data bag:
[image]
I could not run the cartographer and there is no map build
[image]
[image]
my files:
[demo_rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration.git)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration.git)
Can you help me in this?

Duke-Allen, 2019-03-16:
@DriftingTimmy 
Hi, I have been running 3D cartographer successfully with no error by a 16-lines LIDAR and a MPU6050. But the track is inconsistent with the real movement, like this:
[image]
The result of bag is:
- I0316 12:20:33.418805 11044 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 18056  Min: 0.004353  Max: 0.013893  Mean: 0.010000
[0.004353, 0.005307)	                    	Count: 1 (0.005538%)	Total: 1 (0.005538%)
[0.005307, 0.006261)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.006261, 0.007215)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.007215, 0.008169)	                    	Count: 281 (1.556269%)	Total: 282 (1.561808%)
[0.008169, 0.009123)	                    	Count: 73 (0.404298%)	Total: 355 (1.966105%)
[0.009123, 0.010077)	 ###################	Count: 17197 (95.242577%)	Total: 17552 (97.208687%)
[0.010077, 0.011031)	                    	Count: 151 (0.836287%)	Total: 17703 (98.044968%)
[0.011031, 0.011985)	                    	Count: 266 (1.473194%)	Total: 17969 (99.518166%)
[0.011985, 0.012939)	                    	Count: 86 (0.476296%)	Total: 18055 (99.994461%)
[0.012939, 0.013893]	                    	Count: 1 (0.005538%)	Total: 18056 (100.000000%)
E0316 12:20:33.421423 11044 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101069 s, recommended is [0.0005, 0.05] s with no jitter.
I0316 12:20:33.421811 11044 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1791  Min: 0.100548  Max: 0.101069  Mean: 0.100801
[0.100548, 0.100600)	                    	Count: 10 (0.558347%)	Total: 10 (0.558347%)
[0.100600, 0.100652)	                    	Count: 7 (0.390843%)	Total: 17 (0.949190%)
[0.100652, 0.100704)	                    	Count: 10 (0.558347%)	Total: 27 (1.507538%)
[0.100704, 0.100756)	                   #	Count: 51 (2.847571%)	Total: 78 (4.355109%)
[0.100756, 0.100809)	       #############	Count: 1190 (66.443329%)	Total: 1268 (70.798439%)
[0.100809, 0.100861)	               #####	Count: 478 (26.689001%)	Total: 1746 (97.487434%)
[0.100861, 0.100913)	                    	Count: 21 (1.172529%)	Total: 1767 (98.659966%)
[0.100913, 0.100965)	                    	Count: 10 (0.558347%)	Total: 1777 (99.218315%)
[0.100965, 0.101017)	                    	Count: 9 (0.502513%)	Total: 1786 (99.720825%)
[0.101017, 0.101069]	                    	Count: 5 (0.279174%)	Total: 1791 (100.000000%)

Here is my [bag](https://share.babel.cc/share.do?s=4553105679638233). 
The account: 15838868117
Password: 123456789
Any suggestions would be highly appreciated!

ManChrys, 2021-07-12:
Hello @Duke-Allen !!! I have exactly the same equipment. I want to ask you how did you connect imu and lidar. They are one above the other ? Did you set that offset in any conf file or anywhere else? Thanks in advance.