cartographer-project/cartographer_ros #868: running error with one Lidar and a imu

coldrainy, 2018-05-16:
Hi,I got problems when running the cartographer_ros with my own 3d 16-laser lidar and one IMU rosbag data.Here are the details.
1.The rosbag_validate result link is [here](https://gist.github.com/coldrainy/a06027e37c9a23e4c0db16e115381cb2).
2.My configuration and launch and URDF files are [here](https://github.com/coldrainy/carto_configfiles.git).

I can successfully run the program.But there was no map generated and the /scan_matched_points2 look like the picture below.So what is the problem and what should I do to fix it?

[image]
[image]

kentsommer, 2018-05-16:
A couple things to check:
* `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160` should likely be set to `1` or `2` in your case if your `rs_lider` driver is publishing a full rotation per `PointCloud2` message. 
* Your URDF is almost certainly incorrect (specifically the orientation of the IMU). Your linear acceleration  is way off (`Average IMU linear acceleration is 1.84624 m/s^2, expected is [9.5, 10.5] m/s^2`) which likely means the URDF does not match the real world orientation of your IMU.

coldrainy, 2018-05-18:
@kentsommer Thank you!I fixed the parameter that you point out and it worked!

closed 2018-05-18 by coldrainy

Duke-Allen, 2019-02-22:
@coldrainy @kentsommer 
Hello, I also got the problem when running cartographer with IMU and 16-lines lidar.
The rosbag_validate result is:
[image]
My data bag:
[image]
I could not run the cartographer and there is no map build
[image]
[image]
my files:
[demo_rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration.git)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration.git)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration.git)
Can you help me in this?

Duke-Allen, 2019-03-16:
@kentsommer @coldrainy 
Hi, I have been running 3D cartographer successfully with no error by a 16-lines LIDAR and a MPU6050. But the track is inconsistent with the real movement, like this:
[image]

Here is my [bag](https://share.babel.cc/share.do?s=4553105679638233). 
The account: 15838868117
Password: 123456789