cartographer-project/cartographer_ros #876: Build error on ubuntu 16.04

danningwang, 2018-05-24:
Hello,

I should tell you my ubuntu upgraded from 14.04 to 16.04 a few days ago and I successfully installed and used cartographer on 14.04. But now It shows error when I build cartographer using "catkin_make_isolated --install --use-ninja" following the instructions on https://google-cartographer-ros.readthedocs.io/en/latest/index.html.  I think the error related to protobuf and also try the solution on https://gist.github.com/kdaun/51b7d19bbcd5e0798c8415cd093078fb. However, it still showed errors. Could you please tell me what I should do now?

The error is shown below:

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/danny/cartographer_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/danny/cartographer_ws/build_isolated/cartographer/install'
[1/1] Re-running CMake...
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /home/danny/cartographer_ws/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/danny/cartographer_ws/build_isolated/cartographer/install
ninja: no work to do.
==> ninja -j4 -l4 in '/home/danny/cartographer_ws/build_isolated/cartographer/install'
[6/131] Linking CXX executable cartographer.sensor.internal.voxel_filter_test
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   CMakeFiles/cartographer.sensor.internal.voxel_filter_test.dir/cartographer/sensor/internal/voxel_filter_test.cc.o  -o cartographer.sensor.internal.voxel_filter_test  -rdynamic libcartographer.a gmock/libgmock_main.a -lpthread libcartographer_test_library.a libcartographer.a /home/danny/cartographer_ws/install_isolated/lib/libceres.a -lglog -lgflags -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog -lgflags -lcairo /usr/local/lib/x86_64-linux-gnu/libprotobuf.a -lpthread && :
libcartographer.a(adaptive_voxel_filter_options.pb.cc.o): In function `cartographer::sensor::proto::protobuf_cartographer_2fsensor_2fproto_2fadaptive_5fvoxel_5ffilter_5foptions_2eproto::(anonymous namespace)::protobuf_AssignDescriptors()':
adaptive_voxel_filter_options.pb.cc:(.text+0x570): undefined reference to `google::protobuf::internal::AssignDescriptors(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::MessageFactory*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
libcartographer.a(adaptive_voxel_filter_options.pb.cc.o): In function `cartographer::sensor::proto::protobuf_cartographer_2fsensor_2fproto_2fadaptive_5fvoxel_5ffilter_5foptions_2eproto::(anonymous namespace)::AddDescriptorsImpl()':
adaptive_voxel_filter_options.pb.cc:(.text+0x3dd): undefined reference to `google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))'
libcartographer.a(adaptive_voxel_filter_options.pb.cc.o):(.data.rel.ro._ZTVN12cartographer6sensor5proto26AdaptiveVoxelFilterOptionsE[_ZTVN12cartographer6sensor5proto26AdaptiveVoxelFilterOptionsE]+0x20): undefined reference to `google::protobuf::Message::GetTypeName[abi:cxx11]() const'
libcartographer.a(adaptive_voxel_filter_options.pb.cc.o):(.data.rel.ro._ZTVN12cartographer6sensor5proto26AdaptiveVoxelFilterOptionsE[_ZTVN12cartographer6sensor5proto26AdaptiveVoxelFilterOptionsE]+0x58): undefined reference to `google::protobuf::Message::InitializationErrorString[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
[6/131] Linking CXX executable cartogr...her.sensor.compressed_point_cloud_test
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   CMakeFiles/cartographer.sensor.compressed_point_cloud_test.dir/cartographer/sensor/compressed_point_cloud_test.cc.o  -o cartographer.sensor.compressed_point_cloud_test  -rdynamic libcartographer.a gmock/libgmock_main.a -lpthread libcartographer_test_library.a libcartographer.a /home/danny/cartographer_ws/install_isolated/lib/libceres.a -lglog -lgflags -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog -lgflags -lcairo /usr/local/lib/x86_64-linux-gnu/libprotobuf.a -lpthread && :
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
sensor.pb.cc:(.text+0x651): undefined reference to `google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::io::CodedOutputStream*)'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::protobuf_cartographer_2fsensor_2fproto_2fsensor_2eproto::(anonymous namespace)::AddDescriptorsImpl()':
sensor.pb.cc:(.text+0xe99): undefined reference to `google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::protobuf_cartographer_2fsensor_2fproto_2fsensor_2eproto::(anonymous namespace)::protobuf_AssignDescriptors()':
sensor.pb.cc:(.text+0x100b): undefined reference to `google::protobuf::internal::AssignDescriptors(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::MessageFactory*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::SharedCtor()':
sensor.pb.cc:(.text+0x1a63): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::SharedDtor()':
sensor.pb.cc:(.text+0x1aa0): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::Clear()':
sensor.pb.cc:(.text+0x3268): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::MergeFrom(cartographer::sensor::proto::LandmarkData_LandmarkObservation const&)':
sensor.pb.cc:(.text+0x3865): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::LandmarkData_LandmarkObservation(cartographer::sensor::proto::LandmarkData_LandmarkObservation const&)':
sensor.pb.cc:(.text+0x3e32): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
libcartographer.a(sensor.pb.cc.o):sensor.pb.cc:(.text+0x5f71): more undefined references to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]' follow
libcartographer.a(sensor.pb.cc.o): In function `cartographer::sensor::proto::LandmarkData_LandmarkObservation::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)':
[... cut ...]
<== Failed to process package 'cartographer': 
  Command '['/home/danny/cartographer_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/danny/cartographer_ws/build_isolated/cartographer && /home/danny/cartographer_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

----------------------------------------------------------------------------------------------------------------------------------------------------
Thank you in advance.

Danning

danningwang, 2018-05-25:
Add some information:

protoc --version 
libprotoc 3.4.0

env | grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/home/danny/kinetic_ws/src/cartographer_ros/cartographer_ros_msgs:/home/danny/kinetic_ws/src/ceres-solver:/home/danny/kinetic_ws/src/cartographer:/home/danny/kinetic_ws/src/cartographer_ros/cartographer_ros:/home/danny/kinetic_ws/src/cartographer_ros/cartographer_rviz:/home/danny/kinetic_ws/src/g3_mapping:/home/danny/kinetic_ws/src/my_cartographer:/home/danny/kinetic_ws/src/dev-env/cli:/opt/ros/kinetic/share
ROS_MASTER_URI=http://localhost:11311
ROS_VERSION=1
ROSLISP_PACKAGE_DIRECTORIES=/home/danny/kinetic_ws/devel/share/common-lisp
ROS_DISTRO=kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros

gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609

PLUS: 
Cartographer is installed in a new separate workspace.

danningwang, 2018-05-28:
It is already solved. The reason is my ros problem, not protobuf problem.

closed 2018-05-28 by danningwang

lzn27, 2018-06-30:
I have the same problem, can you tell me how to fix it ? @danningwang

liguanyi11111, 2019-02-04:
I have the same problem too, can you tell me how to fix it ? @danningwang @lzn27

liguanyi11111, 2019-02-04:
It is already solved. Check your ${PROTOBUF_LIBRARY} in cartograoher CMakeLists.