cartographer-project/cartographer_ros #885: cartographer on a quadruped robot

nyxaria, 2018-05-30:
For my next project, I plan on building a quadruped robot, and at the moment and planning on how to implement SLAM. I have some ideas on how to implement the needed elements for cartographer to work, but am unsure if this is the right path for implementing SLAM.

- For the laser, I am planning on using LIDAR-Lite v3 with a 360 degree diy jig. This LIDAR can measure at ~750Hz with some tinkering. 
- For odom, I cannot use encoders. Instead, I was thinking of using a stereo/RGB-D camera to get visual odometry. I would then use viso2_ros which publishes a nav_msgs/Odometry msg. 
- An IMU would not, I think, have a different implementation than a wheel-based robot.

In terms of these assumptions/plans, is:

1. Is a LIDAR viable on a robot which is not always flat (I know cartographer uses that assumption)? Would it be possible to compensate for this with the IMU? If not, would constraining the robot's body to stay level be good enough for cartographer to work? In moments where it is not flat, is there a way to tell cartographer to throw away new data?
2. Is there a point in even implementing visual odom? ie. would it help cartographer, hinder it or be useless?
3. Is there a way to use point clouds from Kinect-like cameras to replace the LIDAR? Is this a good solution?
4. Lastly, is cartographer the wrong path for this problem? If yes, what avenue would you recommend?

Thank you for your time!

gaschler, 2018-05-30:
2D Lidar with IMU for orientation correction sounds fine.

> robot which is not always flat (I know cartographer uses that assumption)? Would it be possible to compensate for this with the IMU? 

IMU can compensate that. Not sure how much your robot tilts, but most data you find in the docs was recorded by a backpack that does tilt.

> Is there a point in even implementing visual odom?

It's likely not worth the effort and work just fine without.

> point clouds from Kinect-like cameras

Possible, but Cartographer is currently not well tuned for the high noise in depth direction and requires a larger field of view.

AchmadFathoni, 2018-05-31:
@nyxaria I sent you an e-mail

closed 2020-10-16 by MichaelGrupp