cartographer-project/cartographer_ros #890: Help wanted: Test ROS Melodic packages

gaschler, 2018-06-06:
Dear community,

we are glad that our version 1.0 release candidate made it into ROS Melodic release, its packages are currently in the ROS testing repository.

It would help us tremendously if you install the `ros-melodic-cartographer*_1.0.0` packages and test a little bit that the packages are fine.

A prerequisite is that you have ROS Melodic installed.
Then, you can enable the *testing repositories* as explained in http://wiki.ros.org/ShadowRepository and (temporarily) install the latest version with 

`sudo apt-get install ros-melodic-cartographer-ros ros-melodic-cartographer-rviz`

After that, please verify that version 1.0 is installed, for instance with `rosversion cartographer_ros`.

Some suggestions for testing are of course running roslaunch demo_backpack_2d.launch with bag files, running rviz and visualize submap, running executables like cartographer_pbstream_to_ros_map.

Feel free to post results here (including successful ones) and mention your OS version and architecture.

Thank you!

MichaelGrupp, 2018-06-07:
Gave it a shot  on my private dinosaur laptop with Sandy Bridge CPU... aaand it crashed with an illegal instruction in `cartographer_node`. 

Seems like the instruction `vmovdqu64 ...` in the binary (see below) is the cause. This is an AVX instruction and is probably not compatible with the AVX version this CPU supports (only avx, **not** avx2 or later).

`(gdb) run`
```
Program received signal SIGILL, Illegal instruction.
0x0000555555666598 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, char const*) ()
```

`(gdb) disassemble 0x0000555555666598`
<details>
<summary>Click to show ASM dump</summary>

```
Dump of assembler code for function _ZStplIcSt11char_traitsIcESaIcEENSt7__cxx1112basic_stringIT_T0_T1_EEOS8_PKS5_:
   0x0000555555666520 <+0>:	push   %r12
   0x0000555555666522 <+2>:	push   %rbp
   0x0000555555666523 <+3>:	mov    %rsi,%rbp
   0x0000555555666526 <+6>:	push   %rbx
   0x0000555555666527 <+7>:	mov    %rdi,%rbx
   0x000055555566652a <+10>:	mov    %rdx,%rdi
   0x000055555566652d <+13>:	mov    %rdx,%r12
   0x0000555555666530 <+16>:	callq  0x55555564cc10 <strlen@plt>
   0x0000555555666535 <+21>:	movabs $0x7fffffffffffffff,%rcx
   0x000055555566653f <+31>:	sub    0x8(%rbp),%rcx
   0x0000555555666543 <+35>:	cmp    %rcx,%rax
   0x0000555555666546 <+38>:	ja     0x5555556665a6 <_ZStplIcSt11char_traitsIcESaIcEENSt7__cxx1112basic_stringIT_T0_T1_EEOS8_PKS5_+134>
   0x0000555555666548 <+40>:	mov    %rax,%rdx
   0x000055555566654b <+43>:	mov    %r12,%rsi
   0x000055555566654e <+46>:	mov    %rbp,%rdi
   0x0000555555666551 <+49>:	callq  0x55555564d220 <_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcm@plt>
   0x0000555555666556 <+54>:	lea    0x10(%rbx),%rdx
   0x000055555566655a <+58>:	mov    %rdx,(%rbx)
   0x000055555566655d <+61>:	mov    (%rax),%rcx
   0x0000555555666560 <+64>:	lea    0x10(%rax),%rdx
   0x0000555555666564 <+68>:	cmp    %rdx,%rcx
   0x0000555555666567 <+71>:	je     0x555555666598 <_ZStplIcSt11char_traitsIcESaIcEENSt7__cxx1112basic_stringIT_T0_T1_EEOS8_PKS5_+120>
   0x0000555555666569 <+73>:	mov    %rcx,(%rbx)
   0x000055555566656c <+76>:	mov    0x10(%rax),%rcx
   0x0000555555666570 <+80>:	mov    %rcx,0x10(%rbx)
   0x0000555555666574 <+84>:	mov    0x8(%rax),%rcx
   0x0000555555666578 <+88>:	mov    %rcx,0x8(%rbx)
   0x000055555566657c <+92>:	mov    %rdx,(%rax)
   0x000055555566657f <+95>:	movq   $0x0,0x8(%rax)
   0x0000555555666587 <+103>:	movb   $0x0,0x10(%rax)
   0x000055555566658b <+107>:	mov    %rbx,%rax
   0x000055555566658e <+110>:	pop    %rbx
   0x000055555566658f <+111>:	pop    %rbp
   0x0000555555666590 <+112>:	pop    %r12
   0x0000555555666592 <+114>:	retq   
   0x0000555555666593 <+115>:	nopl   0x0(%rax,%rax,1)
=> 0x0000555555666598 <+120>:	vmovdqu64 0x10(%rax),%xmm0
   0x000055555566659f <+127>:	vmovups %xmm0,0x10(%rbx)
   0x00005555556665a4 <+132>:	jmp    0x555555666574 <_ZStplIcSt11char_traitsIcESaIcEENSt7__cxx1112basic_stringIT_T0_T1_EEOS8_PKS5_+84>
   0x00005555556665a6 <+134>:	lea    0x1641d7(%rip),%rdi        # 0x5555557ca784
   0x00005555556665ad <+141>:	callq  0x55555564d010 <_ZSt20__throw_length_errorPKc@plt>
End of assembler dump.
```
</details>

---
* OS: Ubuntu 18.04
* Kernel: 4.15.0-22-generic
* CPU: [lscpu.txt](https://github.com/googlecartographer/cartographer_ros/files/2078320/lscpu.txt)

gaschler, 2018-06-07:
@MichaelGrupp thank you reporting this illegal instruction and tracking it down.
Let's follow up in https://github.com/ros-gbp/cartographer_ros-release/issues/10

I think the `-march=native -msse4.2` needs to be removed from the ros package before more testing on the package makes sense.

I will post here when the package is ready again for more testing.

MichaelGrupp, 2018-06-07:
Looks like the exact same problem being discussed there, thanks for pointing me to it.

mikaelarguedas, 2018-06-08:
New packages with https://github.com/ros-gbp/cartographer_ros-release/issues/10 resolved are available in the testing repository.
Thanks @MichaelGrupp and @cschuet for reporting and investigating the issue

gaschler, 2018-06-08:
+1 Dear community, we gladly appreciate your results testing the 1.0.0 packages.

MichaelGrupp, 2018-06-08:
Maybe an announcement of this issue in the mailing list reaches more people, don't think too many people regularly check the issues.

gaschler, 2018-06-08:
I tested the packages (expect -rviz) in a Bionic docker container and did not find any issues.

For reference, here are the commands I ran:
```bash
wget -P / https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b2-2016-04-05-14-44-52.bag
wget -P / https://storage.googleapis.com/cartographer-public-data/bags/backpack_3d/b3-2016-02-02-13-32-01.bag
roslaunch cartographer_ros offline_backpack_2d.launch no_rviz:=true bag_filenames:=/b2-2016-04-05-14-44-52.bag
# manually remove rviz from /opt/ros/melodic/share/cartographer_ros/launch/*backpack_3d.launch
roslaunch cartographer_ros offline_backpack_3d.launch bag_filenames:=/b3-2016-02-02-13-32-01.bag
roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=/b3-2016-02-02-13-32-01.bag
/opt/ros/melodic/bin/cartographer_pbstream_to_ros_map -pbstream_filename /b2-2016-04-05-14-44-52.bag.pbstream 
/opt/ros/melodic/bin/cartographer_rosbag_validate -bag_filename /b2-2016-04-05-14-44-52.bag
/opt/ros/melodic/bin/cartographer_rosbag_validate -bag_filename /b3-2016-02-02-13-32-01.bag
/opt/ros/melodic/bin/cartographer_migrate_serialization_format -original_pbstream_file /b2-2016-04-05-14-44-52.bag.pbstream -output_pbstream_file test.pbstream
/opt/ros/melodic/bin/cartographer_autogenerate_ground_truth -pose_graph_filename b2-2016-04-05-14-44-52.bag.pbstream -output_filename test.txt
roslaunch cartographer_ros assets_writer_backpack_2d.launch bag_filenames:=/b2-2016-04-05-14-44-52.bag pose_graph_filename:=/b2-2016-04-05-14-44-52.bag.pbstream
```

MichaelGrupp, 2018-06-08:
Works now also on the laptop mentioned above. 
```bash
# standard backpack stuff
roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=$(pwd)/b0-2014-07-11-10-58-16.bag
rosservice call /write_state "filename: 'test.pbstream'"
roslaunch cartographer_ros demo_backpack_2d_localization.launch load_state_filename:=$HOME/.ros/test.pbstream bag_filename:=$(pwd)/b0-2014-07-11-10-58-16.bag

# with launch files & data from https://github.com/magazino/cartographer_magazino
roslaunch cartographer_toru offline_toru_2d.launch bag_filenames:=$(pwd)/hallway_return.bag
roslaunch cartographer_ros visualize_pbstream.launch pbstream_filename:=$(pwd)/hallway_return.bag.pbstream
rosrun cartographer_ros cartographer_pbstream_map_publisher -pbstream_filename $(pwd)/hallway_return.bag.pbstream
rosrun cartographer_ros cartographer_pbstream_to_ros_map -pbstream_filename $(pwd)/hallway_return.bag.pbstream
```

cschuet, 2018-06-08:
Same here on my machine. All good. Thanks everyone who helped track this
down and fix it!

-- 

Christoph Schütte

Software Engineer

cschuet@google.com
+49 89 839309134

Google Germany GmbH

Erika-Mann-Straße 33

80636 München

Geschäftsführer: Paul Manicle, Halimah DeLaine Prado

Registergericht und -nummer: Hamburg, HRB 86891

Sitz der Gesellschaft: Hamburg

Diese E-Mail ist vertraulich. Wenn Sie nicht der richtige Adressat sind,
leiten Sie diese bitte nicht weiter, informieren Sie den Absender und
löschen Sie die E-Mail und alle Anhänge. Vielen Dank.

This e-mail is confidential. If you are not the right addressee please do
not forward it, please inform the sender, and please erase this e-mail
including any attachments. Thanks.

mikaelarguedas, 2018-06-08:
Looks like these will hit the main repository on Tuesday https://discourse.ros.org/t/preparing-for-melodic-sync-2018-06-08/5056/2

@cschuet @gaschler should we make an announcement on [ROS discourse](https://discourse.ros.org/c/general) to notify user of the release of 1.0.0 ?

cschuet, 2018-06-09:
@mikaelarguedas that’s a great idea! Can you send out an announcement?

gaschler, 2018-06-14:
I just tested the bionic 1.0 packages (excluding rviz) from the main ROS repository 
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-cartographer-ros/
(with the same commands as in https://github.com/googlecartographer/cartographer_ros/issues/890#issuecomment-395736971) and did not find any issues.

mikaelarguedas, 2018-06-18:
Sorry I was hoping to get around making an announcement this weekend and did not get to it. Thanks @cschuet for taking care of it !

cschuet, 2018-06-18:
@mikaelarguedas np thanks for all your help getting the package out!

closed 2018-06-18 by cschuet