cartographer-project/cartographer_ros #894: Cartographer 2d slam: BAD RESULTS

letsdoitagain, 2018-06-07:
gist:  https://gist.github.com/letsdoitagain/ed901f722212cba0685e037ce996cc01

files:  https://github.com/letsdoitagain/cartographerIssue

bagfile:  https://drive.google.com/file/d/1CM9fc06Cdgl7QrrGfBSrxvq-NOpCQ_Nv/view

Hello,

It's taking forever to produce a refine map for cartographer. I am tuning it for days but the results are EXTREMELY bad. I am using odometry and 2d lidar.
Below is the screen shot for cartographer:
[image]

It's worth noting that gmapping has produced an extremely good map when compared to cartographer. Below is the screenshot:
[image]

gaschler, 2018-06-11:
> TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 270

Please verify. This should be the number of range messages you receive within a physical lidar scan rotation.

> TRAJECTORY_BUILDER_2D.submaps.num_range_data = 270

This should be longer. It should be the number of range messages you receive within 5-60 s.

> I0607 22:06:51.339933 21394 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/hokuyo_lidar_scan" (frame_id: "velodyne"):
> Count: 7369  Min: 0.000000  Max: 0.155330  Mean: 0.050439
> [0.000000, 0.015533)	          ##########	Count: 3684 (49.993214%)	Total: 3684 (49.993214%)
> [0.015533, 0.031066)	                    	Count: 0 (0.000000%)	Total: 3684 (49.993214%)
> [0.031066, 0.046599)	                    	Count: 0 (0.000000%)	Total: 3684 (49.993214%)
> [0.046599, 0.062132)	                    	Count: 0 (0.000000%)	Total: 3684 (49.993214%)
> [0.062132, 0.077665)	                    	Count: 1 (0.013570%)	Total: 3685 (50.006786%)
> [0.077665, 0.093198)	                    	Count: 1 (0.013570%)	Total: 3686 (50.020355%)
> [0.093198, 0.108731)	          ##########	Count: 3681 (49.952503%)	Total: 7367 (99.972862%)
> [0.108731, 0.124264)	                    	Count: 1 (0.013570%)	Total: 7368 (99.986427%)

This distribution is weird. Does the component send duplicates?

> TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.

This is too short. It is usually at the order of `max_range`.

> E0607 22:05:58.179749 21394 rosbag_validate_main.cc:318] frame_id "velodyne" is send on multiple topics. It was seen at least on /hokuyo_lidar_scan and /velodyne_points

Please verify cartographer_ros only subscribes to one of them.

> TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

Please check if this is really better than the default.

Hope this helps.

As always, a good way to start is to tune local SLAM only first, as in the tuning guide (set optimize_every_n_nodes = 1e10).
Seeing your image, it should really be possible to tune that the drift after walking around the four hallways is < 1 meter.

SubMishMar, 2018-08-06:
@gaschler, I think the[ tuning guide](http://google-cartographer-ros.readthedocs.io/en/latest/tuning.html) as to set `POSE_GRAPH.optimize_every_n_nodes = 0` but here you have asked to set `POSE_GRAPH.optimize_every_n_nodes = 1e10` and your suggestion seems more correct  to me, can you tell me which one is actually correct?

cschuet, 2018-08-14:
See e.g. [here](https://github.com/googlecartographer/cartographer/blob/213882a9b3d8aee10b7ea059e732f1632d3684de/cartographer/mapping/internal/2d/pose_graph_2d.cc#L372) both of these essentially do the same thing.

closed 2020-10-16 by MichaelGrupp