cartographer-project/cartographer_ros #902: advice in calibrating indoor application with 1 VLP16 and IMU

billcai, 2018-06-15:
Hi all, 

I am using cartographer in a large indoor setting, with a single VLP16 and imu. Being new to ROS, I went through ROS tutorials, and also read other similar issues. I've managed to get the frame_ids right (I think), but I cannot seem to produce an accurate 3d map of the short test area with the lidar and imu mounted on a trolley. 

My rosbug_validate output is: [here](https://gist.github.com/billcai/161934151fc996139ef44cd1dbf9f81a).
My forked repo is: [here](https://github.com/billcai/cartographer_ros)
The file to run is vlp16-imu.launch. I used replay_vlp16-imu.launch to calibrate.
My bag file is: [here](https://drive.google.com/file/d/1gVylbblAA8JPZH8p9zABWJMHjm5DP3ln/view?usp=sharing)

Things that I have played around with include changing the `TRAJECTORY_BUILDER_3D.num_accumulated_range_data` and `POSE_GRAPH.optimize_every_n_nodes`. 

Thank you in advance for your help! Appreciate any and all feedback and comments. And also many thanks to contributors and cartographer authors.

gaschler, 2018-06-18:
Thanks for giving all information related to this issue.

I took a quick look at the text files.

> E0615 18:03:47.420536 21330 rosbag_validate_main.cc:350] frame_id "velodyne" on topic /velodyne_points has serialization time 1529100095.068097455 but sensor time 1528998386.107721000 differing by -101709 s.
> E0615 18:03:47.421025 21330 rosbag_validate_main.cc:350] frame_id "map" on topic /scan_matched_points2 has serialization time 1529100095.093976695 but sensor time 1528998386.107721000 differing by -101709 s.
> E0615 18:03:47.422089 21330 rosbag_validate_main.cc:350] frame_id "velodyne" on topic /velodyne_points has serialization time 1529100095.111101422 but sensor time 1528998386.208568000 differing by -101709 s.

Differing timestamps are incompatible with the output stage of cartographer_ros. SLAM only looks at the sensor=header timestamps, but the transform publisher extrapolates to "Time::now" = serialization time, which will lead to nowhere. SLAM may be running, but published tf are off, you won't see any results in rviz. Please fix this first.

> E0615 18:03:47.422263 21330 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "base_link". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)

What the warning says: You cannot use this bag as is as an input for cartographer because it contains the output.

> TRAJECTORY_BUILDER_3D.submaps.num_range_data = 10

This should be the number of messages you get within 5-30 seconds.

> TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 2

This should be the number of messages you get within one physical lidar rotation. (I think this should be way higher for the typical VLP16 driver.)

> <origin xyz="0 0 0" rpy="3.1416 0 0" />

IMU to lidar rotation *must* be correct for 3D. 
If unsure, use the 2D pipeline first until you are happy with the 2D results. "Large indoor" usually works perfect with 2D SLAM. 2D is easier to tune.

Hope that helps.

gaschler, 2018-06-18:
Another thing, you can enable log output

`POSE_GRAPH.optimization_problem.log_solver_summary = true`

After 3D SLAM is tuned and it converges, it will print the optimized IMU to Lidar rotation and you can copy that value to URDF.

jojo666666, 2018-06-18:
Can you tell me how to calibrate the lidar and camera? Thank you so much!

billcai, 2018-06-19:
Thank you @gaschler! I followed your advice and managed to tune appropriately.

closed 2018-06-19 by billcai

ludennis, 2018-08-30:
Hi @gaschler and @billcai,

I have turned on `POSE_GRAPH.optimization_problem.log_solver_summary = true`, and have spotted "IMU correction" throughout the mapping process. My question is: which correction should I use? 

Thanks in advance,

Dennis Lu

albertchang1989, 2018-11-23:
> Thank you @gaschler! I followed your advice and managed to tune appropriately.

@billcai Hi, Where can I get the solver smmuary?

Duke-Allen, 2019-02-28:
@billcai 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?

ChenHuanJian, 2019-04-02:
@gaschler 
[image]

ChenHuanJian, 2019-04-02:
@gaschler I think my URDF is problematic. As shown in the figure above, it is wrong in terms of effect. But when I set POSE_GRAPH.optimization_problem.log_solver_summary = true, the log output from the terminal shows IMU correction was: 0 deg (1, 0, 0, 0).
At this time, I am confused, is the backend not implementing optimization or closed-loop detection caused by incorrect mapping?
In addition, I also tried to run 2D, from the results, the effect is OK,
From the information I provided, can you help me analyze the possible reasons for this?

[image]

ChenHuanJian, 2019-04-02:
@gaschler The output of the validation tool are as follows.
W0402 19:37:59.987740 27452 rosbag_validate_main.cc:353] frame_id "horizontal_vlp16_link" on topic /points2 has serialization time 1554129541.329265789 but sensor time 1554129541.004609000 differing by -0.324657 s.
E0402 19:38:00.142844 27452 rosbag_validate_main.cc:351] frame_id "horizontal_vlp16_link" on topic /points2 has serialization time 1554129622.013921476 but sensor time 1554129621.451861000 differing by -0.56206 s.
E0402 19:38:00.205253 27452 rosbag_validate_main.cc:351] frame_id "horizontal_vlp16_link" on topic /points2 has serialization time 1554129671.615146824 but sensor time 1554129671.058153000 differing by -0.556994 s.
E0402 19:38:00.387591 27452 rosbag_validate_main.cc:383] Point data (frame_id: "horizontal_vlp16_link") has a large gap, largest is 0.531893 s, recommended is [0.0005, 0.05] s with no jitter.
I0402 19:38:00.387670 27452 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/points2" (frame_id: "horizontal_vlp16_link"):
Count: 1217  Min: 0.042261  Max: 0.531893  Mean: 0.228924
[0.042261, 0.0912242)	                    	Count: 9 (0.739523%)	Total: 9 (0.739523%)
[0.0912242, 0.140187)	                    	Count: 10 (0.821693%)	Total: 19 (1.56122%)
[0.140187, 0.189151)	             #######	Count: 402 (33.032%)	Total: 421 (34.5933%)
[0.189151, 0.238114)	               #####	Count: 295 (24.2399%)	Total: 716 (58.8332%)
[0.238114, 0.287077)	               #####	Count: 304 (24.9795%)	Total: 1020 (83.8127%)
[0.287077, 0.33604)	                 ###	Count: 173 (14.2153%)	Total: 1193 (98.0279%)
[0.33604, 0.385003)	                    	Count: 21 (1.72555%)	Total: 1214 (99.7535%)
[0.385003, 0.433967)	                    	Count: 1 (0.0821693%)	Total: 1215 (99.8357%)
[0.433967, 0.48293)	                    	Count: 0 (0%)	Total: 1215 (99.8357%)
[0.48293, 0.531893]	                    	Count: 2 (0.164339%)	Total: 1217 (100%)
I0402 19:38:00.388211 27452 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "imu_link"):
Count: 34840  Min: 0.00232616  Max: 0.016077  Mean: 0.0080025
[0.00232616, 0.00370125)	                    	Count: 1 (0.00287026%)	Total: 1 (0.00287026%)
[0.00370125, 0.00507633)	                    	Count: 2 (0.00574053%)	Total: 3 (0.00861079%)
[0.00507633, 0.00645142)	                    	Count: 12 (0.0344432%)	Total: 15 (0.043054%)
[0.00645142, 0.00782651)	                   #	Count: 1459 (4.18772%)	Total: 1474 (4.23077%)
[0.00782651, 0.0092016)	 ###################	Count: 33345 (95.709%)	Total: 34819 (99.9397%)
[0.0092016, 0.0105767)	                    	Count: 8 (0.0229621%)	Total: 34827 (99.9627%)
[0.0105767, 0.0119518)	                    	Count: 6 (0.0172216%)	Total: 34833 (99.9799%)
[0.0119518, 0.0133269)	                    	Count: 0 (0%)	Total: 34833 (99.9799%)
[0.0133269, 0.014702)	                    	Count: 1 (0.00287026%)	Total: 34834 (99.9828%)
[0.014702, 0.016077]	                    	Count: 6 (0.0172216%)	Total: 34840 (100%)