cartographer-project/cartographer_ros #904: Issues in mapping with a IMU and VLP16(map overlapped)

ZhouYixuanRobtic, 2018-06-19:
Hello cartographer developers,
 Thanks for your excellent works. I am a freshman to cartographer and ROS. I try to generate map by using a IMU and VLP16, however something wrong must happened in my configuration files I can't get right results.
## Results
  Just like the figure below, my map overlapped.
[image]
 I wanna know what makes this wrong.
## Bag Validate
  I'm so sorry for that I can't use gits, so I post this screenshot
[image]
As this screenshot says, my velodyne_points topic must be published at a rate more than 20 HZ, while I can't find out how to change that rate. I use velodyne driver from [Ros-driver/velodyne](https://github.com/ros-drivers/velodyne). Can you help me in this? Or where I can get some correlative information? Appreciate much.
## Configuration files and bag
My configuration files can be found in my [github](https://github.com/ZhouYixuanRobtic/cartographer_ros),or in this [Configuration files.zip](https://github.com/googlecartographer/cartographer_ros/files/2115276/Configuration.files.zip).
My bag file can be found in this [bag.zip](https://github.com/googlecartographer/cartographer_ros/files/2115282/bag.zip).

I do apologize if this post disturbed any of you. Thanks a lot whether this issue can be answered or not!

gaschler, 2018-06-19:
The reason for strong circular drift is almost always a wrong rotation config between IMU and lidar.
You need to correct the imu-lidar transform in your urdf file.

The cartographer_rosbag_validate output doesn't seem complete. What's on the screen shot looks fine, please post text next time.

ZhouYixuanRobtic, 2018-06-20:
Thank you for noticing this issue.
## urdf file
I modified urdf file  as the way I put IMU and LIDAR (same origin and same X/Y axis direction). I don't understand how to correct the urdf file, so may you please specify how to do so? Thanks again for your help.
By the way, my urdf file can be found in [here](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/urdf/vlp16imu3d.urdf).
## Bag validate
You mean this line,
> E0620 09:24:09.715848  2936 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.100966 s, recommended is [0.0005, 0.05] s with no jitter.

**don't** influence cartographer mapping. Am I right?
And complete the cartographer_rosbag_validate output is posted below
>rosrun cartographer_ros cartographer_rosbag_validate -bag_filename imu_velodyne.bag 
>I0620 09:24:09.715647  2936 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "imu_link"):
>Count: 8394  Min: 0.009754  Max: 0.010286  Mean: 0.010000
>[0.009754, 0.009807)	                    	Count: 1 (0.011913%)	Total: 1 (0.011913%)
>[0.009807, 0.009860)	                    	Count: 2 (0.023827%)	Total: 3 (0.035740%)
>[0.009860, 0.009914)	                    	Count: 20 (0.238265%)	Total: 23 (0.274005%)
>[0.009914, 0.009967)	                 ###	Count: 1344 (16.011436%)	Total: 1367 (16.285442%)
>[0.009967, 0.010020)	        ############	Count: 4933 (58.768169%)	Total: 6300 (75.053612%)
>[0.010020, 0.010073)	               #####	Count: 2023 (24.100548%)	Total: 8323 (99.154160%)
>[0.010073, 0.010126)	                    	Count: 68 (0.810102%)	Total: 8391 (99.964264%)
>[0.010126, 0.010179)	                    	Count: 2 (0.023827%)	Total: 8393 (99.988091%)
>[0.010179, 0.010233)	                    	Count: 0 (0.000000%)	Total: 8393 (99.988091%)
>[0.010233, 0.010286]	                    	Count: 1 (0.011913%)	Total: 8394 (100.000000%)
>E0620 09:24:09.715848  2936 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.100966 s, recommended is [0.0005, 0.05] s with no jitter.
>I0620 09:24:09.716104  2936 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
>Count: 831  Min: 0.100763  Max: 0.100966  Mean: 0.100858
>[0.100763, 0.100783)	                    	Count: 11 (1.323706%)	Total: 11 (1.323706%)
>[0.100783, 0.100804)	                   #	Count: 37 (4.452467%)	Total: 48 (5.776173%)
>[0.100804, 0.100824)	                  ##	Count: 90 (10.830325%)	Total: 138 (16.606499%)
>[0.100824, 0.100844)	                 ###	Count: 143 (17.208183%)	Total: 281 (33.814682%)
>[0.100844, 0.100864)	               #####	Count: 217 (26.113117%)	Total: 498 (59.927799%)
>[0.100864, 0.100885)	                ####	Count: 161 (19.374249%)	Total: 659 (79.302048%)
>[0.100885, 0.100905)	                  ##	Count: 102 (12.274368%)	Total: 761 (91.576416%)
>[0.100905, 0.100925)	                   #	Count: 48 (5.776173%)	Total: 809 (97.352585%)
>[0.100925, 0.100946)	                    	Count: 19 (2.286402%)	Total: 828 (99.638992%)
>[0.100946, 0.100966]	                    	Count: 3 (0.361011%)	Total: 831 (100.000000%)

Thanks again for your kindly help

gaschler, 2018-06-20:
> By the way, my urdf file can be found in here.
>    `<origin xyz="0 0 0" rpy="0 0 0" />`

Correct the IMU rotation. Format is roll, pitch, yaw in radians.

> E0620 09:24:09.715848 2936 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.100966 

Agree, this is unrelated to your issue, and less severe.
The remaining part of the validate results seems fine.

ZhouYixuanRobtic, 2018-06-20:
Thanks gaschler for your generous help. 
To be honest, I'm confused about this rotation. I know its format,however I don't know what's the right values. Should these values be satisfied with the way I put the IMU and LIDAR in real time? 
## 
I put the LIDAR and IMU in this way
[image]
and my new urdf files can be foud in [here](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/urdf/new_VLP16.urdf)
Intuitively, in rviz the urdf can be visualized like
[image].
##
One more question, when I run cartographer with my own sensor,it gave out this INFO
>  INFO: I0620 16:43:13.000000 22598 collated_trajectory_builder.cc:72] imu rate: 100.00 Hz 1.00e-02 s +/- 3.39e-05 s (pulsed at 99.47% real time)
> INFO: I0620 16:43:13.000000 22598 collated_trajectory_builder.cc:72] points2 rate: 9.91 Hz 1.01e-01 s +/- 4.18e-05 s (pulsed at 100.02% real time)

When I run the official demo, I found that it gave out this INFO
>  INFO: I0620 16:43:13.000000 22598 collated_trajectory_builder.cc:72] imu rate: 250.04 Hz 4.00e-03 s +/- 3.01e-03 s (pulsed at 99.47% real time)
> INFO: I0620 16:43:13.000000 22598 collated_trajectory_builder.cc:72] points2 rate: 1507 Hz 6.64e-04 s +/- 1.37e-04 s (pulsed at 100.02% real time)

Dose this difference matter?
## 
I hope these preliminary questions won't disturb you. Thanks very much for your patience.

ZhouYixuanRobtic, 2018-06-26:
Hello @gaschler, I found there's something wrong with my imu data. My imu data's units are wrong. Then I corrected them, now I got this picture. I think no submaps are generated, right?
## Results
[image]

## Bag and configuration files
My configuration files can be found in [here](https://github.com/ZhouYixuanRobtic/cartographer_ros/tree/master/cartographer_ros/cartographer_ros)
My new record bag file can be found in [here](https://drive.google.com/open?id=1PwQc-F0a03lG3Cvd5rD5OQXnlLa8aU0w)

So still my urdf is wrong? Appreciate your generous help much.

gaschler, 2018-06-26:
> I think no map is generated while with some submaps, right? 

For submaps, please look at the left tab in rviz in click into the submaps menu. Please report how many seconds it takes until a new submap is created.

(The screenshot is not the best way to explain the issue, but I can see at least one noisy submap.)

"Map" is an undefined/overloaded term. If you mean occupancy grid and want such topic published, you need to run occupancy_grid_node.

ZhouYixuanRobtic, 2018-06-26:
Thanks your advice @gaschler . 
## Submap gap
I got these lines (I think the gap is what you wanna know?)
> Trajectory 0
0.160
1.00

Then it changes to 
> 0.320
1.320

So I think it takes ** 0.160 ** to create a new submap.
## Occupancy_grid_node
I run rqt_graph and I am sure I've run occupancy_grid_node already.

gaschler, 2018-06-26:
> 0.320
> 1.320

If it always stays just two submaps, which is the initial value, a submap is never finished.
Does it always stay just two submaps?

Most likely, you need to correct your values for
`TRAJECTORY_BUILDER_3D.num_accumulated_range_data` and `TRAJECTORY_BUILDER_3D.submaps.num_range_data` (probably make them smaller).

Also, please take a look at the terminal output if the motion filter is always active (it should not filter 100%).

> So I think it takes ** 0.160 ** to create a new submap.

My question was more at which period submaps are created, this is just the delay of the first one.

ZhouYixuanRobtic, 2018-06-26:
> Does it always stay just two submaps?

No, I just report two lines.
> TRAJECTORY_BUILDER_3D.num_accumulated_range_data and TRAJECTORY_BUILDER_3D.submaps.num_range_data

I've changed `TRAJECTORY_BUILDER_3D.num_accumulated_range_data` from 160 to 1 and `TRAJECTORY_BUILDER_3D.submaps.num_range_data` from 160 to 40.But still noisy submaps just like aforementioned picture.

> please take a look at the terminal output

I post my output below
> [INFO][1530005111.921205134]: I0626 **17:25:11**.000000 17762 submap_3d.cc:321] Added submap 1
>[ INFO][1530005111.921274619]: I0626 17:25:11.000000 17762 map_builder_bridge.cc:130] Added trajectory with ID '0'.
>[ INFO][1530005112.185521820]: I0626 17:25:12.000000 17762 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636656019121713630'.
>[ INFO][1530005121.796776418]: I0626 **17:25:21**.000000 17762 submap_3d.cc:321] Added submap 2
>[ INFO][1530005127.004939724]: I0626 17:25:27.000000 17762 collated_trajectory_builder.cc:72] imu rate: 99.92 Hz 1.00e-02 s +/- 2.65e-04 s (pulsed at 99.41% real time)
>[ INFO][1530005127.005067929]: I0626 17:25:27.000000 17762 collated_trajectory_builder.cc:72] points2 rate: 9.91 Hz 1.01e-01 s +/- 4.60e-05 s (pulsed at 100.07% real time)
>[ INFO][1530005131.876605143]: I0626 17:25:31.000000 17762 submap_3d.cc:321] Added submap 3
>[ INFO][1530005132.453464787]: I0626 17:25:32.000000 17834 constraint_builder_3d.cc:296] Node (0, 40) with 170 points on submap (0, 0) differs by translation 0.02 rotation 0.011 with score 62.2%.
>[ INFO][1530005134.401066965]: I0626 17:25:34.000000 17762 submap_3d.cc:321] Added submap 4
>[ INFO][1530005134.518086208]: I0626 17:25:34.000000 17833 constraint_builder_3d.cc:296] Node (0, 13) with 166 points on submap (0, 1) differs by translation 0.05 rotation 0.006 with score 86.0%.
>[ INFO][1530005136.418121964]: I0626 **17:25:36**.000000 17762 submap_3d.cc:321] Added submap 5
>[ INFO][1530005139.624747867]: I0626 **17:25:39**.000000 17762 submap_3d.cc:321] Added submap 6
>[ INFO][1530005141.671432679]: I0626 17:25:41.000000 17762 submap_3d.cc:321] Added submap 7
>[ INFO][1530005142.035062183]: I0626 17:25:42.000000 17762 collated_trajectory_builder.cc:72] imu rate: 100.00 Hz 1.00e-02 s +/- 3.71e-05 s (pulsed at 100.03% real time)
>[ INFO][1530005142.035189698]: I0626 17:25:42.000000 17762 collated_trajectory_builder.cc:72] points2 rate: 9.91 Hz 1.01e-01 s +/- 4.19e-05 s (pulsed at 100.01% real time)
>[ INFO][1530005146.107174682]: I0626 17:25:46.000000 17762 submap_3d.cc:321] Added submap 8
>[ INFO][1530005156.194111645]: I0626 17:25:56.000000 17762 submap_3d.cc:321] Added submap 9
>[ INFO][1530005157.064094877]: I0626 17:25:57.000000 17762 collated_trajectory_builder.cc:72] imu rate: 100.00 Hz 1.00e-02 s +/- 3.67e-05 s (pulsed at 100.02% real time)
>[ INFO][1530005157.064295662]: I0626 17:25:57.000000 17762 collated_trajectory_builder.cc:72] points2 rate: 9.91 Hz 1.01e-01 s +/- 4.32e-05 s (pulsed at 100.01% real time)
>**[ INFO][1530005162.622981531]: I0626 17:26:02.000000 17762 motion_filter.cc:42] Motion filter reduced the number of nodes to 34.4%.**

> My question was more at which period submaps are created

Sorry for misunderstanding your question. To be honest, I still do not comprehend your requirements. Are you talking about the gap cartographer added new submaps?  Then it should be 10 seconds or 3 seconds as I marked them black.

Thanks  for your help. You're such a patience instructor whom I can't express my appreciate to .

Entropy512, 2018-06-29:
One thing (warning, I'm going from memory, I haven't touched Cartographer in over a month and I had to return the loaner Velodynes I was using so anything I was doing with Cartographer is on hold for quite some time) - One of the Velodyne nodes has an "npackets" option which determines how many packets are accumulated into a single point cloud message.  The default option accumulates, if I recall correctly, a full scan for single return mode and a half scan for dual return mode.  The reason the num_accumulated_range_data in the demo config files is so high is because the Cartographer demos were collected with npackets forced to 1.

I forget exactly which launch file I needed to change to force npackets to 1, but trust me - you really want to do this.  I had the exact same issues you were having until I made that change.

ZhouYixuanRobtic, 2018-07-09:
Sorry for late reply, I had a vacation these days. And thanks for your advice, I've tried your suggestion. 
I changed `napckets` parameter in [driver.cc](https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/driver/driver.cc) Line 79 and re-catkin_make them in ros. However things are not getting better. Still I have such noisy submaps. But thanks for providing me another direction to tune cartographer by check the sensor.It's really help. Thanks anyway.@Entropy512 .

ZhouYixuanRobtic, 2018-09-05:
Since I retry cartographer with 2D mode, I've got perfect map in 2d. Then I turn to 3D mapping and my submaps performs well. So I'll close this issue since it's been solved.

closed 2018-09-05 by ZhouYixuanRobtic

Duke-Allen, 2019-02-28:
@ZhouYixuanRobtic 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?

Duke-Allen, 2019-03-16:
@ZhouYixuanRobtic 
Hi, I correct my angular velocity on x and y axis, and change the urdf file. This is my new [bag](https://pan.baidu.com/s/1sySfAAGw8eBtpJ7elKCT6A).The password is bakh.

- I0316 11:43:52.342375  9917 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 18056  Min: 0.004353  Max: 0.013893  Mean: 0.010000
[0.004353, 0.005307)	                    	Count: 1 (0.005538%)	Total: 1 (0.005538%)
[0.005307, 0.006261)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.006261, 0.007215)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.007215, 0.008169)	                    	Count: 281 (1.556269%)	Total: 282 (1.561808%)
[0.008169, 0.009123)	                    	Count: 73 (0.404298%)	Total: 355 (1.966105%)
[0.009123, 0.010077)	 ###################	Count: 17197 (95.242577%)	Total: 17552 (97.208687%)
[0.010077, 0.011031)	                    	Count: 151 (0.836287%)	Total: 17703 (98.044968%)
[0.011031, 0.011985)	                    	Count: 266 (1.473194%)	Total: 17969 (99.518166%)
[0.011985, 0.012939)	                    	Count: 86 (0.476296%)	Total: 18055 (99.994461%)
[0.012939, 0.013893]	                    	Count: 1 (0.005538%)	Total: 18056 (100.000000%)
E0316 11:43:52.343436  9917 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101069 s, recommended is [0.0005, 0.05] s with no jitter.
I0316 11:43:52.343727  9917 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1791  Min: 0.100548  Max: 0.101069  Mean: 0.100801
[0.100548, 0.100600)	                    	Count: 10 (0.558347%)	Total: 10 (0.558347%)
[0.100600, 0.100652)	                    	Count: 7 (0.390843%)	Total: 17 (0.949190%)
[0.100652, 0.100704)	                    	Count: 10 (0.558347%)	Total: 27 (1.507538%)
[0.100704, 0.100756)	                   #	Count: 51 (2.847571%)	Total: 78 (4.355109%)
[0.100756, 0.100809)	       #############	Count: 1190 (66.443329%)	Total: 1268 (70.798439%)
[0.100809, 0.100861)	               #####	Count: 478 (26.689001%)	Total: 1746 (97.487434%)
[0.100861, 0.100913)	                    	Count: 21 (1.172529%)	Total: 1767 (98.659966%)
[0.100913, 0.100965)	                    	Count: 10 (0.558347%)	Total: 1777 (99.218315%)
[0.100965, 0.101017)	                    	Count: 9 (0.502513%)	Total: 1786 (99.720825%)
[0.101017, 0.101069]	                    	Count: 5 (0.279174%)	Total: 1791 (100.000000%)

[image]
The track is discrepant from the real movement. Could you help me?

ZhouYixuanRobtic, 2019-03-18:
@Duke-Allen I'm sorry for this late reply, I turn to another project and so on not working on cartographer anymore, but I can still give you some information. For the sake of using the real and in-time published topics rather than replayed from a bag you record, you should set your `us_sim_time` as **false** in your [demo_rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch). This parameter can be set as **true** together with bag play lines like 
```
<node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
```
May there are some other wrong parameter set, but I'm out of this project for months, so please forgive that I can not provide more advice. And I do recommend you turn to a 2d cartographer mode first and back to 3d mode then. Hope this answer may be useful.

Duke-Allen, 2019-03-19:
@ZhouYixuanRobtic 
Thanks for you reply. Should I set my `use_sim_time` as **false** in my demo_rslidar_3d.launch? I use a bag I record so I set it true. You means that only set the `<node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />` as true in real 3D cartographer?

Duke-Allen, 2019-03-19:
I also have a question. What is the angular velocity and linear-acceleration  unit when running 3D cartographer? I guess my parameters' unit may be wrong.

ZhouYixuanRobtic, 2019-03-19:
@Duke-Allen I presumed you were using a real robot and all topics were published in-time, because I didn't see your `palybag` node in your launch file. Since you are using a bag recorded, so keep `use_sim_time` as `true` is right. 
> What is the angular velocity and linear-acceleration unit when running 3D cartographer?

the angular velocity uses `rad/s`, linear-acceleration uses m/s^2

ChenHuanJian, 2019-04-02:
@ZhouYixuanRobtic My situation is basically the same as yours, and I feel that there is no post-processing and loopback detection (maybe the positional relationship given in urd is not accurate). I want to know how you first pass 2D, and then try 3D is OK. I ran the 2D effect very well, but 3D has a ghost image. I hope you can give me some advice.