cartographer-project/cartographer_ros #905: 3D Occupancy Grid

Sanderi44, 2018-06-19:
Hello, I have been working with cartographer for a while now and have been able to get pretty good results.  My main question is about how to save, load and view the 3D occupancy grid that is created by cartographer (in 3d mode).  I have been using the point cloud that I get from the assets_writer to describe my space, but would rather have an occupancy grid.  After examining the code and examples, I have realized that I can use the assets_writer with option 'action = "write_hybrid_grid"' to write the hybrid grid to a serialized format.  My questions are:

1. If anyone has a script to load, analyze and view this hybrid grid, and if not, what do I need to do?
2. Is there a difference between the hybridgrid and a 3d occupancy grid.  My thought is that the hybridgrid is a thresholded occupancy grid in which cells with probability higher than some value are set to 1 and below some value are set to 0.  Is that correct?
3.  If this hybrid grid is what I described, is it possible to get the raw 3d occupancy grid instead?

Thanks

gaschler, 2018-06-20:
> Is there a difference between the hybridgrid and a 3d occupancy grid. My thought is that the hybridgrid is a thresholded occupancy grid in which cells with probability higher than some value are set to 1 and below some value are set to 0. Is that correct?

`HybridGrid` is a sparse 3D grid with 15 Bit per voxel. So no, it does have a probability for each voxel and it *not thresholded*.
I think the name hybrid is to indicate is has both flat layers and dynamic, tree-like layers to organize the data.

>  assets_writer with option 'action = "write_hybrid_grid"'

It would run this batch processor: https://github.com/googlecartographer/cartographer/blob/master/cartographer/io/hybrid_grid_points_processor.cc 
(Haven't used it, I don't even know if any other processor can use the result.)
 
> view this hybrid grid

You can see projections of it in rviz. Or you need to implement something that reads the contents of `GetAllSubmapData`.

There is one important property of the 3D pipeline: It only writes into a few voxels in the hybrid grid around the lidar reflection. The default is `TRAJECTORY_BUILDER_3D.submaps.range_data_inserter.num_free_space_voxels = 2`.
So the grid is mostly empty, especially the high resolution one.
If you plan to export a dense occupancy grid (because you need free voxels to find free paths to navigate), you could of course try changing that parameter, but performance may be extremely slow.

Hope this answers your questions.

cschuet, 2018-08-14:
Answered.

closed 2018-08-14 by cschuet