cartographer-project/cartographer_ros #907: Sensor Bridge Warning Ignored Subdivision of a Laserscan

boesenbe, 2018-06-22:
Hi,

I started to work with cartographer and the examples provided. Now I try to use it for my own robot. 

I use gazebo to simulate the robot. It is publishing IMU and Laserscanner data like in the example from cartographer documentation. But when I start cartographer with the simulation I get a warning.

```
[ WARN] [1529659045.435797513, 765.124823928]: W0622 11:17:25.000000  4220 sensor_bridge.cc:203] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355975394560000 is not before current subdivision time 621355975394560000
```

To the **map** topic is nothing published as well.

To start cartographer I launch a launchfile from the **demo_backpack_2d**. 

Here is the output of my bagfile when I run `cartographer_rosbag_validate`

```
W0622 12:16:01.566934  5175 rosbag_validate_main.cc:103] frame_id imu_link time 737699000000: IMU linear acceleration is 0.00162683 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -4.68042e-11 -1.16851e-11 000.00162683
W0622 12:16:01.567340  5175 rosbag_validate_main.cc:103] frame_id imu_link time 737749000000: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0
W0622 12:16:01.567502  5175 rosbag_validate_main.cc:103] frame_id imu_link time 737799000000: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0
E0622 12:16:02.126138  5175 rosbag_validate_main.cc:365] Average IMU linear acceleration is 0.45241 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0622 12:16:02.126271  5175 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.05 s, recommended is [0.0005, 0.005] s with no jitter.
I0622 12:16:02.126328  5175 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu_link"):
Count: 3823  Min: 0.050000  Max: 0.050000  Mean: 0.050000
I0622 12:16:02.126552  5175 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "horizontal_laser_2d" (frame_id: "sick_s300_laser_back_link"):
Count: 7602  Min: 0.023000  Max: 0.034000  Mean: 0.025136
[0.023000, 0.024100)	                ####	Count: 1461 (19.218627%)	Total: 1461 (19.218627%)
[0.024100, 0.025200)	         ###########	Count: 4323 (56.866614%)	Total: 5784 (76.085241%)
[0.025200, 0.026300)	                ####	Count: 1588 (20.889240%)	Total: 7372 (96.974480%)
[0.026300, 0.027400)	                    	Count: 62 (0.815575%)	Total: 7434 (97.790055%)
[0.027400, 0.028500)	                    	Count: 47 (0.618258%)	Total: 7481 (98.408314%)
[0.028500, 0.029600)	                    	Count: 45 (0.591949%)	Total: 7526 (99.000263%)
[0.029600, 0.030700)	                    	Count: 19 (0.249934%)	Total: 7545 (99.250197%)
[0.030700, 0.031800)	                    	Count: 35 (0.460405%)	Total: 7580 (99.710602%)
[0.031800, 0.032900)	                    	Count: 17 (0.223625%)	Total: 7597 (99.934228%)
[0.032900, 0.034000]	                    	Count: 5 (0.065772%)	Total: 7602 (100.000000%)
I0622 12:16:02.126828  5175 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "vertical_laser_2d" (frame_id: "sick_s300_laser_front_link"):
Count: 7364  Min: 0.016000  Max: 0.055000  Mean: 0.025951
[0.016000, 0.019900)	                   #	Count: 297 (4.033134%)	Total: 297 (4.033134%)
[0.019900, 0.023800)	                 ###	Count: 1274 (17.300380%)	Total: 1571 (21.333514%)
[0.023800, 0.027700)	        ############	Count: 4259 (57.835416%)	Total: 5830 (79.168930%)
[0.027700, 0.031600)	                 ###	Count: 1100 (14.937534%)	Total: 6930 (94.106464%)
[0.031600, 0.035500)	                   #	Count: 189 (2.566540%)	Total: 7119 (96.673004%)
[0.035500, 0.039400)	                    	Count: 6 (0.081477%)	Total: 7125 (96.754481%)
[0.039400, 0.043300)	                    	Count: 19 (0.258012%)	Total: 7144 (97.012493%)
[0.043300, 0.047200)	                    	Count: 122 (1.656708%)	Total: 7266 (98.669202%)
[0.047200, 0.051100)	                    	Count: 85 (1.154264%)	Total: 7351 (99.823466%)
[0.051100, 0.055000]	                    	Count: 13 (0.176534%)	Total: 7364 (100.000000%)
```

Here is the link to the [repo](https://github.com/boesenbe/cartographer_issue) with all config files I used.

Here you will find my [bag file](https://www.dropbox.com/sh/nh8f3a37szr4cts/AADu6V8e0drKe1aUDHF8NnyMa?dl=0).

Many thanks guys !!!

gaschler, 2018-06-22:
> I use gazebo to simulate the robot.
> [ WARN] [1529659045.435797513, 765.124823928]: W0622 11:17:25.000000  4220 sensor_bridge.cc:203] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355975394560000 is not before current subdivision time 621355975394560000

Interesting. Perhaps the field `float32 time_increment` of laser scan messages is not set?
Can you please check and print one of the laser scan message you get (without range data)?

> W0622 12:16:01.567340  5175 rosbag_validate_main.cc:103] frame_id imu_link time 737749000000: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0

Some (or all) IMU messages are invalid.
However, this is fine as since you set `TRAJECTORY_BUILDER_2D.use_imu_data = false`.

> num_subdivisions_per_laser_scan = 90,

This seems unnecessarily high (because we do per point unwarping). I'm not sure if we support subdivisions with zero points, not sure if that can become an issue.

> TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 270

This should be worth one physical laser rotation.
So usually it is `number_of_lidar * num_subdivisions * num_laser_messages_per_sensor_rotation`.

> TRAJECTORY_BUILDER_2D.submaps.num_range_data = 270

(I think the is a misunderstanding that these values should match.)
`num_range_data` is how many physical laser rotations should be inserted into one submap.
A good choice is to have 5-30 seconds worth of laser rotations. So usually it is `20 / duration_sensor_rotation`.

boesenbe, 2018-06-27:
Hi,

Many thanks for the detailed explanation!

I changed the `float_32 time_increment` and `float_32 scan_time` to the values from the demo (cartographer_paper_deutsches_museum.bag) and now it is working. Should I get these two values from the documentation of my laser scanner? If I just give it random numbers lets say 0.1 for `time_increment` and  0.2 for `scan_time` it is still not working.

Cheers in advance!

gaschler, 2018-06-27:
> float_32 time_increment and float_32 scan_time

If you don't no know the per-point time increment, specify zero and don't do subdivisions.
scan_time is not read by cartographer AFAIK.

>  If I just give it random numbers lets say 0.1 for time_increment and 0.2 for scan_time it is still not working.

Why would you want to provide wrong input?

The most important thing is that the header time stamps are correct (and of course messages are non-empty and not duplicated...).
You could inspect your header time stamps with rqt_bag.
Hope this helps.

boesenbe, 2018-07-25:
I found another mistake as well, when I use **ira_laser_tools** to merge laserscanes Cartographer is not happy with it. When I give all laserscanes seperately into Cartographer it is working. Maybe interesting for someone that is reading this issue.

closed 2018-07-25 by boesenbe

Awacky, 2022-04-15:
> 您可以使用 rqt_bag 检查您的标头时间戳。

 I met the same problem with you ,have you solved it out?

mmc-n33, 2022-04-27:
Setting `num_subdivisions_per_laser_scan` to 1 suppressed the warning in my case.