cartographer-project/cartographer_ros #909: How to use Landmarks

nyxaria, 2018-06-26:
Hi,

Currently I'm trying to use the lamdmarks feature, and so have written a node that publishes LandmarkList messages to /landmark. This works as intended, however, my implementation relies on a callback that detects said landmarks which only happens sparsely. As a result, Cartographer hangs during the times when no messages are published to this topic as it seems to consider landmarks with the same priority as sensor data from, i.e., a laser. 

When no messages are published by my node, cartographer spams the following:

`` [/cartographer  WARN 1530018173.562266708, 1524734268.807942259]: W0626 15:02:53.000000 31787 ordered_multi_queue.cc:155] Queue waiting for data: (0, landmark) ``

When it receives a single message, this is outputted:

`` [/cartographer  INFO 1530018143.665043171, 1524734238.912012996]: I0626 15:02:23.000000 31787 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636603310389015659'. ``

Afterwards, it continues to log the "Queue waiting..." message.

Does landmark data have to be constantly published or does Cartographer also support feeding in single landmarks at sparse timestamps? I didn't find documentation on how to do it correctly.

cschuet, 2018-06-26:
In order to make Cartographer calculations deterministic we strictly time order sensor data using the collator. But you only know in what order to publish data when you have already seen data from all other sources. Sparse sources pose a problem to that strategy.

You should publish empty landmark messages when you have nothing observed to unblock the queue. BTW this is also done for fixed frame poses (e.g. GPS): if there is no fix, you publish an empty fixed frame pose message.

nyxaria, 2018-06-26:
Thanks for the quick response-- this seems to have fixed all the issues.

Does Cartographer require a specified frequency, or is 10Hz okay?

cschuet, 2018-06-26:
Hmm, I’d probably match my slowest sensor, so 10Hz sounds right.

-- 

Christoph Schütte

Software Engineer

cschuet@google.com
+49 89 839309134

Google Germany GmbH

Erika-Mann-Straße 33

80636 München

Geschäftsführer: Paul Manicle, Halimah DeLaine Prado

Registergericht und -nummer: Hamburg, HRB 86891

Sitz der Gesellschaft: Hamburg

Diese E-Mail ist vertraulich. Wenn Sie nicht der richtige Adressat sind,
leiten Sie diese bitte nicht weiter, informieren Sie den Absender und
löschen Sie die E-Mail und alle Anhänge. Vielen Dank.

This e-mail is confidential. If you are not the right addressee please do
not forward it, please inform the sender, and please erase this e-mail
including any attachments. Thanks.

nyxaria, 2018-06-26:
Thanks for your help @cschuet =)

MengNan-Li, 2018-08-08:
Hi，could you provide a demo that publishes the landmarker data. Thinks.
My email address is "lmn.cnu.g@gmail.com".

mojovski, 2018-10-17:
Does cartographer expect the landmark observations to contain the pose in the world (map) frame?
I actually would like to provide the observations only in the sensor frame (e.g. base_laser_link).

MichaelGrupp, 2018-10-17:
@mojovski No, the landmark observations are relative to the tracking frame of Cartographer. 

Here you can see the respective landmark cost function that is minimized during optimization: https://google-cartographer.readthedocs.io/en/latest/cost_functions.html#landmark-cost-function

Implementations of these cost functors are here: https://github.com/googlecartographer/cartographer/tree/master/cartographer/mapping/internal/optimization/cost_functions

pifon2a, 2018-10-25:
#1077 this might help. I uploaded a bag file with landmarks and instructions how to launch it.

MengNan-Li, 2018-11-01:
Thanks for your help @pifon2a

Lorjuo, 2018-12-12:
An additional question:
Currently, I think every landmark has to deliver a unique ID. If the same landmark is observed again, the same ID needs to be provided. However, quite often there is the case that the camera is able to detect the object and classify it into groups. E.g. the camera is able to detect traffic signs and to state that this sign is a "priority road sign" but it's not able to say if it's exactly the same sign as seen before.
Do you know if there is a possibility to handle these classified but indistinguishable types of landmarks?

pifon2a, 2018-12-13:
@Lorjuo, currently we do not support non-unique landmarks, but we have plans for 2019 to experiment with object instance reconciliation and also with tracking non-static objects.

mojovski, 2018-12-13:
Hi @Lorjuo and @pifon2a we identify the landmarks by its position assuming, that there is only one within the support of x meters. This makes them "unique". However, I dont have results demonstrating the improvement of the robot localization yet. To come probably next week or next year. ;)

closed 2020-10-17 by MichaelGrupp