cartographer-project/cartographer_ros #913: Wrong pose assigned when start new trajectory with initial pose

iceicehyhy, 2018-06-29:
Hi,

I'm trying the pure localization mode now, and i'm able to end the previous trajectory and add a new trajectory with initial pose.

However, the initial pose being added does not follow the instruction I sent. It somehow assigns the robot to another point in the map.

iceicehyhy, 2018-06-29:
Hi,

thanks for your prompt reply.

the initial pose is set exactly on the map_frame that specified in the config file. What i've done is, I wrote a simple script subscribe to /initial_pose topic that RVIZ published, everytime i got the pose sent from RVIZ, i'll call the service to end the previous trajectory and start a new trajectory with the pose that i got from RVIZ.

The interesting thing is , the initial pose is working on some maps, but the pose is not correct for others. I wonder is there anything i need to do when I build the map? Or  any other thing i need to do before i call the /write_state service to save the serialized slam state?

MichaelGrupp, 2018-06-29:
Your approach sounds good, that's how I'm also doing it more or less. 

This shouldn't be affected by mapping. The only thing that could come to my mind is that the frame ID of the initial pose message doesn't match the frame ID of Cartographer's map frame.

---

Side note: since today you can use the `/get_trajectory_states` service to conveniently determine the active trajectories that need to be finished in your script. See https://github.com/googlecartographer/cartographer_ros/commit/8bbec6bff77a7e63669db34b9023ad1811e741fd ([msg](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros_msgs/msg/TrajectoryStates.msg) / [srv](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros_msgs/srv/GetTrajectoryStates.srv))

iceicehyhy, 2018-06-29:
Coz the map I used is the grid map that generated by Cartographer occupancy grid node, is possible that the frame origin of the generated grid map differ from the origin of serialized state map ?

derekhe89, 2018-06-30:
I got the same problem, if you solve this problem, can you give me some advice?@ iceicehyhy

dan9thsense, 2018-06-30:
I would like to do the exact same thing.  @iceicehyhy can you share that script?

MichaelGrupp, 2018-06-30:
The `to_trajectory_id` parameter of the initial pose is important, this defines relative to which origin the pose is defined. See [here for the concept](https://github.com/googlecartographer/cartographer/blob/ce18ec72950974cf9ed7e818ee0693ed4ed05308/cartographer/mapping/pose_graph.h#L138) for and [here for the syntax](https://github.com/googlecartographer/cartographer/blob/915ebb19a68bfb1b1ff4e8fe8e972db9e5e32000/cartographer/cloud/internal/handlers/add_trajectory_handler_test.cc#L53) (ignore the dummy values).
The ID has to be the one that is used by your frozen, global map. Usually it is 0 (better: check with `/get_trajectory_states`, see above).

My initial pose handler code works very reliable, unfortunately I can't share it at the moment as it is currently embedded in an proprietary project.

zhenliysu, 2018-08-09:
@MichaelGrupp @iceicehyhy 
 Hi, guys. I set  -initial_pose="{to_trajectory_id=0,relative_pose={translation={0.0,0.0,0.},rotation={0.,0.,0.0}}}".
But it appear at the last submap of trajectory_id 0.  Then i changed relative_pose some times, it changes too. But not the correct position( or not the desired position). So i guess the position is relative about the last submap of trajectory_id 0.   So what is the reason? could you give me some advice about how to solve this problem, thanks.
I am sincerely looking forward to your reply.

MichaelGrupp, 2018-08-09:
Oh and: try setting 

```
-initial_pose="{
  to_trajectory_id=0,
  relative_pose={translation={0.0,0.0,0.},rotation={0.,0.,0.0}},
  timestamp = 0
}"
```
...the "timestamp" field is used to determine the trajectory node to which the initial pose is relative to, as far as I know. I always set it to 0, and it works fine for me.

zhenliysu, 2018-08-09:
@MichaelGrupp Thank you for your prompt reply. I will try to set  timestamp = 0. In my opinons, i think the origin of trajectory 0 is the first submap pose of trajectory 0. So i feel confused.

MichaelGrupp, 2018-08-09:
> i think the origin of trajectory 0 is the first submap pose of trajectory 0

Yes, but as far as I know the timestamp parameter can be used to select later poses in trajectory 0 as the relative frame (not sure, never used this feature). Maybe it uses the last known pose in the last submap by default if you don't specify it as `timestamp=0`.

@gaschler

zhenliysu, 2018-08-09:
@MichaelGrupp Yes, when i set timestamp=0, it works normally. thank you very much.

cschuet, 2018-08-14:
Closing as answered. Thanks @MichaelGrupp !

closed 2018-08-14 by cschuet

carl1218, 2018-10-31:
@MichaelGrupp ，
Hi,
I am running 2d slam online ,when i call rosservice call /finish_trajectory,and set the new trajectory to z-axis=5,i runned:
rosrun cartographer_ros cartographer_start_trajectory -configuration_basename=backpack_2d.lua -configuration_directory=/home/calyu/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -initial_pose="{to_trajectory_id=0,relative_pose={translation={0.0,0.0,5},rotation={0.,0.,0.0}}}
i can see the map look like this:
[image]

I want to save this map ,so than i runned rosservice call /finish_trajectory 1,but the Z axis will change to the initial position.(Z-axis = 0m).
look like this:
[image]

How can i to set the new trajectory in the new plane (Z-axis ≠ 0) and save this map? could you give me some advice about how to solve this problem, thanks.
I am sincerely looking forward to your reply.

carl1218, 2018-11-05:
@MichaelGrupp Thank you for your reply.
yes ,I want to have e.g multiple floors with 2D SLAM, how can i use separate 2D maps per floor, could you give me some advice about how to solve this problem? thank you.

carl1218, 2018-11-05:
@MichaelGrupp Thank you very much.

carl1218, 2018-11-11:
> I would just process and store the data separately and use separate Cartographer nodes for each floor. On-the-fly floor changes are not possible with 2D SLAM at the moment.
>
> If you want to have multiple 2D maps aligned with a z-offset in 3D space, you would need an additional coordinate frame to model the floors. If you use ROS, you can define an additional parent TF frame (let's call it "world"), in which your floor maps are located. The static transformation from  "world" to each map can be adjusted to model the floor height.
>         world       /     \      /       \ map1           map2

Hi，
 Thank you very much for your advice.
 Because I don't know much about the cartographer, in order to achieve the effect I need, can you help me modify it, of course, it's not free, I can pay for it.
 Expected for your  reply.

carl1218, 2018-11-14:
@MichaelGrupp
 Hi，
 Thank you very much for your advice.
 in order to achieve the effect I need? can you help me modify it, of course, it's not free.
 Expected for your  reply.

MarcelSoler, 2019-04-17:
@MichaelGrupp timestamp=0 info was very helpful. 
Maybe it would be nice to update the documentation with this.It was not easy to reach the solution.

Thank you

MichaelGrupp, 2019-04-17:
I know. The timestamp option is weird and we are currently working on a complete refactor of the start trajectory service that will simplify things a lot. See #1245 and the corresponding RFC.

This will break the current API of this service, but in this case it's a good thing 😉