cartographer-project/cartographer_ros #924: build error

forlinchow, 2018-07-06:
Source space: /home/fy/catkin_ws/src
Build space: /home/fy/catkin_ws/build_isolated
Devel space: /home/fy/catkin_ws/devel_isolated
Install space: /home/fy/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'cartographer_ros_msgs'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/fy/catkin_ws/build_isolated/cartographer_ros_msgs'
ninja: no work to do.
==> ninja -j4 -l4 in '/home/fy/catkin_ws/build_isolated/cartographer_ros_msgs'
[20/20] cd /home/fy/catkin_ws/build_is...v cartographer_ros_msgs/StatusResponse
==> ninja install in '/home/fy/catkin_ws/build_isolated/cartographer_ros_msgs'
[21/21] Install the project...
-- Install configuration: ""
-- Up-to-date: /home/fy/catkin_ws/install_isolated/_setup_util.py
-- Up-to-date: /home/fy/catkin_ws/install_isolated/env.sh
-- Up-to-date: /home/fy/catkin_ws/install_isolated/setup.bash
-- Up-to-date: /home/fy/catkin_ws/install_isolated/setup.sh
-- Up-to-date: /home/fy/catkin_ws/install_isolated/setup.zsh
-- Up-to-date: /home/fy/catkin_ws/install_isolated/.rosinstall
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/LandmarkEntry.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/LandmarkList.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/SensorTopics.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/StatusCode.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/StatusResponse.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/SubmapEntry.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/SubmapList.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/SubmapTexture.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/TrajectoryOptions.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/TrajectoryStates.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/metrics/HistogramBucket.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/metrics/Metric.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/metrics/MetricFamily.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/msg/metrics/MetricLabel.msg
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/FinishTrajectory.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/GetTrajectoryStates.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/ReadMetrics.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/StartTrajectory.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/SubmapQuery.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/srv/WriteState.srv
-- Up-to-date: /home/fy/catkin_ws/install_isolated/share/cartographer_ros_msgs/cmake/cartographer_ros_msgs-msg-paths.cmake
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/WriteState.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/LandmarkEntry.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/WriteStateRequest.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/HistogramBucket.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/SubmapTexture.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/GetTrajectoryStatesResponse.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/StatusCode.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/SubmapQueryResponse.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/SubmapList.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/MetricLabel.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/SubmapQuery.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/StartTrajectoryResponse.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/LandmarkList.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/FinishTrajectory.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/MetricFamily.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/Metric.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/StartTrajectory.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/SubmapEntry.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/FinishTrajectoryResponse.h
-- Up-to-date: /home/fy/catkin_ws/install_isolated/include/cartographer_ros_msgs/ReadMetrics.h
[... 229 lines cut ...]
-- Up-to-date: /home/fy/catkin_ws/install_isolated/lib/libceres.a
<== Finished processing package [2 of 5]: 'ceres-solver'

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/fy/catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/fy/catkin_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j4 -l4 in '/home/fy/catkin_ws/build_isolated/cartographer/install'
[4/360] Running C++ protocol buffer co...relative_scan_matcher_options_2d.proto
FAILED: cd /home/fy/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/fy/catkin_ws/build_isolated/cartographer/install -I /home/fy/catkin_ws/src/cartographer /home/fy/catkin_ws/src/cartographer/cartographer/mapping/proto/scan_matching/fast_correlative_scan_matcher_options_2d.proto
cartographer/mapping/proto/scan_matching/fast_correlative_scan_matcher_options_2d.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[4/360] Running C++ protocol buffer co...rapher/transform/proto/transform.proto
FAILED: cd /home/fy/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/fy/catkin_ws/build_isolated/cartographer/install -I /home/fy/catkin_ws/src/cartographer /home/fy/catkin_ws/src/cartographer/cartographer/transform/proto/transform.proto
cartographer/transform/proto/transform.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[4/360] Running C++ protocol buffer co...ommon/proto/ceres_solver_options.proto
FAILED: cd /home/fy/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/fy/catkin_ws/build_isolated/cartographer/install -I /home/fy/catkin_ws/src/cartographer /home/fy/catkin_ws/src/cartographer/cartographer/common/proto/ceres_solver_options.proto
cartographer/common/proto/ceres_solver_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[4/360] cd /home/fy/catkin_ws/build_is...d/cartographer/install/AllFiles.cmake"
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/fy/catkin_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/fy/catkin_ws/build_isolated/cartographer && /home/fy/catkin_ws/install_isolated/env.sh ninja -j4 -l4

cschuet, 2018-07-06:
Cartographer requires protobuf version 3. You have only proto2 on your system. `scripts/install_proto3.sh` shows how to compile and install it.

forlinchow, 2018-07-09:
Thank you very much. I solved that last week. I type "which protoc" and found the default protobuf version is 2, then I replaced it .