cartographer-project/cartographer_ros #932: Loop closure the nth

herrhannes, 2018-07-09:
Hello everybody,

I have the following setup (rather lowcost...)

- 1 Velodyne VLP16 mounted in a height of about 2m (I know that this is not ideal because of its limited vertical angle, right?), operating at 20Hz, which is also the publishing rate of its driver
- 1 Tinkerforge IMU Brick publishing at about 100Hz, obviously mounted at a different position, more to the center of the vehicle, about 1m away in z-direction, about 15cm in x-direction
- no external odometry
- the cartographer config is in the following [gist](https://gist.github.com/herrhannes/2d7bf87a70f65bd2bd2687d826d09320) and is based on something I found somewhere around here
- I cannot provide the urdf (respectively sdf), but in rviz, everything looks well in place

Now my findings:
- while the vehicle is moving, everything seems fine
- as soon as it is NOT, meaning standing completely still without vibrations etc. the loop closure destroys everything (the maps rolls and nicks relatively to the map, until it is pointing in a completely wrong direction). Until the loop closure, the map also still looks fine
- use_online_correlative_scan_matching does not change anything, so it cannot only be the IMU's fault, right?
- in the log_solver_summary, it states that the IMU is misaligned by about 90 to 250°, although it is positioned as it should (verified in rviz): z-axis up (positive acceleration due to gravity), x-axis in direction of the front, y-axis to the left side. This is correct, isn't it? (a simulation with gazebo leads to the same issue)

So, what can I do that the loop closure also optimizes the map correctly when not moving apart from deactivation it?

Best regards

herrhannes, 2018-07-10:
@gaschler Thank you for your reply. Here are the results:
[robag_validate with cartographer running](https://gist.github.com/herrhannes/daf3a587d09adae0a919347ee3f174c0)
[rosbag_validate w/o cartographer running](https://gist.github.com/herrhannes/6c8d2a0d1fa192d5be4e1ba5e69da13f)

Although cartographer is also doing the odom transform, odom->map diverges. That should no be the case, I assume?

gaschler, 2018-07-10:
rosbag_validate w/o cartographer running results look fine.

> odom->map diverges

The transform "map to odom" is the drift correction coming from loop closure and optimization. It may grow larger over time but not excessively.
"odom to base_link" is the transform coming from SLAM without loop clousure, so it does not jump but it drifts.

herrhannes, 2018-07-10:
The "odom to base_link" only differs slightly, which seems reasonable for me.
But the "map to odom" is not really translated but rather rotated, which should not happen.
E.g. "log_solver_summary" claims that the imu was misaligned by a very big angle (up to over 270°), which actually should not be the case.
Are there any errors in the assumptions I made in the first post?

FlorentRevest, 2018-07-16:
If you have a [recent cartographer](https://github.com/googlecartographer/cartographer/commit/7fcc4576a81100c67c1bb49700daf0754240a64e), could you try setting: `POSE_GRAPH.optimization_problem.use_online_imu_extrinsics_in_3d = false` ?
I have very similar problems to the ones you reported and this helped quite a bit.
However, I don't think it is a clean solution... I still think I might have something wrong with my IMU pose but I can't find anything.

closed 2020-10-17 by MichaelGrupp