cartographer-project/cartographer_ros #949: Does not read odometry data

kevindo94, 2018-07-25:
Hello. I am new to Cartographer so I was hoping that you could help me fix some problems.
When I launch the cartographer, I keep getting this message.
> Queue waiting for data: (0, odom)

Although the rqt_graph shows that the odometry data is being subscribed by the cartographer node.
I have a bag file that has imu, point cloud, odometry, and scan data.
I remapped the topics as below in the launch file "velodyne_3d.launch" that I made using "backpack_3d.launch". 
```
<remap from="points2" to="velodyne_points" />
<remap from="imu" to="/imu/data" />
<remap from="odom" to="/odometry/filtered" />
```
For the .lua file, "velodyne_3d.lua" I changed the tracking_frame, use_odometry, num_laser_scans, and num_point_clouds from "backpack_3d.lua"
```
  tracking_frame = "imu_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
```
I also modified "backpack_3d.urdf" as follows.
```
<robot name="cartographer_backpack_3d">
  <material name="orange">
    <color rgba="1.0 0.5 0.2 1" />
  </material>
  <material name="gray">
    <color rgba="0.2 0.2 0.2 1" />
  </material>

  <link name="imu_link">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

  <link name="velodyne">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.07" radius="0.05" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="odom">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.07" radius="0.05" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link" />

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu_link" />
    <origin xyz="0 0 0" rpy="3.1416 0 0" />
  </joint>

  <joint name="velodyne_joint" type="fixed">
    <parent link="base_link" />
    <child link="velodyne" />
    <origin xyz="0 0 0" rpy="0 0 0" />
  </joint>

  <joint name="odom_joint" type="fixed">
    <parent link="base_link" />
    <child link="odom" />
    <origin xyz="0 0 0" rpy="0 0 0" />
  </joint>
```
These are the files that I used.
[bag file](https://drive.google.com/open?id=1mroPsZ68YYU9nXPi9dBdkvYjlu9VC7Se)
[launch](https://drive.google.com/open?id=1rO0bffT6YHI_OW1JonC95vVN3nD5vCuy)
[lua](https://drive.google.com/open?id=13aOKg9IzEzZDjM6jE_9cbFD9ykNC92Hs)
[urdf](https://drive.google.com/open?id=1Xs3dBbtHR_SFSAuQwwk-rqB7Wmv5vjUl)

If I've made any mistakes, please tell me how I can fix them.
Thank you for your time.

gaschler, 2018-07-25:
Please provide the output of rosbag_validate and rosbag info of your bag file.
Perhaps the timestamps in your odometry messages are off?
Have you verified it works without using odometry input? 
When tuning a new setup, you should always start with a simple setup and go from there.

kevindokaist, 2018-07-26:
@gaschler 
This is what I got when I ran rosbag_validate.

```
kd@kd-H170-Gaming-3:~/Desktop/Data$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename 11-29-17-odom_imu_velodyne.bag
E0726 11:25:26.557795  4073 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1511943472.501245505 but sensor time 1511943218.709390144 differing by -253.792 s.
E0726 11:25:26.560966  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475402724830570, current one at time 636475402724830570
E0726 11:25:26.560982  4073 rosbag_validate_main.cc:326] frame_id "velodyne" is send on multiple topics. It was seen at least on /velodyne_points and /scan
E0726 11:25:26.561028  4073 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1511943472.521474820 but sensor time 1511943218.729313772 differing by -253.792 s.
E0726 11:25:26.561043  4073 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1511943472.541220599 but sensor time 1511943218.749345821 differing by -253.792 s.
E0726 11:25:26.562378  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475402725838990, current one at time 636475402725838990
E0726 11:25:26.562391  4073 rosbag_validate_main.cc:326] frame_id "velodyne" is send on multiple topics. It was seen at least on /velodyne_points and /scan
E0726 11:25:26.563320  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475402726848210, current one at time 636475402726848210
E0726 11:25:26.563331  4073 rosbag_validate_main.cc:326] frame_id "velodyne" is send on multiple topics. It was seen at least on /velodyne_points and /scan
E0726 11:25:26.708547  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475402938653020, current one at time 636475402938653020
W0726 11:25:26.814049  4073 rosbag_validate_main.cc:103] frame_id imu_link time 1511943507878688250: IMU linear acceleration is 41.0307 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -37.9301 06.31402 -14.3165
W0726 11:25:26.814100  4073 rosbag_validate_main.cc:103] frame_id imu_link time 1511943507882025108: IMU linear acceleration is 48.1366 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -46.3687 07.05701 -10.8295
W0726 11:25:26.814131  4073 rosbag_validate_main.cc:103] frame_id imu_link time 1511943507895289950: IMU linear acceleration is 33.2993 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -32.4882 0.477347 -7.28939
E0726 11:25:27.878453  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475404403461920, current one at time 636475404403461920
E0726 11:25:28.003545  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475404555760680, current one at time 636475404555760680
W0726 11:25:32.403448  4073 rosbag_validate_main.cc:203] Sensor with frame_id "velodyne" range measurements have longest overlap of 0 s
I0726 11:25:32.404682  4073 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu/data" (frame_id: "imu_link"):
Count: 73082  Min: 0.003157  Max: 0.030183  Mean: 0.010000
[0.003157, 0.005860)	                  ##	Count: 5952 (8.144277%)	Total: 5952 (8.144277%)
[0.005860, 0.008562)	               #####	Count: 18339 (25.093731%)	Total: 24291 (33.238007%)
[0.008562, 0.011265)	              ######	Count: 21531 (29.461428%)	Total: 45822 (62.699432%)
[0.011265, 0.013967)	             #######	Count: 24904 (34.076790%)	Total: 70726 (96.776222%)
[0.013967, 0.016670)	                    	Count: 957 (1.309488%)	Total: 71683 (98.085709%)
[0.016670, 0.019373)	                    	Count: 874 (1.195917%)	Total: 72557 (99.281631%)
[0.019373, 0.022075)	                    	Count: 479 (0.655428%)	Total: 73036 (99.937057%)
[0.022075, 0.024778)	                    	Count: 18 (0.024630%)	Total: 73054 (99.961685%)
[0.024778, 0.027480)	                    	Count: 1 (0.001368%)	Total: 73055 (99.963058%)
[0.027480, 0.030183]	                    	Count: 27 (0.036945%)	Total: 73082 (100.000000%)
I0726 11:25:32.405455  4073 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/odometry/filtered" (frame_id: "odom"):
Count: 36540  Min: 0.018734  Max: 0.021243  Mean: 0.020000
[0.018734, 0.018985)	                    	Count: 4 (0.010947%)	Total: 4 (0.010947%)
[0.018985, 0.019236)	                    	Count: 36 (0.098522%)	Total: 40 (0.109469%)
[0.019236, 0.019486)	                    	Count: 250 (0.684182%)	Total: 290 (0.793651%)
[0.019486, 0.019737)	                   #	Count: 1056 (2.889984%)	Total: 1346 (3.683634%)
[0.019737, 0.019988)	            ########	Count: 14609 (39.980843%)	Total: 15955 (43.664478%)
[0.019988, 0.020239)	          ##########	Count: 18297 (50.073891%)	Total: 34252 (93.738365%)
[0.020239, 0.020490)	                   #	Count: 1785 (4.885057%)	Total: 36037 (98.623428%)
[... cut ...]
[ INFO] [1532572475.690400676]: I0726 11:34:35.000000  5896 local_trajectory_builder_3d.cc:78] Range data collator filling buffer.
[ INFO] [1532572475.790546762]: I0726 11:34:35.000000  5896 local_trajectory_builder_3d.cc:78] Range data collator filling buffer.
[ INFO] [1532572475.890435760]: I0726 11:34:35.000000  5896 local_trajectory_builder_3d.cc:78] Range data collator filling buffer.
[ INFO] [1532572475.990725259]: I0726 11:34:35.000000  5896 local_trajectory_builder_3d.cc:78] Range data collator filling buffer.

```

gaschler, 2018-07-26:
Thanks for posting the validate results. 

> E0726 11:25:27.878453  4073 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636475404403461920, current one at time 636475404403461920

Fix this, this is not supported. (Are you sending double messages?)

> I0726 11:25:32.405814  4073 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"): Count: 14491  Min: 0.000000 

Same thing.

> E0726 11:25:26.561028  4073 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1511943472.521474820 but sensor time 1511943218.729313772 differing by -253.792 s.

This is mostly unsupported. cartographer_ros published poses will be extrapolated by that time and be miles off. Visualization won't work.

> And when I try without the odometry, I don't get any errors but the map doesn't build accurately.

Fix this first. Again, as always, get 2D SLAM working without loop closure and only using laser scans. Then add IMU, then switch to 3D. Then, add odometry if you wish. At the very last, enable loop closure (contraint search and optimization).

Perhaps your imu to lidar rotation is incorrect? We don't know.

kevindo94, 2018-07-26:
I'm sorry. I am still new to all of these so it may be a dumb question but what do you mean by 

> Fix this, this is not supported.

> Are you sending double messages?

Is it a matter of modifying lua, urdf, or launch files? or the bag file itself?

I tried doing 2D SLAM with just the scan data. So I created a new bag file by filtering out all the topics except /scan. This is what pops up if I run bag validate.

```
kd@kd-H170-Gaming-3:~$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename /home/kd/Desktop/Data/scan.bag 
E0726 20:13:10.939683 18025 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.103675 s, recommended is [0.0005, 0.05] s with no jitter.
I0726 20:13:10.940085 18025 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):
Count: 7245  Min: 0.100512  Max: 0.103675  Mean: 0.100870
[0.100512, 0.100828)	             #######	Count: 2418 (33.374741%)	Total: 2418 (33.374741%)
[0.100828, 0.101145)	       #############	Count: 4753 (65.603867%)	Total: 7171 (98.978607%)
[0.101145, 0.101461)	                    	Count: 42 (0.579710%)	Total: 7213 (99.558319%)
[0.101461, 0.101777)	                    	Count: 0 (0.000000%)	Total: 7213 (99.558319%)
[0.101777, 0.102094)	                    	Count: 0 (0.000000%)	Total: 7213 (99.558319%)
[0.102094, 0.102410)	                    	Count: 4 (0.055210%)	Total: 7217 (99.613525%)
[0.102410, 0.102726)	                    	Count: 3 (0.041408%)	Total: 7220 (99.654938%)
[0.102726, 0.103042)	                    	Count: 5 (0.069013%)	Total: 7225 (99.723946%)
[0.103042, 0.103359)	                    	Count: 13 (0.179434%)	Total: 7238 (99.903381%)
[0.103359, 0.103675]	                    	Count: 7 (0.096618%)	Total: 7245 (100.000000%)

```
Is there a problem when 

> Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):

shows up???

When I run 2D SLAM with just this scan topic, it doesn't work and only prints out

```
[ INFO] [1532607609.311631152]: I0726 21:20:09.000000 29579 submap_2d.cc:187] Added submap 1
[ INFO] [1532607609.311681388]: I0726 21:20:09.000000 29579 map_builder_bridge.cc:130] Added trajectory with ID '0'.
```
Then, nothing shows up. I tried looking at the topics like /tf, /map, and others, but nothing comes out of the cartographer_node.

Thanks for replying!

gaschler, 2018-07-26:
>>Fix this, this is not supported.
>>Are you sending double messages?

>Is it a matter of modifying lua, urdf, or launch files? or the bag file itself?

It's a matter of the nodes that publish the data. It's is a bug in the lidar driver node when it sends two messages with the same timestamp.

> Is there a problem when
> 
>> Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):
> 
> shows up???

No, the histogram is always output by the validate tool.

Actually, your newest validate results look fine.

> When I run 2D SLAM with just this scan topic, it doesn't work and only prints out

Please verify your lua number of laser scanners is correct and match the ros message type.
The newest version of cartographer_ros actually warns about such possible mismatches.
Post the entire log, look for log messages like "waiting for data (0, ...)".

And of course, verify rosbag play runs and simulated time is on.

kevindo94, 2018-07-27:
Like you suggested, I verified the number of laser scanners and matched the ros message type. 
At first I got this message.
```
[ WARN] [1532656205.040509478]: W0727 10:50:05.000000 26376 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1532656214.676527463]: W0727 10:50:14.000000 26376 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636475410032822220 is not before current subdivision time 636475402724830570
```
I did not publish the /imu. I only played the /scan topic. Is there an option in the lua file to make sure I'm won't use the imu data? In lua, I only see 
`imu_sampling_ratio = 1.,`

As for the subdivision part, I read in [one of the previous issues](https://github.com/googlecartographer/cartographer_ros/issues/907)

> > float_32 time_increment and float_32 scan_time

> 
> If you don't no know the per-point time increment, specify zero and don't do subdivisions.
> scan_time is not read by cartographer AFAIK.

So I changed the setting in lua. 
`num_subdivisions_per_laser_scan = 1,`

Then I ran cartographer again which displays no result. There are topics that are being published by the cartographer node, but the messages are empty. Also, I get the same message. 
`[ WARN] [1532658390.389158036]: W0727 11:26:30.000000   469 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)`

These are the [configurations](https://github.com/kevindo94/cartographer_ros) that I used. 
Thank you.

gaschler, 2018-07-27:
`TRAJECTORY_BUILDER_2D.use_imu_data = false`

For the 3D algorithm, IMU is required.

kevindo94, 2018-07-27:
I added
> TRAJECTORY_BUILDER_2D.use_imu_data = false
to the lua file. But still I had no luck. The cartographer runs without any error but no correct map is built. 

```
[ INFO] [1532685521.375467601]: I0727 18:58:41.000000 12363 submap_2d.cc:187] Added submap 1
[ INFO] [1532685521.375555645]: I0727 18:58:41.000000 12363 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1532685639.681121709]: I0727 19:00:39.000000 12363 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636475402724830570'.
[ INFO] [1532685639.681189977]: I0727 19:00:39.000000 12363 collated_trajectory_builder.cc:72] scan rate: unknown
[ INFO] [1532685654.708664227]: I0727 19:00:54.000000 12363 collated_trajectory_builder.cc:72] scan rate: 9.91 Hz 1.01e-01 s +/- 9.05e-05 s (pulsed at 99.98% real time)
[ INFO] [1532686089.486700550]: I0727 19:08:09.000000 12363 submap_2d.cc:187] Added submap 2
[ INFO] [1532686090.566329733]: I0727 19:08:10.000000 12363 collated_trajectory_builder.cc:72] scan rate: 9.91 Hz 1.01e-01 s +/- 9.61e-05 s (pulsed at 100.00% real time)
[ INFO] [1532686094.540213623]: I0727 19:08:14.000000 12432 constraint_builder_2d.cc:281] 0 computations resulted in 0 additional constraints.
[ INFO] [1532686094.540249803]: I0727 19:08:14.000000 12432 constraint_builder_2d.cc:283] Score histogram:
Count: 0
[ INFO] [1532686105.594410227]: I0727 19:08:25.000000 12363 collated_trajectory_builder.cc:72] scan rate: 9.91 Hz 1.01e-01 s +/- 1.02e-04 s (pulsed at 100.00% real time)
[ INFO] [1532686144.953881286]: I0727 19:09:04.000000 12363 motion_filter.cc:42] Motion filter reduced the number of nodes to 20%.
```

These were the messages that I got in running the 2D SLAM with only scan data (no imu, odometry).
The map that is built looks like the data that I see in rviz.

gaschler, 2018-07-27:
> The map that is built looks like the data that I see in rviz.

This means you see submaps in rviz?

Please see the rest of the tuning tutorial. Likely you add to much range data to a submap at this low laser frequency (only 10 Hz).

NZK1993, 2018-10-09:
@kevindo94 Hello,have you slove the problem? When I run cartographer using my own device ,I meet the same problem . If you solved the problem ,please tell me how to modify the  lua. Thanks!

MichaelGrupp, 2020-10-21:
Inactive.

closed 2020-10-21 by MichaelGrupp