cartographer-project/cartographer_ros #956: pure localization with live velodyne

hks95, 2018-07-26:
I was successful in running the pure localization with 2 bagfiles, one for map and other for sensor. is it possible to run pure localization with just the map coming from bag.pbstream and using a live VLP-16 and imu, instead of using the second bag file?
roslaunch cartographer_ros lead_localization_online.launch map_filename:=${HOME}/log/july26_map.bag.pbstream
It seems that when I launch velodyne+imu, the map doesnt show up on rviz when I run pure localization. The tf published by robot state publisher is not recognized. When i run pure localization first, the velodyne and  imu do not get launched. The sensor data is not published.

gaschler, 2018-07-27:
If you have issues which nodes are launched, please post the launch file as text that you tried.

> I was successful in running the pure localization with 2 bagfiles, one for map and other for sensor

Just to avoid confusion here. When you run 2 bag files without `load_state_from_file`, this is not what we call pure localization mode.

> is it possible to run pure localization with just the map coming from bag.pbstream and using a live VLP-16

Sure, this is the main use case.

hks95, 2018-07-27:
I have attached the launch file I am currently facing problems with.
[lead_localization_online.txt](https://github.com/googlecartographer/cartographer_ros/files/2235788/lead_localization_online.txt)

In the documentation, pure localization demo has two bag files. I am also not clear with the load state from file term you mentioned. Could you elaborate on that?

gaschler, 2018-07-27:
> not clear with the load state from file term you mentioned

I mean the argument `load_state_filename` of the node, like in demo_backpack_2d_localization.launch

```
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename backpack_2d_localization.lua
          -load_state_filename $(arg load_state_filename)"
      output="screen">
    <remap from="echoes" to="horizontal_laser_2d" />
  </node>
```

(You seem to be using an old version where it was named "map_filename".)

hks95, 2018-07-27:
Could you clarify why does the cartographer localization prevent my velodyne and imu sensor from launching? I first run the 'lead localization online' launch file i attached to this thread, then I try to launch VLP16_points.launch and my custom imu.launch but they do not show any sensor readings. 
When I try to launch the sensors first and then localization, the sensor readings are seen but the submaps from pbstream do not show up.

hks95, 2018-07-31:
Resolved the issue. I had to change the use sim time to false, run the sensor launch files and then run the pure localization launch file without the second bag file argument.

closed 2018-07-31 by hks95