cartographer-project/cartographer_ros #960: what data from the imu and odometry does cartographer use?

dan9thsense, 2018-07-31:
I know that it uses the gravity vector from the accelerometers, but am unable to find what other data it uses.  Does it use angular velocities from the imu?  Does it use linear velocities from odometry?

Is there a difference in what data is used for 2D vs 3D?

Thanks for the help.

elgarbe, 2018-08-11:
I'm working on generating IMU msg with an ICM20602 and a STM32F722 and publish it using rosserial. I have the same question. Do I need to provide orientation part of IMU msg? I have Angular velocity and Linear Acceleration.

Thank

ojura, 2018-08-11:
No need to provide orientation. Cartographer filters angular velocity and acceleration.

Note: you need the raw acceleration (with g = 9.81 m/s^2), NOT linear acceleration (which has gravity removed, so it's zero if not accelerating). Cartographer needs the gravity reference!

dan9thsense, 2018-08-16:
Does it use anything other than angular velocity and z axis acceleration?
What about odometry?
Is there any difference in 2D vs 3D in what it uses?

ojura, 2018-08-16:
It uses angular vecoltieis and all raw accelerations, not just the Z one. If you have a pitch or roll, gravity will not be fully in Z obviously. Cartographer needs all three axes' accelerations to estimate the gravity orientation.

dan9thsense, 2018-08-16:
Does it use anything from odometry that is not included in the imu?

elgarbe, 2018-08-25:
what if I have no good angular velocity?
I have raw gyro data from my IMU but it drift over time. This is my bag file. 
https://drive.google.com/open?id=1ZfoIJP8t_8jl2-QrdE5rl-UfR8lS5YSL

When I enable IMU on cartographer I can see fast rotations that don't correspond with my movement.
and a short video:
https://drive.google.com/open?id=11RhA5scSWGDt3ZGeVXt-jpElN3B6YyqT

is my gyro data a problem or is my accel data the problem?

MichaelGrupp, 2020-10-21:
Closing due to inactivity.

closed 2020-10-21 by MichaelGrupp