cartographer-project/cartographer_ros #962: SLAM 3D with cartographer

alfred31, 2018-07-31:
Hello,
I have been working with cartographer a couple of months. Since After having read all the documentation I managed to get a good performance for both 2d and 3d. The point is that I would like to get a map sharper and more clear, like the gotten with the 2d SLAM.
Is there anyway to detect height changes with the 2D SLAM?

Thanks in advance,.
Regards

gaschler, 2018-07-31:
> I managed to get a good performance for both 2d and 3d

Always glad to hear!

> anyway to detect height changes with the 2D SLAM?

Not sure what you mean. Can you explain your the setup, what results you currently get and the expected behavior you'd like to get?

alfred31, 2018-08-01:
I have a rotary hokuyo to perform a 3d visualization. I would like to get a good 3d slam performance so i can get the information of the z axis meaning that I have either changed the floor where the robot stands  or not. Plus, I would like to have a  high quality map  so I can send a pose to the robot in he map. The thing is that the quality of the 2D map is awesome, while the one that I am getting from the 3D SLAM is not hat good (at least in the test in which i tried to perform a floor change). I have changed lots of parameters like the submaps resolution to get a map with a higher quality. I have seen in this video https://www.youtube.com/channel/UCbRcsb-eNL7cviFj9y0QqHQ uploaded by you from the deutches museum that the results are also awesome for the 3d SLAM, since the submaps look like the ones that I have obtained for the 2d SLAM, adding the Z information. 
[image]
This is what I am getting changing going downstairs and upstairs a couple o times. However I did another test in a different environment and the resuls were way better, so I don't know if it can be caused due to a lack of features within this environment.

Thanks a lot for your support!!

alfred31, 2018-08-01:
The question is if I am missing something to get a performance simiar to the video.

Regards

alfred31, 2018-08-01:
Sorry, I meant this video https://www.youtube.com/watch?v=DM0dpHLhtX0

dan9thsense, 2018-08-08:
I am seeing something very similar.  I get great 2d maps with a SICK lidar.  However, with a velodyne producing 3D data, the 3d SLAM produces maps that look just like the one shown here.  It does not generate solid walls.

gaschler, 2018-08-08:
@alfred31 @dan9thsense 
Glad you are interested for tuning 3D SLAM. However, you'd need to provide more technical details before the community can help here (please see the issue template).
A common reason is that 3D requires an accurate transform between IMU and lidar sensors (set in the urdf), but there are several other settings to look at.
Also check your input data with cartographer_rosbag_validate and post the *text* output here.

Duke-Allen, 2019-02-22:
@alfred31 @gaschler 
I am tuning in 3D slam with imu and rs_16 lidar. But it no map output.
It displays "No tf data, Actual error: [map] does not exist" and the robotmodel' status is error.
my files:
demo_rslidar_3d.launch:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
rslidar_3d.launch:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
mydemo_3d.rviz:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
rslidar_scan_3d.lua:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
rslidar_2d.urdf:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
I don't know how to solve it. Could you help me?

MichaelGrupp, 2020-10-21:
Closing due to inactivity...

closed 2020-10-21 by MichaelGrupp