cartographer-project/cartographer_ros #963: assets writer map output doesnt show all obstacles

hks95, 2018-07-31:
I am using the assets writer map function given in this repo to convert my cartographer pbstream output to a map server compatible map that can be used with the navigation stack. With the default parameter values, a lot of obstacles that show up on the submaps do not appear in the final map pgm.  
[image]

I tried modifying the hit/miss probabilities in the lua file but it gets worse than the default. Any insights on how this can be solved?

hks95, 2018-08-01:
It seems like the assets writer considers only the last few portions of the pbstream file. I noticed that the obstacles appear with a much smaller bag file where I moved around it twice. Is there any way to increase the capability to retain history in cartographer

cschuet, 2018-08-13:
@SirVer Can you explain this behavior?

Leisa2019, 2019-08-12:
Hey there, 
I've been having the same issue using the assets writer map function as well. @hks95 , did you find a solution yet? I also tried changing the parameters in the lua-file like you with no success.

When i tried out @SirVer 's hunch #1, I kind of got a better result obsacle-wise (so the shapes are better) but first it is all grey and black (for my application, I actually need the nuances) and second of all it also does not represent the seen rViz Map either. 

SirVer's solution #2 kind of makes sense to me. But on the other hand, following that logic, changing the hit/miss probabilitys should actually make a difference for the better right? 
I would really appreciate some help. :)

facontidavide, 2019-09-23:
Having the same problem. As anyone found a solution?

facontidavide, 2019-10-09:
Related to this issue. I think adding a min_z and max_z to the asset_writer should be very simple (I looked at the source code).

Would the cartographer team accept a PR with this additional feature?

facontidavide, 2019-10-10:
The problem reported here can be solved with this PR:

Pull request already sent: https://github.com/googlecartographer/cartographer/pull/1636

A typical config will look like

```lua
options = {
    tracking_frame = "base_link",
    pipeline = {
      {
        action = "min_max_range_filter",
        min_range = 1.,
        max_range = 60.
      },
      {
        action = "vertical_range_filter",
        min_z = -0.2,
        max_z = 0.5
      },
      {
        action = "write_ros_map",
        range_data_inserter = {
          insert_free_space = true,
          hit_probability = 0.55,
          miss_probability = 0.49,
        },
        filestem = "map",
        resolution = 0.05
      }
    }
}
```

closed 2020-06-09 by wohe