cartographer-project/cartographer_ros #965: Getting Cartographer's final pose graph optimized trajectory

SubMishMar, 2018-08-01:
I want to get the final optimized trajectory generated by cartographer in some sort of text or json file and I went through some of the issues which suggested that on running rosservice call /finishtrajectory 0 a protobuf file will be generated inside ./ros but no such file was generated in my case. FYI, I am not using the offline node. 

Is there anyway to fix this?

Thanks.

ojura, 2018-08-01:
You also have to call the service for saving the state.

SubMishMar, 2018-08-01:
@ojura thanks for your responce, is it ``/write_state`` ?

SubMishMar, 2018-08-01:
@ojura so I rosserive called /write_state and a binary file was generated inside .ros folder. How to read this file?
Thanks.

ojura, 2018-08-01:
Have a look here: https://github.com/googlecartographer/cartographer_ros/issues/460

SubMishMar, 2018-08-01:
Thanks!
I suppose that the discussion in the above mentioned issue asks me to run ``cartographer_pbstream_map_publisher -pbstream_filename <output_filename>`` and running it prompts the following output in the command line:
``.ros$ cartographer_pbstream_map_publisher -pbstream_filename output
I0801 10:29:45.845443 16013 pbstream_map_publisher_main.cc:56] Loading submap slices from serialized data.
[ INFO] [1533144585.845735264]: I0801 10:29:45.000000 16013 pbstream_map_publisher_main.cc:56] Loading submap slices from serialized data.
I0801 10:29:45.942812 16013 pbstream_map_publisher_main.cc:76] Generating combined map image from submap slices.
[ INFO] [1533144585.942857743]: I0801 10:29:45.000000 16013 pbstream_map_publisher_main.cc:76] Generating combined map image from submap slices.
I0801 10:29:45.950736 16013 pbstream_map_publisher_main.cc:92] Publishing occupancy grid topic map (frame_id: map, resolution:0.050000).
[ INFO] [1533144585.950760768]: I0801 10:29:45.000000 16013 pbstream_map_publisher_main.cc:92] Publishing occupancy grid topic map (frame_id: map, resolution:0.050000).

ojura, 2018-08-01:
That tool publishes a map from a .pbstream. You can open rviz and visualize it there. It‘s not frozen, it‘s just continuing to publish the map to other nodes („spinning“ in ROS lingo).

SubMishMar, 2018-08-01:
Yes I can see the map in rviz, thanks. But I want the trajectory node list not the map.

ojura, 2018-08-01:
There isn't a tool for that, but you could easily write one yourself, like this:
https://gist.github.com/ojura/85507cd6c8be419357305b30d9bb949d

SubMishMar, 2018-08-01:
Thanks @ojura I was able to compile it.

closed 2018-08-01 by SubMishMar