cartographer-project/cartographer_ros #968: Fail to produce 2D map with /write_state

SeanGuoXuan, 2018-08-06:
1. rosbag_validate result: https://github.com/SeanGuoXuan/cartographer_ros/blob/master/validation_result
2. Git repository:  https://github.com/SeanGuoXuan/cartographer_ros
3. bag file: https://drive.google.com/open?id=1xDRRJ1txde-9ZLwNU759GMT1ab_ljJMx

Hi, 
I am using Kinect2 and razor imu to produce 2D map. This is the command for online mode:

> roslaunch cartographer_ros demo_kinect_2d.launch bag_filename:=subset.bag

Below is the screen shot:
[image]

Then, I run those two services to produce pbstream file:

> rosservice call /finish_trajectory 0
> rosservice call /write_state subset.bag.pbstream

The file is created. However, when I execute command:

> roslaunch cartographer_ros assets_writer_kinect.launch bag_filenames:=subset.bag pose_graph_filename:=subset.bag.pbstream

I got the following error:

> I0806 23:46:19.241068  7152 configuration_file_resolver.cc:41] Found '/home/sean/sean/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/assets_writer_kinect.lua' for 'assets_writer_kinect.lua'.
> I0806 23:46:19.241536  7152 configuration_file_resolver.cc:41] Found '/home/sean/sean/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
> I0806 23:46:19.241559  7152 configuration_file_resolver.cc:41] Found '/home/sean/sean/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
> I0806 23:46:19.242890  7152 assets_writer.cc:188] Processing /home/sean/sean/data_repository/cartographer/subset.bag...
> I0806 23:46:19.323696  7152 assets_writer.cc:254] Processed 0 of 78.6413 bag time seconds...
> W0806 23:46:19.722715  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723078  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723129  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723171  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723218  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723268  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723366  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723407  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> W0806 23:46:19.723450  7152 xray_points_processor.cc:147] Not writing output: bounding box is empty.
> I0806 23:46:19.723475  7152 counting_points_processor.cc:43] Processed 0 and finishing.

Can someone explain what cause this error?

Thanks in advance.

cschuet, 2018-08-13:
@SirVer Why would the asset generation fail like this?

mojovski, 2018-10-01:
I think I got a similar issue. What file size do you have, @SeanGuoXuan ? Mine is 98bytes

jub92, 2018-11-27:
@mojovski I also get a file with size of 98bytes. Have you found out anything? I am trying to write out a pbstream file containing the state created from a 3.5 minute short bag record. Parameters set so that only one submap should be built.

eugenfunk, 2018-12-05:
Yes, Pass true for saving unfinished submaps

mojovski, 2018-12-05:
Yes, pass true for saving unfinished submaps

closed 2020-10-21 by MichaelGrupp