cartographer-project/cartographer_ros #970: cartographer_rosbag_validate gives segmentation fault

SubMishMar, 2018-08-08:
On running `rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>` for my bag file I had a segmentation fault. My bag file has  3 topics and has the following shows the info regarding it.

```
path:        2018-08-07-15-53-39.bag
version:     2.0
duration:    3:19s (199s)
start:       Aug 03 2018 14:59:19.63 (1533333559.63)
end:         Aug 03 2018 15:02:38.76 (1533333758.76)
size:        29.5 MB
messages:    40793
compression: none [39/39 chunks]
types:       nav_msgs/Odometry     [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/LaserScan [90c7ef2dc6895d81024acba2ac42f369]
             tf2_msgs/TFMessage    [94810edda583a504dfda3829e70d7eec]
topics:      /odom   19913 msgs    : nav_msgs/Odometry    
             /scan     992 msgs    : sensor_msgs/LaserScan
             /tf     19888 msgs    : tf2_msgs/TFMessage
```
But running the validator node gives me a segfault and the segfault happens at: https://github.com/googlecartographer/cartographer_ros/blob/41c74de649fca92b55c66ab0f2ee26be1c84ce69/cartographer_ros/cartographer_ros/rosbag_validate_main.cc#L222

Since all the points in a laser scan should have the same timestamp, I just made the following change 
 `*from = *to` and it worked without any fault. What am I missing if I make this change?

ojura, 2018-08-08:
Please post the stack trace.

Not all points physically have the same time - the lidar is rotating. For 3D, they are typically 5 us apart. IIRC for 2D lidars this time is calculated from scan metadata, and for 3D you can provide it on a separate channel in the PointCloud2. 

This is used for scan unwarping, where Cartographer applies the velocity estimate to unwarp the scan.

SubMishMar, 2018-08-08:
Okay I will post the stack trace but my scans are extracted from a point cloud, I dont actually use a lidar and I am doing 2d slam here. I kind of make sure that each scan has the same time stamp for all the points. Still, I will post the stack for your perusal and may be post the bag file too. Thanks.

koelsinha, 2019-03-11:
Hi,

I am getting the same error. I know how to use prefix with rosrun open a program in gdb to get a trace. I am not sure what is the syntax of the same with this tool. I am new to this so can you please tell me how to get a stack trace?

koelsinha, 2019-03-11:
@cschuet  
Hi,

I am getting the same error. I know how to use prefix with rosrun open a program in gdb to get a trace. I am not sure what is the syntax of the same with this tool. I am new to this so can you please tell me how to get a stack trace?

MichaelGrupp, 2020-10-21:
Duplicate of https://github.com/cartographer-project/cartographer_ros/issues/1405, please answer there if there are new insights.

closed 2020-10-21 by MichaelGrupp