cartographer-project/cartographer_ros #971: Locate failed in 2D pure localization mode

ZhouYixuanRobtic, 2018-08-08:
Hi, thanks for your excellent works, I'm working on cartographer pure localization mode now. I have several issues with it.
- **What makes odom position updated?**
   In official 2D pure localization demo,I mean 
```
roslaunch cartographer_ros demo_backpack_2d_localization.launch \
   load_state_filename:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag.pbstream \
   bag_filename:=${HOME}/Downloads/b2-2016-04-27-12-31-41.bag

```
I notice that after a while the odom position (parent-map) changed,while in my own demo the odom position sticks to (0,0,0).  **I guess the pure localization mode works because it compares the load_state map and new map to locate the robot. Am I right?** Is there anything needed that I don't notice?

- **What makes the new map generated then vanished?**
  In official 2D localization demo, I noticed that a new map generated then vanished? However in my own demo **the new map generated then kept exist**. What's wrong?

- Files needed
Here's my [launch](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/launch/demo_velodyne2d_localization.launch) ,[urdf](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/urdf/velodyne2d.urdf) ,[configuration](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/configuration_files/velodyne2d_localization.lua).
Here's my [bag](https://drive.google.com/open?id=1dTt4Csu5cBY1eoC7FFwmm1lD-vwWOUqA).
Here's my [bag_validate](https://github.com/ZhouYixuanRobtic/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/bag_validate.txt).

ZhouYixuanRobtic, 2018-08-08:
Thanks your reply,@zhenliysu . 
> I guess "provide_odom_frame = true" makes odom position updated.

In [here](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html?highlight=tf), it described what the parameter `provide_odom_frame` makes. So I suppose this just provides the odom frame rather than updates.

 I guess the pure localization mode works because it compares the load_state map and new map to locate the robot . Am I right?  If so how this process works?

I suppose these [screenshots](https://drive.google.com/open?id=1rcSh44f6zwYo9OwZZk1vuI5eh0YQt5ia) may be helpful

zhenliysu, 2018-08-08:
@ZhouYixuanRobtic Yes, iagree with you about  "provide_odom_frame". In my opinions, odom will update while pure_localization works. that is, pure_localization update the robot position and odom position  in map. 
if you odom position sticks to (0,0,0), maybe you should check your Fixed Frame  in Global Options(it should be map instead of odom).
i hope it's useful.

ZhouYixuanRobtic, 2018-08-08:
Thanks for your generous help,@zhenliysu . I checked my fixed frame in global options, however it's map. 
**So in other words, in pure localization mode, the initial pose of trajectory 1 is generated by cartographer after a while,right?** 
I understand we can set the initial pose manually by 
```
rosrun cartographer_ros cartographer_start_trajectory -configuration_basename=velodyne2d_localization.lua -configuration_directory=/home/xcy/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/ -initial_pose="{to_trajectory_id=0,relative_pose={translation={1.0,-0.2,0.},rotation={0.,0.,0.}}}"
```
But this command works with trajectory 2.

I'm struggling with the consistent initial pose for a while. 
The trajectory 1 always started at the origin of trajectory 0 (landmark (0,0) ) rather than the actual position in real space. In official demo the new trajectory also started at the landmark(0,0) but it quickly changed while my demo didn't.
By the way, my two bags just last about 30 seconds. **Do you think short_term data causes the localization failed?**
Many thanks

zhenliysu, 2018-08-09:
@ZhouYixuanRobtic  I think  the initial pose of trajectory 1 is updated by cartographer global SLAM.  if your initial pose not update, maybe the global SLAM have not found  a good match.

ZhouYixuanRobtic, 2018-08-09:
Thanks your reply, @zhenliysu . I agree with you about the global SLAM didn't find a good match. Then can you give me some suggestions about how to make it have a good match?
Many thanks.

zhenliysu, 2018-08-09:
you can try run  pure_localization for a long time. And through this way, you can get a result.
This is my options, i hope it's helpful.

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.5,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 2e-2,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}
--pure_localization
POSE_GRAPH.optimize_every_n_nodes = 1
MAP_BUILDER.num_background_threads = 8
POSE_GRAPH.global_sampling_ratio = 0.003
POSE_GRAPH.constraint_builder.sampling_ratio = 0.3
POSE_GRAPH.constraint_builder.min_score = 0.65

--pure_localization end

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35
TRAJECTORY_BUILDER_2D.min_range = 0.05
TRAJECTORY_BUILDER_2D.max_range = 25.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

POSE_GRAPH.optimization_problem.huber_scale = 1e2

TRAJECTORY_BUILDER.pure_localization = true

return options

ZhouYixuanRobtic, 2018-08-10:
Much appreciate your generous help. @zhenliysu . I've tried your advice, however it didn't work except improved the submaps quality. Then I noticed that may the simple environment causes failed localization. To be more specific, I used to run pure localization mode in indoor lab with big glasses,and now I try it around the corridor. Then I run it right.
[image]

So here A recommendation
**Run cartographer in a "complex" environment,especially with distinct walls**
Also leave my [succeed bag](https://drive.google.com/open?id=1WrLwXO_x-hhbo9hQ6AzhgVgQY8DMqbWp) (12 MB) for others who are troubled with similar  issues.My launch files can be found above comments

closed 2018-08-10 by ZhouYixuanRobtic