cartographer-project/cartographer_ros #974: Build failed for cartographer_ros package

stephijung, 2018-08-09:
using ROS Kinetic, Ubuntu 16.04 & current version of cartographer_ros

I followed the provided [installation guide](https://google-cartographer-ros.readthedocs.io/en/latest/index.html) but use a local protobuf installation: 
`catkin_make_isolated --install --use-ninja \
  -DCMAKE_PREFIX_PATH="${PWD}/install_isolated;${PWD}/protobuf/install;${CMAKE_PREFIX_PATH}"`

 After successful build for cartographer, ceres-solver & cartographer_ros_msgs following error message appear during build process of cartographer_ros package:
```shell 
[10/38] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/tf_bridge.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/djungste/cartographer_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -isystem /home/djungste/cartographer_ws/protobuf/install/include -isystem abseil/src/abseil -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/tf_bridge.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/tf_bridge.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/tf_bridge.cc.o -c /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/tf_bridge.cc
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/tf_bridge.cc:17:45: fatal error: cartographer/common/make_unique.h: No such file or directory
compilation terminated.
[10/38] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/sensor_bridge.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/djungste/cartographer_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -isystem /home/djungste/cartographer_ws/protobuf/install/include -isystem abseil/src/abseil -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/sensor_bridge.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/sensor_bridge.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/sensor_bridge.cc.o -c /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc
In file included from /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:17:0:
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h:22:42: fatal error: cartographer/common/optional.h: No such file or directory
compilation terminated.
[10/38] Building CXX object CMakeFiles/cartographer_ros....r/cartographer_ros/ros_map_writing_points_processor.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/djungste/cartographer_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -isystem /home/djungste/cartographer_ws/protobuf/install/include -isystem abseil/src/abseil -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map_writing_points_processor.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map_writing_points_processor.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map_writing_points_processor.cc.o -c /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.cc
In file included from /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.cc:17:0:
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.h:24:77: fatal error: cartographer/mapping/2d/proto/range_data_inserter_options_2d.pb.h: No such file or directory
compilation terminated.
[10/38] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/djungste/cartographer_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -isystem /home/djungste/cartographer_ws/protobuf/install/include -isystem abseil/src/abseil -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o -c /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc: In function ‘std::unique_ptr<nav_msgs::OccupancyGrid_<std::allocator<void> > > cartographer_ros::CreateOccupancyGridMsg(const cartographer::io::PaintSubmapSlicesResult&, double, const string&, const ros::Time&)’:
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc:308:7: error: ‘make_unique’ is not a member of ‘cartographer::common’
       ::cartographer::common::make_unique<nav_msgs::OccupancyGrid>();
       ^
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc:308:7: note: suggested alternative:
In file included from /home/djungste/cartographer_ws/install_isolated/include/cartographer/mapping/id.h:30:0,
                 from /home/djungste/cartographer_ws/install_isolated/include/cartographer/io/submap_painter.h:24,
                 from /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.h:22,
[... cut ...]
In file included from /home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.cc:17:0:
/home/djungste/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.h:24:77: fatal error: cartographer/mapping/2d/proto/range_data_inserter_options_2d.pb.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
```

I would be very grateful if somebody can help or give a hint how to solve this build problem.

cschuet, 2018-08-13:
We had some breakage in the repo. Should be fixed now. Please sync to head and try again. Sorry for the inconvenience.

closed 2018-08-13 by cschuet

Avinaashmani, 2022-02-13:
> using ROS Kinetic, Ubuntu 16.04 & current version of cartographer_ros
>
> I followed the provided [installation guide](https://google-cartographer-ros.readthedocs.io/en/latest/index.html) but use a [...]

same, i have the exact same issue......Could anyone help me.