cartographer-project/cartographer_ros #975: Could I run 3D Cartographer Without some IMU message?

Rolanding, 2018-08-10:
I have a 32-HDL LIDAR sensor, and I'm trying to run 3D cartographer. 
But I can only get the following IMU messages :
1. heading
2. pitch
3. roll
And also I have a  odometry. But there are no  linear acceleration and Gyroscope axis rate messages.
So how can I run 3D cartographer with this kind of situation.
Any advice would be very appreciated.

gaschler, 2018-08-10:
Angular velocities are the important part of IMU for the 3D pipeline. If you can provide correct angular velocities, it will likely work with a fake gravity vector pointing in the right direction. 
See https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/cartographer_turtlebot/flat_world_imu_node_main.cc for an example of that work around.

As always, tune first the 2D pipeline until that runs well before trying out 3D.

Rolanding, 2018-08-10:
@gaschler Thanks for your reply.
As I can get the messages of heading and pick and roll , I'm wondering weather if it is possible to get the angular velocities by using the methods of differential and linear interpolation.

Rolanding, 2018-08-10:
@gaschler And of course, It's going to bring a some deviation.