cartographer-project/cartographer_ros #976: If there's pose topic published already,or I need to calculate it by my own?

ZhouYixuanRobtic, 2018-08-10:
Hi,I'm working with cartographer 2D pure localization mode and it works well.Thanks for developers' excellent work. 
In localization mode, I wanna get the robot pose which is not published (May published,but I didn't see it in `rostopic list`). **I am wondering if there's pose topic publihsed already,or I need to calculate it by my own?**
I notice that in the tf topic,the cartographer published the tf between map and odom also odom and base-link. Should I use these two tf to calculate the robot pose?

cschuet, 2018-08-13:
We publish the pose of the robot as a TF transform. The frame_ids depend on your configuration, but usually base_link (this is your robot) to map (this is origin of the map you are localizing against).

ZhouYixuanRobtic, 2018-08-14:
Thanks for your help. My query is well answered. Thanks again for your kindly help. Now I'll close this issue.

closed 2018-08-14 by ZhouYixuanRobtic