cartographer-project/cartographer_ros #977: Filtering out artifacts due to ambient light interference in sunny outdoor environments

sabetAI, 2018-08-10:
I'm using a cheap 2D lidar with poor ambient light resistance to map an outdoor environment. This means I get small artifacts near the robots that show up in the map ie a stray set of sparse points that look like small islands with the 2d costmap enabled. I'd like to filter these points out so they don't interfere with the robot's navigation. I've thought of some ways to do this (ie by point density in moving window, or by removing points with distant nearest neighbours). _Where in the code is best to add this functionality_?

cschuet, 2018-08-13:
Have you considered writing a ROS node that does the filtering a republishes the cleaned up point clouds? That's what I would do.

sabetAI, 2018-08-13:
Yeah that's what I decided to do. Thanks!

closed 2018-08-13 by cschuet